davidusb-geek/emhass #195: days_to_retrieve default is documented wrong

tomasbedrich, 2024-02-12:
**Describe the bug**
Default value of `days_to_retrieve` in HASS addon is described to be 2, while in reality it is 8.

**To Reproduce**
Install EMHASS addon and configure it (leave `days_to_retrieve` default), create a new `sensor.house_consumption_no_var_loads`, let it collecting values for 3 days, attempt to run Perfect optim.

**Expected behavior**
Perfect optim should pass.

**Screenshots**
[image]
[image]

**Home Assistant installation type**
 - Home Assistant OS
 - Home Assistant Supervised
 - Home Assistant Core

**Your hardware**
- OS: HA OS, Windows, Linux, etc
- Architecture: amd64, armhf, armv7, aarch64

**EMHASS installation type**
 - Add-on, Docker Standalone, Legacy Python virtual environment

**Additional context**
```
root@5b918bf2-emhass:/usr/local/lib/python3.11/dist-packages/emhass# head -5 /usr/src/config_emhass.yaml 
# Configuration file for EMHASS

retrieve_hass_conf:
  freq: 30 # The time step to resample retrieved data from hass in minutes
  days_to_retrieve: 8 # We will retrieve data from now and up to days_to_retrieve days
  ```
  
  + additional log output manually added to https://github.com/davidusb-geek/emhass/blob/master/src/emhass/retrieve_hass.py#L116 - note the datetime which it attempts to load data for:
  
  ```
  2024-02-12 11:45:20,898 - web_server - INFO - Retrieve hass get data method initiated...
2024-02-12 11:45:20,901 - web_server - INFO - getting url='http://supervisor/core/api/history/period/2024-02-04T10:00:00+00:00?filter_entity_id=sensor.house_consumption_no_var_loads'
2024-02-12 11:45:20,930 - web_server - ERROR - The retrieved JSON is empty, check that correct day or variable names are passed
  ```

davidusb-geek, 2024-02-12:
You are right. It was always 2 and that' what is reflected in the documentation and it is what is logic for some forecast methods. But at some point we changed it for testing and it wrongly stayed at 8.
Fixed: https://github.com/davidusb-geek/emhass/commit/f7e98416ae84e1e7989e0bd2f6b364b579a75da5

tomasbedrich, 2024-02-13:
Thanks for the fix.

Perhaps there is some way to synchronize the values so that it doesn't happen again in future?

davidusb-geek, 2024-02-22:
I can't come up with an easy way to synchronize these values, these are two separate config file, one from the add-on and the other with the default values. Any suggestions are welcome. Closing for now.

closed 2024-02-22 by davidusb-geek