davidusb-geek/emhass #197: Make fails with „Temporary failure resolving deb.debian.org“

lythie-de, 2024-02-13:
**Describe the bug**
Running the makefile on a Raspberry pi results in failure mesages like these:

Err:1 http://deb.debian.org/debian buster/main armhf libkeyutils1 armhf 1.6-6
  Temporary failure resolving 'deb.debian.org'
Err:2 http://deb.debian.org/debian-security buster/updates/main armhf libkrb5support0 armhf 1.17-3+deb10u6
  Temporary failure resolving 'deb.debian.org'

**To Reproduce**
After a git pull I executed the make command:

pi@raspberrypi:~/emhass/emhass $ make -f deploy_docker.mk clean_deploy

**Expected behavior**
The make process should create the tar-file without errors/warnings.

**Screenshots**

pi@raspberrypi:~/emhass/emhass $ make -f deploy_docker.mk clean_deploy
docker build -t emhass/core:f7e9841 -f Dockerfile .
Sending build context to Docker daemon  25.62MB
Step 1/24 : FROM python:3.11-slim-buster
 ---> e439b63e6a4b
Step 2/24 : WORKDIR /app
 ---> Using cache
 ---> 2f809f16719d
Step 3/24 : COPY requirements.txt requirements.txt
 ---> Using cache
 ---> b4dd15a978a8
Step 4/24 : COPY requirements_webserver.txt requirements_webserver.txt
 ---> Using cache
 ---> 031fe6ab2926
Step 5/24 : COPY setup.py setup.py
 ---> Using cache
 ---> 55dce8dda359
Step 6/24 : COPY README.md README.md
 ---> Using cache
 ---> f237e8428e33
Step 7/24 : RUN apt-get update     && apt-get install -y --no-install-recommends         gcc         coinor-cbc         coinor-libcbc-dev         libhdf5-dev         libhdf5-serial-dev         netcdf-bin         libnetcdf-dev     && ln -s /usr/include/hdf5/serial /usr/include/hdf5/include     && export HDF5_DIR=/usr/include/hdf5     && pip3 install --no-cache-dir --break-system-packages -r requirements_webserver.txt     && apt-get purge -y --auto-remove         gcc         libhdf5-dev         libhdf5-serial-dev         netcdf-bin         libnetcdf-dev     && rm -rf /var/lib/apt/lists/*
 ---> Running in 5d556eb5494e
Get:1 http://deb.debian.org/debian buster InRelease [122 kB]
Get:2 http://deb.debian.org/debian-security buster/updates InRelease [34.8 kB]
Get:3 http://deb.debian.org/debian buster-updates InRelease [56.6 kB]
Get:4 http://deb.debian.org/debian buster/main armhf Packages [7706 kB]
Get:5 http://deb.debian.org/debian-security buster/updates/main armhf Packages [577 kB]
Get:6 http://deb.debian.org/debian buster-updates/main armhf Packages [8784 B]
Fetched 8505 kB in 9s (897 kB/s)
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  binutils binutils-arm-linux-gnueabihf binutils-common coinor-libcbc3
  coinor-libcgl-dev coinor-libcgl1 coinor-libclp-dev coinor-libclp1
  coinor-libcoinutils-dev coinor-libcoinutils3v5 coinor-libosi-dev
  coinor-libosi1v5 cpp cpp-8 gcc-8 hdf5-helpers libaec-dev libaec0 libasan5
  libatomic1 libbinutils libblas-dev libblas3 libbz2-dev libc-dev-bin
  libc6-dev libcc1-0 libcurl3-gnutls libcurl4-gnutls-dev libgcc-8-dev
  libgfortran5 libgomp1 libgssapi-krb5-2 libhdf5-103 libhdf5-cpp-103 libisl19
  libjpeg-dev libjpeg62-turbo libjpeg62-turbo-dev libk5crypto3 libkeyutils1
  libkrb5-3 libkrb5support0 liblapack-dev liblapack3 libldap-2.4-2
  libldap-common libmpc3 libmpfr6 libnetcdf13 libnghttp2-14 libpsl5 librtmp1
  libsasl2-2 libsasl2-modules-db libssh2-1 libsz2 libubsan1 linux-libc-dev
  zlib1g-dev
Suggested packages:
  binutils-doc cpp-doc gcc-8-locales gcc-multilib make manpages-dev autoconf
  automake libtool flex bison gdb gcc-doc gcc-8-doc libgcc1-dbg libgomp1-dbg
  libitm1-dbg libatomic1-dbg libasan5-dbg liblsan0-dbg libtsan0-dbg
  libubsan1-dbg libmpx2-dbg libquadmath0-dbg liblapack-doc glibc-doc
  libcurl4-doc libgnutls28-dev libidn11-dev libkrb5-dev libldap2-dev
[... 149 lines cut ...]
^Cmake: *** [deploy_docker.mk:12: deploy] Unterbrechung

pi@raspberrypi:~/emhass/emhass $ docker exec -it homeassistant bash
raspberrypi:/config# ping deb.debian.org
PING deb.debian.org (2a04:4e42:8d::644): 56 data bytes
64 bytes from 2a04:4e42:8d::644: seq=0 ttl=60 time=25.564 ms
64 bytes from 2a04:4e42:8d::644: seq=1 ttl=60 time=23.764 ms
64 bytes from 2a04:4e42:8d::644: seq=2 ttl=60 time=23.827 ms
64 bytes from 2a04:4e42:8d::644: seq=3 ttl=60 time=24.496 ms
^C
--- deb.debian.org ping statistics ---
4 packets transmitted, 4 packets received, 0% packet loss
round-trip min/avg/max = 23.764/24.412/25.564 ms
raspberrypi:/config# exit
exit
pi@raspberrypi:~/emhass/emhass $ make -f deploy_docker.mk clean_deploy > make.log
^Cmake: *** [deploy_docker.mk:12: deploy] Unterbrechung

pi@raspberrypi:~/emhass/emhass $ make -f deploy_docker.mk clean_deploy > make.log &
[1] 9440
pi@raspberrypi:~/emhass/emhass $ 
pi@raspberrypi:~/emhass/emhass $ docker exec -it homeassistant bash
raspberrypi:/config# ping deb.debian.org
ping: bad address 'deb.debian.org'
raspberrypi:/config#  `

davidusb-geek, 2024-02-14:
Hi, currently armhf architecture are not supported. They were until not too long time ago when we switched to fully support Python 3.11.

So this has been hold since then and I have not found a solution to this. So you can actually help finding a solution.

You can start trying to modify the Dockefile to use a different Python version.
Change this:
```
python:3.10-slim-buster
```
Or even:
```
python:3.9-slim-buster
```

davidusb-geek, 2024-03-11:
All these issues should be solved now. Wait for the v0.8.3 to be available.

closed 2024-03-11 by davidusb-geek