davidusb-geek/emhass #198: MPC: No proper scheduling when using def_end_timestep

sti0, 2024-02-14:
**Describe the bug**
When scheduling my dishwasher I reduce the `def_total_hours` by runtime. I reduce it every 0.5 hours (-1 timestep). I set `def_end_timestep` = `def_total_hours * 2`. I expect that the result schedules the deferrable load one by one for 6 timesteps (3h to run). Each 30m the hours will be reduced by 0.5 and the timesteps will be reduced by 1. Nevertheless there are gaps in some of the calculations. The gaps do not happen when `def_end_timestep` is not set OR when its set to a higher value e.g. 12.

Long story short: I like to make EMHASS know that the dishwasher is running and that the energy is needed, therefore I need to use `def_end_timestep` to force it to this given timeframe otherwise it will schedule the "hours to run" somewhen later which makes no sense in this case.

**To Reproduce**

1. iteration def_total_hours = 3, def_end_timestep = 6 
```
curl -i -H "Content-Type: application/json" -X POST -d '{
        "load_cost_forecast": [0.2554, 0.2536, 0.2536, 0.2484, 0.2484, 0.2459, 0.2459, 0.2452, 0.2452, 0.2442, 0.2442, 0.2435, 0.2435, 0.2462, 0.2462, 0.2511, 0.2511, 0.2584, 0.2584, 0.2695, 0.2695, 0.2708, 0.2708, 0.2662, 0.2662, 0.2607, 0.2607, 0.2557, 0.2557, 0.2517, 0.2517, 0.2512, 0.2512, 0.2556, 0.2556, 0.2618, 0.2618, 0.2665, 0.2665, 0.276, 0.276, 0.2759, 0.2759, 0.2624, 0.2624, 0.2563, 0.2563, 0.2474],
        "prediction_horizon": 36,
        "pv_power_forecast": [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 408, 798, 1200, 1608, 2330, 4349, 5582, 5938, 5293, 4092, 3819, 3779, 3645, 3471, 3129, 3126, 2708, 1445, 758, 274, 0, 0, 0, 0, 0, 0, 0],
        "alpha": 0.75,
        "beta": 0.25,
        "num_def_loads": 3,
        "def_total_hours": [
            0,
            3,
            0
          ],
        "P_deferrable_nom":  [
            2750,
            2000,
            2200
          ],
        "treat_def_as_semi_cont": [true, true, true],
        "set_def_constant": [true, true, true],
        "def_start_timestep": [
            0,
            0,
            0
          ],
        "def_end_timestep": [
            0,
            6,
            0
          ]
      }' http://localhost:5000/action/naive-mpc-optim
```

now reduce the `def_total_hours` and `def_end_timestep` each 30 minutes:
2. iteration: 2.5 / 5
3. iteration: 2 / 4
4. iteration: 1.5 / 3
5. iteration: 1 / 2
6. iteration: 0.5 / 1

**Expected behavior**
In each iteration the loads should be scheduled without any gap!

**Screenshots**

## 1. iteration (`def_total_hours`=3, `def_end_timestep`=6)
[image]
=> Not ok, gap between timestep 1 and 3, last timestep is doubled up (4000 W instead of 2000 W)

## 2. iteration (`def_total_hours`=2.5, `def_end_timestep`=5)
[image]
=> Not ok, gap and double watts

## 3. iteration (`def_total_hours`=2, `def_end_timestep`=4)
[image]
=> Ok

## 4. iteration (`def_total_hours`=1.5, `def_end_timestep`=3)
[image]
=> Ok

## 5. iteration (`def_total_hours`=1, `def_end_timestep`=2)
[image]
[image]
=> Not ok, second timestep is scheduled to the very last element (+18h)

## 5. iteration (`def_total_hours`=0.5, `def_end_timestep`=1)
[image]
=> Ok

**Home Assistant installation type**
 - Home Assistant OS

**Your hardware**
- OS: HA OS
- Architecture: amd64

**EMHASS installation type**
 - Add-on

**Additional context**

When not using `def_end_timestep` the calculation is done like expected in "one go":
3 hours:
[image]
2.5 hours:
[image]
1 hour:
[image]

When using a greater `def_end_timestep` it works like expected, too:
2.5 hours, def_end_timestep = 12:
[image]

1 hour, def_end_timestep = 12:
[image]

davidusb-geek, 2024-02-14:
@michaelpiron do you have any ideas on this?

michaelpiron, 2024-02-15:
@sti0 for each of the iterations, could you tell what was the status of the optimization? Was it “optimal” in each iteration? 
can you share the log (DEBUG level) of a bad iteration?

When did you run each iteration? Did you run it at exact at the start of a new half hour? (Xh00 and Xh30)

Last info I’d need: which load cost profile did you feed to the optimizer? Flat tariff?

thx

sti0, 2024-02-15:
@michaelpiron will try to reproduce it with the needed information tonight

michaelpiron, 2024-02-15:
In fact, the start and end timestep feature was not built to accomodate your use case. Your case would need a logic that tells emhass that the deferrable load has started its program, and hence the remaining hours need to be planned at the beginningof the optimization window (without having to specify the end timestep).

Let’s for now try to debug further and see how you can use start-end timesteps to accomodate your use case as good as possible. In the longer run however, we’ll need a new feature.

sti0, 2024-02-15:
But in fact the time endstep should not produce wrong results because only because it's forced to produce in a given timeframe. It should produce the same result if the endtimestep is 6 or 12 as long as the time condition was met.

But I agree a feature to tell EMHASS that the load is running and can't be stopped during the run would be a better solution.

sti0, 2024-02-15:
@michaelpiron please find the wanted information next:

> which load cost profile did you feed to the optimizer? Flat tariff?
I'm using `self-consumption`. With dynamic tariff from Tibber feed into when calling MPC.

`method_ts_round` is set to `last`.

# 1. Iteration - NOK

MPC Call @ 22:39 with `def_total_hours` = 3 and `def_end_timestep` = 6:
[image]
Debug log:
```
2024-02-15 22:39:05,195 - web_server - INFO - Setting up needed data
2024-02-15 22:39:05,242 - web_server - INFO - Retrieve hass get data method initiated...
2024-02-15 22:39:07,078 - web_server - INFO - Retrieving weather forecast data using method = list
2024-02-15 22:39:07,080 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2024-02-15 22:39:07,080 - web_server - INFO - Retrieve hass get data method initiated...
2024-02-15 22:39:09,444 - web_server - INFO -  >> Performing naive MPC optimization...
2024-02-15 22:39:09,444 - web_server - INFO - Performing naive MPC optimization
2024-02-15 22:39:09,449 - web_server - INFO - Perform an iteration of a naive MPC controller
2024-02-15 22:39:09,458 - web_server - DEBUG - Deferrable load 0: Proposed optimization window: 0 --> 0
2024-02-15 22:39:09,458 - web_server - DEBUG - Deferrable load 0: Validated optimization window: 0 --> 0
2024-02-15 22:39:09,461 - web_server - DEBUG - Deferrable load 1: Proposed optimization window: 0 --> 6
2024-02-15 22:39:09,461 - web_server - DEBUG - Deferrable load 1: Validated optimization window: 0 --> 6
2024-02-15 22:39:09,465 - web_server - DEBUG - Deferrable load 2: Proposed optimization window: 0 --> 0
2024-02-15 22:39:09,465 - web_server - DEBUG - Deferrable load 2: Validated optimization window: 0 --> 0
2024-02-15 22:39:09,532 - web_server - INFO - Status: Optimal
2024-02-15 22:39:09,532 - web_server - INFO - Total value of the Cost function = -2165.71
```

[image]

# 2. Iteration - NOK

MPC Call @ 22:45 with `def_total_hours` = 2.5 and `def_end_timestep` = 5:
[image]
```
2024-02-15 22:45:14,503 - web_server - INFO - Setting up needed data
2024-02-15 22:45:14,511 - web_server - INFO - Retrieve hass get data method initiated...
2024-02-15 22:45:16,249 - web_server - INFO - Retrieving weather forecast data using method = list
2024-02-15 22:45:16,250 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2024-02-15 22:45:16,251 - web_server - INFO - Retrieve hass get data method initiated...
2024-02-15 22:45:18,440 - web_server - INFO -  >> Performing naive MPC optimization...
2024-02-15 22:45:18,440 - web_server - INFO - Performing naive MPC optimization
2024-02-15 22:45:18,446 - web_server - INFO - Perform an iteration of a naive MPC controller
2024-02-15 22:45:18,453 - web_server - DEBUG - Deferrable load 0: Proposed optimization window: 0 --> 0
2024-02-15 22:45:18,453 - web_server - DEBUG - Deferrable load 0: Validated optimization window: 0 --> 0
2024-02-15 22:45:18,456 - web_server - DEBUG - Deferrable load 1: Proposed optimization window: 0 --> 5
2024-02-15 22:45:18,456 - web_server - DEBUG - Deferrable load 1: Validated optimization window: 0 --> 5
2024-02-15 22:45:18,461 - web_server - DEBUG - Deferrable load 2: Proposed optimization window: 0 --> 0
2024-02-15 22:45:18,461 - web_server - DEBUG - Deferrable load 2: Validated optimization window: 0 --> 0
2024-02-15 22:45:18,534 - web_server - INFO - Status: Optimal
2024-02-15 22:45:18,534 - web_server - INFO - Total value of the Cost function = -1917.27
2024-02-15 22:45:27,078 - web_server - INFO - EMHASS server online, serving index.html...
```
[image]

# 3. Iteration - OK

MPC Call @ 22:47 with `def_total_hours` = 2 and `def_end_timestep` = 4:
[image]
```
2024-02-15 22:47:07,647 - web_server - INFO - Setting up needed data
2024-02-15 22:47:07,654 - web_server - INFO - Retrieve hass get data method initiated...
2024-02-15 22:47:08,765 - web_server - INFO - Retrieving weather forecast data using method = list
2024-02-15 22:47:08,766 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2024-02-15 22:47:08,767 - web_server - INFO - Retrieve hass get data method initiated...
2024-02-15 22:47:11,600 - web_server - INFO -  >> Performing naive MPC optimization...
2024-02-15 22:47:11,600 - web_server - INFO - Performing naive MPC optimization
[... 102 lines cut ...]
MPC Call @ 23:05 with `def_total_hours` = 1 and `def_end_timestep` = 2:
[image]
=> WRONG RESULT
```
2024-02-15 23:05:13,647 - web_server - INFO - Setting up needed data
2024-02-15 23:05:13,654 - web_server - INFO - Retrieve hass get data method initiated...
2024-02-15 23:05:14,832 - web_server - INFO - Retrieving weather forecast data using method = list
2024-02-15 23:05:14,834 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2024-02-15 23:05:14,834 - web_server - INFO - Retrieve hass get data method initiated...
2024-02-15 23:05:17,173 - web_server - INFO -  >> Performing naive MPC optimization...
2024-02-15 23:05:17,173 - web_server - INFO - Performing naive MPC optimization
2024-02-15 23:05:17,178 - web_server - INFO - Perform an iteration of a naive MPC controller
2024-02-15 23:05:17,183 - web_server - DEBUG - Deferrable load 0: Proposed optimization window: 0 --> 0
2024-02-15 23:05:17,183 - web_server - DEBUG - Deferrable load 0: Validated optimization window: 0 --> 0
2024-02-15 23:05:17,185 - web_server - DEBUG - Deferrable load 1: Proposed optimization window: 0 --> 2
[... cut ...]
2024-02-15 23:05:17,188 - web_server - DEBUG - Deferrable load 2: Proposed optimization window: 0 --> 0
2024-02-15 23:05:17,188 - web_server - DEBUG - Deferrable load 2: Validated optimization window: 0 --> 0
2024-02-15 23:05:17,241 - web_server - INFO - Status: Optimal
2024-02-15 23:05:17,241 - web_server - INFO - Total value of the Cost function = -1192.86
2024-02-15 23:05:21,132 - web_server - INFO - EMHASS server online, serving index.html...
```

MPC Call @ 23:06 with `def_total_hours` = 0.5 and `def_end_timestep` = 1:
[image]

sti0, 2024-02-16:
Another strange behavior which I noticed right now. While my deferrable load 2 (washing maschine) is running, deferrable load 0 got scheduled but its "deactivated" (running hours are always 0). And the load seems to be a very small value (not typical usage).

In the following screenshot p_deferrable load 2 is supposed to run for 1.5h with end timestep set to 3 (2200W each timestep).

[image]
[image]

After running it again a few minutes later it comes to deferrable load 1 (which is also defined with def_total_hours = 0 atm).
[image]

```
2024-02-16 12:24:48,198 - web_server - INFO - Setting up needed data
2024-02-16 12:24:48,232 - web_server - INFO - Retrieve hass get data method initiated...
2024-02-16 12:24:49,729 - web_server - INFO - Retrieving weather forecast data using method = list
2024-02-16 12:24:49,731 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2024-02-16 12:24:49,731 - web_server - INFO - Retrieve hass get data method initiated...
2024-02-16 12:24:52,597 - web_server - INFO -  >> Performing naive MPC optimization...
2024-02-16 12:24:52,597 - web_server - INFO - Performing naive MPC optimization
2024-02-16 12:24:52,601 - web_server - INFO - Perform an iteration of a naive MPC controller
2024-02-16 12:24:52,607 - web_server - DEBUG - Deferrable load 0: Proposed optimization window: 0 --> 0
2024-02-16 12:24:52,607 - web_server - DEBUG - Deferrable load 0: Validated optimization window: 0 --> 0
2024-02-16 12:24:52,608 - web_server - DEBUG - Deferrable load 1: Proposed optimization window: 0 --> 0
2024-02-16 12:24:52,609 - web_server - DEBUG - Deferrable load 1: Validated optimization window: 0 --> 0
2024-02-16 12:24:52,610 - web_server - DEBUG - Deferrable load 2: Proposed optimization window: 0 --> 3
2024-02-16 12:24:52,610 - web_server - DEBUG - Deferrable load 2: Validated optimization window: 0 --> 3
2024-02-16 12:24:52,649 - web_server - INFO - Status: Optimal
2024-02-16 12:24:52,649 - web_server - INFO - Total value of the Cost function = -1200.78
2024-02-16 12:24:57,922 - web_server - INFO - EMHASS server online, serving index.html...
```

michaelpiron, 2024-02-17:
Very strange...
When enforcing an end timestep, the only constraint that is added, is the following:
[image]

In your case, this means:
Deferrable load 2 needs to operate before timestep 3, so its consumed energy between step 3 & end of prediction horizon needs to equal zero.

I don't immediately grasp how this would affect the operating scheme of deferrable load 0.
I find it also strange that the optimizer delivers a so-called 'optimal' result, whereas not all constraints were respected (in this case: load 0 was not allowed to run, but it got scheduled anyway.)

I'm not sure if the issues you encounter are specifically related to the start_end timestep feature.
Will need to dive deeper. I'll do it as soon as I can.
@davidusb-geek would you have any idea?

davidusb-geek, 2024-02-22:
No idea. This will need testing to debug under these same conditions. I haven't found the time to look at this.

MikaelHoogen, 2024-02-25:
I have the same problem. I do schedule my heatpump to run x hours with a 12 hour interval. I have a 30m timestep.
For each timestep I reduce the def_end_timestep and reduce x with the remaining hours to run.

My theory is that it has something to do with the nominal power and when in the window the optimisation runs. When the problem occurs it distributes the total power in fewer timesteps. It seem to sum up to the total.

The problem seem to be intermittent. Someone with script skillz may be able to reproduce this.

---
Let Deferrable load = x and def_end_timestep = y

If y = 2x I get in trouble. But not always.
Deferrable load 0 = 2000 w
Deferrable load 1 = 1200 w

The log say Optimal in all cases
2024-02-25 10:29:15,384 - web_server - INFO -  >> Performing naive MPC optimization...
2024-02-25 10:29:15,385 - web_server - INFO - Performing naive MPC optimization
2024-02-25 10:29:15,390 - web_server - INFO - Perform an iteration of a naive MPC controller
2024-02-25 10:29:15,393 - web_server - DEBUG - Deferrable load 0: Proposed optimization window: 0 --> y
2024-02-25 10:29:15,393 - web_server - DEBUG - Deferrable load 0: Validated optimization window: 0 --> y
2024-02-25 10:42:01,485 - web_server - DEBUG - Deferrable load 1: Proposed optimization window: 0 --> 1
2024-02-25 10:42:01,485 - web_server - DEBUG - Deferrable load 1: Validated optimization window: 0 --> 1
2024-02-25 10:29:15,452 - web_server - INFO - Status: Optimal

Example:
x = 8 and y = 16
[image]

michaelpiron, 2024-02-25:
So if y = 2x + 1, the optimization runs fine?
I’m trying to reconstruct the issue

MikaelHoogen, 2024-02-25:
> So if y = 2x + 1, the optimization runs fine? I’m trying to reconstruct the issue

Yes I think so. I notised this problem this morning so I'm not really sure. But that is my workaround for now.

MikaelHoogen, 2024-03-10:
Any progress on this?

davidusb-geek, 2024-03-12:
I don't if @michaelpiron was able to reproduce this?

MikaelHoogen, 2024-03-18:
I think on to something. The problem seems to be related to
set_deferrable_load_single_constant: true

I did a quick test now. Will test more tonight.

When I set it to false it behaves as it should it seems.

MikaelHoogen, 2024-03-19:
set_deferrable_load_single_constant: true ---> problem
set_deferrable_load_single_constant: false ---> no problem

To my understanding the villain is set_deferrable_load_single_constant.

@michaelpiron this maybe can be useful if you want to reproduce the issue?

michaelpiron, 2024-04-14:
I don't fully understand the functionality of the list_set_deferrable_load_single_constant ('Set each deferrable load as a constant fixed value variable with just one startup for each optimization task (ex. each 24h)').
I've always worked with "false" as I don't need that functionality for my use cases.

So I might not be the best person to deal with this issue.

As written by @davidusb-geek in the HA forum, this is not really a bug, but just the way this type of optimization works. (https://community.home-assistant.io/t/emhass-an-energy-management-for-home-assistant/338126/2098)
Should the optimization code be changed? perhaps, but at this point I'm not able to say how it should change.

davidusb-geek, 2024-04-15:
Maybe a solution could be to provide a helper to the optimization to indicate that a deferrable load is currently in operation. But this means that the user has to provide this information. Need to figure out the simplest way to achieve this.

closed 2024-05-01 by davidusb-geek

sti0, 2024-05-01:
Is there any new feature in an upcoming release which fixes this issue?