davidusb-geek/emhass #199: web_server - ERROR - Exception on /action/publish-data

BDVGitHub, 2024-02-14:
**Describe the bug**
When running curl -i -H ’Content-Type:application/json’ -X POST -d ’{}’ http://localhost:5000/action/dayahead-optim I get an internal server error 500

In Docker I have edited config_emhass.yaml and secrets_emhass.yaml

In Home Assistant I put in configuration.yaml

shell_command:
  dayahead_optim: "curl -i -H \"Content-Type:application/json\" -X POST -d '{}' http://localhost:5000/action/dayahead-optim"
  publish_data: "curl -i -H \"Content-Type:application/json\" -X POST -d '{}' http://localhost:5000/action/publish-data"

and in automations.yaml

- alias: EMHASS day-ahead optimization
  trigger:
    platform: time
    at: '05:30:00'
  action:
  - service: shell_command.dayahead_optim
- alias: EMHASS publish data
  trigger:
  - minutes: /5
    platform: time_pattern
  action:
  - service: shell_command.publish_data

**To Reproduce**
root@MiniPCServer:/home/brecht# docker exec -it DockerEMHASS bash
root@6185e47f85ef:/app# curl -i -H 'Content-Type:application/json' -X POST -d '{}' http://localhost:5000/action/dayahead-optim
HTTP/1.1 500 INTERNAL SERVER ERROR
Content-Length: 265
Content-Type: text/html; charset=utf-8
Date: Wed, 14 Feb 2024 20:58:38 GMT
Server: waitress

<!doctype html>
<html lang=en>
<title>500 Internal Server Error</title>
<h1>Internal Server Error</h1>
<p>The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.</p>

**Expected behavior**
What do I need to do to run curl -i -H 'Content-Type:application/json' -X POST -d '{}' http://localhost:5000/action/dayahead-optim without error?

**Home Assistant installation type**
  - Home Assistant Supervised
 - Home Assistant Core

**Your hardware**
- OS: Linux (Debian) 
- Architecture: x86_64

**EMHASS installation type**
 -  Docker Standalone
 
**Additional context**
Docker info
lient: Docker Engine - Community
 Version:    25.0.3
 Context:    default
 Debug Mode: false
 Plugins:
  buildx: Docker Buildx (Docker Inc.)
    Version:  v0.12.1
    Path:     /usr/libexec/docker/cli-plugins/docker-buildx
  compose: Docker Compose (Docker Inc.)
    Version:  v2.24.5
    Path:     /usr/libexec/docker/cli-plugins/docker-compose

Server:
[... 116 lines cut ...]
     return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/app/src/emhass/web_server.py", line 53, in action_call
     _ = publish_data(input_data_dict, app.logger)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/usr/local/lib/python3.11/site-packages/emhass-0.7.7-py3.11.egg/emhass/command_line.py", line 474, in publish_data
     input_data_dict['rh'].post_data(opt_res_latest['P_PV'], idx_closest,
   File "/usr/local/lib/python3.11/site-packages/emhass-0.7.7-py3.11.egg/emhass/retrieve_hass.py", line 332, in post_data
     response = post(url, headers=headers, data=json.dumps(data))
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/usr/local/lib/python3.11/site-packages/requests/api.py", line 115, in post
     return request("post", url, data=data, json=json, **kwargs)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/usr/local/lib/python3.11/site-packages/requests/api.py", line 59, in request
     return session.request(method=method, url=url, **kwargs)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/usr/local/lib/python3.11/site-packages/requests/sessions.py", line 589, in request
     resp = self.send(prep, **send_kwargs)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/usr/local/lib/python3.11/site-packages/requests/sessions.py", line 703, in send
     r = adapter.send(request, **kwargs)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/usr/local/lib/python3.11/site-packages/requests/adapters.py", line 519, in send
     raise ConnectionError(e, request=request)
 requests.exceptions.ConnectionError: HTTPConnectionPool(host='localhost', port=80): Max retries exceeded with url: /api/states/sensor.p_pv_forecast (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f16a22ff890>: Failed to establish a new connection: [Errno 111] Connection refused'))

davidusb-geek, 2024-02-14:
Just test that curl command without going inside the docker container. Give it a try from a simple new terminal. Your HA url looks weird though. Is it just localhost? No 8123 port?

BDVGitHub, 2024-02-15:
Hey, thanks for the advice, changed in secrets_emhass.yaml hass_url: http://localhost to hass_url: http://172.17.0.1:8123/ and is now working fine

closed 2024-02-15 by BDVGitHub