davidusb-geek/emhass #203: Documentation peak hour periods

BDVGitHub, 2024-02-19:
**Describe the bug**

In Belgium we have peak hour:
Monday - Friday 07:00 - 22:00

non peak hour
Monday- Fraday 22:00 - 07:00
In the weekend its always non peak hour

how do I need to implement this in the data_load_cost_forecast file?
‘/data/data_load_cost_forecast.csv’

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Home Assistant installation type**
 - Home Assistant Supervised

**Your hardware**
- OS: Linux
- Architecture: amd64

**EMHASS installation type**
 - Docker Standalone

davidusb-geek, 2024-02-22:
The best for this would be to define a template sensor with a list of cost values to be passed to emhass and then you can manage to pass a different list depending on the day of the week.

BDVGitHub, 2024-02-26:
Thx but where can I find how te structure for this template senor needs to look like, what kind of arguments do I need to give to emhass to pass the cost values? I've searched in the manual and in the fora but didn't found it yet

davidusb-geek, 2024-03-03:
There are quite some examples here: https://emhass.readthedocs.io/en/latest/forecasts.html#passing-your-own-forecast-data

But I can help you a bit here.

You need to define your own template sensor and build a list.
In Home Assistant you can define template sensors in the sensors.yaml file using the tamplate platform `platform: template`.

We need to build a list of values. A list of values should look like this: `[1, 2, 3, ...]`

So in your case for Belgium we can try this. I'm supposing here that you are using a time step of 60min (so if you are using 30 you need to adapt this) in EMHASS add-on and that the non-peak cost is 0.2 and the peak cost is 0.3 €/kWh (adapt these values to the real values).

Here is a proposed template:

```
  - platform: template
    sensors:
      load_cost_forecast:
        value_template: >-
          {% if now().weekday() == 5 or now().weekday() == 6 %}
            {%- set load_cost = [0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.2] %}
          {%- else -%}
            {%- set load_cost = [0.2, 0.2, 0.2, 0.2, 0.2, 0.2, 0.3, 0.3, 0.3, 0.3, 0.3, 0.3, 0.3, 0.3, 0.3, 0.3, 0.3, 0.3, 0.3, 0.3, 0.3, 0.3, 0.2, 0.2] %}
          {%- endif %}
          {%- set values_all = namespace(all=[]) %}
          {% for i in range(load_cost | length) %}
            {%- set v = (load_cost[i] | float ) %}
            {%- set values_all.all = values_all.all + [ v ] %}
          {%- endfor %} {{ (values_all.all)[now().hour:24] + (values_all.all)[0:now().hour] }}
```

davidusb-geek, 2024-03-03:
The final step is to define the shell command that will launch the optimization and pass it the list of values from the template sensor that we built before:

```
shell_command:
  dayahead_optim: "curl -i -H \"Content-Type: application/json\" -X POST -d '{\"load_cost_forecast\":{{states('sensor.load_cost_forecast_list')}}}' http://localhost:5000/action/dayahead-optim"
```

BDVGitHub, 2024-03-07:
Thx for the step by step guidance 
I'm using docker and using 30 minutes time stamp so I've put 48 cost values instead of 24 in the value list.
Currently I'm testing it but I don't foresee any problems

BDVGitHub, 2024-03-08:
In the config_emhass.yaml do I need to adjust "load_cost_forecast_method" 
now next to load_cost_forecast_method for the moment stands "hp_hc_periods" do I need to change this to "list" or not?

I'm using the docker version

davidusb-geek, 2024-03-08:
No no need to set those. All the passed data as lists at runtime overrides the settings from the configuration file. Anyway you can if your passed data is being taken into account by launching the optimization and then going to the webui and check the passed data there, in the graphs and in the table.

davidusb-geek, 2024-03-11:
Marking as solved feel free to reopen if needed

closed 2024-03-11 by davidusb-geek

BDVGitHub, 2024-03-11:
Thx is working now, only needed to add `sensor: !include sensors.yaml` in the configuration.yaml file
was the first time using sensors