davidusb-geek/emhass #204: IndexError in DEBUG logging of Proposed optimization window

luuuis, 2024-02-22:
**Describe the bug**
Upon upgrade to 0.6.6 EMHASS stopped working in Home Assistant due to the following error.

```
2024-02-22 15:37:00,247 - web_server - ERROR - Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1463, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 872, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 870, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 855, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/web_server.py", line 66, in action_call
    opt_res = dayahead_forecast_optim(input_data_dict, app.logger)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/command_line.py", line 234, in dayahead_forecast_optim
    opt_res_dayahead = input_data_dict['opt'].perform_dayahead_forecast_optim(
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/optimization.py", line 572, in perform_dayahead_forecast_optim
    self.opt_res = self.perform_optimization(df_input_data, P_PV.values.ravel(),
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/optimization.py", line 288, in perform_optimization
    self.logger.debug("Deferrable load {}: Proposed optimization window: {} --> {}".format(k, def_start_timestep[k], def_end_timestep[k]))
                                                                                              ~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
```

**To Reproduce**
1. Do not configure `def_start_timestep` or `def_end_timestep` in YAML,
2. Send a dayahead-optim request without providing either `def_start_timestep` or `def_end_timestep` in the payload.

**Expected behavior**
Configurations that were previously working should continue to work, especially for optional parameters. Alternatively, if these parameters are required then they should be validated and an appropriate error message issued.

**EMHASS installation type**
 - Add-on

**Additional context**
Add any other context about the problem here.

davidusb-geek, 2024-02-22:
> Do not configure def_start_timestep or def_end_timestep in YAML

Just configure these. They just should have the same length as the number of deferrable loads.

This was an unexpected breaking change, so that's why no warning was issued. But the solution is easy and the error messages were improved after this.

Closing.

closed 2024-02-22 by davidusb-geek

luuuis, 2024-02-23:
Yes, I configured locally once I figured out what was the cause. Just looking to avoid this for others who upgrade, glad it's fixed in later versions. 👍