davidusb-geek/emhass #218: Numpy you should not try to import numpyu from its source directory

Denna93, 2024-03-06:
**Describe the bug**
Getting error message after I try to run newly installed EMHASS on Hassio

**To Reproduce**
Just dowloading EMHASS from addon store and will everytime give me a error message when trying to start. 
**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/local/lib/python3.11/dist-packages/emhass/web_server.py", line 12, in <module>
    import pandas as pd
  File "/usr/local/lib/python3.11/dist-packages/pandas/__init__.py", line 16, in <module>
    raise ImportError(
ImportError: Unable to import required dependencies:
numpy: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.
        
        
**Home Assistant installation type**
 - Home Assistant OS
 - Home Assistant Supervised
 - Home Assistant Core

**Your hardware**
RPI 4

**EMHASS installation type**
Addon
**Additional context**
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/local/lib/python3.11/dist-packages/emhass/web_server.py", line 12, in <module>
    import pandas as pd
  File "/usr/local/lib/python3.11/dist-packages/pandas/__init__.py", line 16, in <module>
    raise ImportError(
ImportError: Unable to import required dependencies:
numpy: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.

davidusb-geek, 2024-03-08:
These problems on arm archs should be solved soon in the coming release.

davidusb-geek, 2024-03-11:
All these issues should be solved now. Wait for the v0.8.3 to be available.

closed 2024-03-11 by davidusb-geek