davidusb-geek/emhass #224: DST when running day ahead optim

choumarin, 2024-03-09:
**Describe the bug**
Running day ahead optimization crashes because of DST change in the next 24h

**To Reproduce**
Run day ahead optim when DST change is in the next 24h

**Expected behavior**
Expected day ahead optim to run, and ignore the hour that is offseted

**Home Assistant installation type**
 - Home Assistant OS

**Your hardware**
- OS: HA OS on VM (Synology)

**EMHASS installation type**
 - Add-on

**Additional context**
***Logs***
```
2024-03-09 15:43:57,096 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2024-03-09 15:43:57,097 - web_server - ERROR - Exception on /action/dayahead-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1463, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 872, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 870, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 855, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/web_server.py", line 103, in action_call
    input_data_dict = set_input_data_dict(config_path, str(data_path), costfun,
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/command_line.py", line 91, in set_input_data_dict
    df_weather = fcst.get_weather_forecast(method=optim_conf['weather_forecast_method'])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/forecast.py", line 187, in get_weather_forecast
    freq=freq_scrap).round(freq_scrap)
                     ^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/indexes/extension.py", line 98, in method
    result = attr(self._data, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/arrays/datetimelike.py", line 2026, in round
    return self._round(freq, RoundTo.NEAREST_HALF_EVEN, ambiguous, nonexistent)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/arrays/datetimelike.py", line 2002, in _round
    return result.tz_localize(
           ^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/arrays/_mixins.py", line 86, in method
    return meth(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/arrays/datetimes.py", line 1040, in tz_localize
    new_dates = tzconversion.tz_localize_to_utc(
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "pandas/_libs/tslibs/tzconversion.pyx", line 417, in pandas._libs.tslibs.tzconversion.tz_localize_to_utc
pytz.exceptions.NonExistentTimeError: 2024-03-10 02:00:00
```

davidusb-geek, 2024-03-10:
This is a reccurent issue every 6 months with DST.
We have tried some solution in the past without much success.
I will try a new solution that I found jsut now: https://pandas.pydata.org/docs/reference/api/pandas.Series.dt.round.html
It will be available in the upcoming release.

davidusb-geek, 2024-03-11:
Could you test this now, the solution has been implemented

davidusb-geek, 2024-03-12:
> Great! If this gets into a new release, I'll defo upgrade before our cut-over to CEST on march 30:th.

Yes exactly, we will try this on a new release before the 30th

davidusb-geek, 2024-04-08:
This specific problem when using the rounding function should be solved.
Feel free to reopen if needed

closed 2024-04-08 by davidusb-geek