davidusb-geek/emhass #225: Undefined variable when running as standalone in 0.8.2

philjohn, 2024-03-10:
**Describe the bug**

Log filled with:

```
NameError: name 'options_json' is not defined

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/usr/local/lib/python3.11/dist-packages/emhass/web_server.py", line 243, in <module>
    if options_json.exists():
       ^^^^^^^^^^^^

NameError: name 'options_json' is not defined
```

**To Reproduce**
Pull latest version (0.8.2) into your docker environment and launch

**Home Assistant installation type**
 - Home Assistant Core

**Your hardware**
- OS: HA OS, Windows, Linux, etc
- Architecture: amd64

**EMHASS installation type**
 - Docker Standalone,

davidusb-geek, 2024-03-10:
You need to pass the environment variable `USE_OPTIONS=True`

GeoDerp, 2024-03-10:
This is interesting. Standalone should not be asking for options file at all. I'll see if I can check this today.

davidusb-geek, 2024-03-10:
> This is interesting. Standalone should not be asking for options file at all. I'll see if I can check this today.

From the code if not using the `USE_OPTION` env then `options_json` is undefined.

GeoDerp, 2024-03-10:
oh I see `use_options` should be False by default yet @philjohn looks like its `True`?

GeoDerp, 2024-03-10:
I think this Is an error of mine from the last commit.

GeoDerp, 2024-03-10:
```python
if __name__ == "__main__":
    # Parsing arguments
    parser = argparse.ArgumentParser()
    parser.add_argument('--url', type=str, help='The URL to your Home Assistant instance, ex the external_url in your hass configuration')
    parser.add_argument('--key', type=str, help='Your access key. If using EMHASS in standalone this should be a Long-Lived Access Token')
    parser.add_argument('--addon', type=strtobool, default='False', help='Define if we are usinng EMHASS with the add-on or in standalone mode')
    parser.add_argument('--no_response', type=strtobool, default='False', help='This is set if json response errors occur')
    args = parser.parse_args()
    
    use_options = os.getenv('USE_OPTIONS', default=False)
    options = None

    #Obtain url and key from ENV or ARG (if any)
    hass_url = os.getenv("EMHASS_URL", default=args.url)
    key =  os.getenv("SUPERVISOR_TOKEN", default=args.key) 
    if hass_url != "http://supervisor/core/api":
        key =  os.getenv("EMHASS_KEY", key)  
    #If url or key is None, Set as empty string to reduce NoneType errors bellow
    if key is None: key = ""
    if hass_url is None: hass_url = ""
    
    # Define the paths
    if args.addon==1:
        OPTIONS_PATH = os.getenv('OPTIONS_PATH', default="/app/options.json")
        options_json = Path(OPTIONS_PATH)
        CONFIG_PATH = os.getenv("CONFIG_PATH", default="/app/config_emhass.yaml")
        DATA_PATH = os.getenv("DATA_PATH", default="/app/data/")
        # Read options info
        if options_json.exists():
            with options_json.open('r') as data:
                options = json.load(data)
        else:
            app.logger.error("options.json does not exists")
    else:
        if use_options:
            OPTIONS_PATH = os.getenv('OPTIONS_PATH', default="/app/options.json")
            options_json = Path(OPTIONS_PATH)
            # Read options info
            if options_json.exists():
                with options_json.open('r') as data:
                    options = json.load(data)
            else:
                app.logger.error("options.json does not exists")
        else:
            options = None       

    #if data path specified by options.json
    if options is not None:
        if options.get('data_path', None) != None and options.get('data_path', None) != "default":
            DATA_PATH = options.get('data_path', None);
```

This may work better

davidusb-geek, 2024-03-10:
Ok we could release a quick patch. I've just released the new version.

davidusb-geek, 2024-03-10:
That solution seems good. 
I'm going out, if tests are succesful put the patch on a PR and I will be able to patch the release tomorrow night.
Thanks @GeoDerp

davidusb-geek, 2024-03-11:
Solved with latest patch

closed 2024-03-11 by davidusb-geek