davidusb-geek/emhass #229: Cannot install the addon from repo

marst444, 2024-03-11:
Cannot install EMHASS add-on to home assistant from advised repo: https://github.com/davidusb-geek/emhass-add-on

giving the log: 
Logger: homeassistant.components.hassio
Källa: components/hassio/websocket_api.py:138
integration: Home Assistant Supervisor (dokumentation, ärenden)
Inträffade först: 22:23:52 (3 händelser)
Senast loggade: 22:28:55

Failed to to call /store/repositories -

davidusb-geek, 2024-03-11:
If you are using an ARM architecture (RPi's), then just wait a few more seconds for the images to build.

neil-ennis, 2024-03-12:
I try to add this repo:
https://github.com/davidusb-geek/emhass-add-on

I get this error:
https://github.com/davidusb-geek/emhass is not a valid add-on repository

Note the error message has a different URI from the repo.

It could just be me - I'm a bit of a gumby, but if I'm doing something wrong, please point me in the right direction.

GeoDerp, 2024-03-12:
Just tried with `https://github.com/davidusb-geek/emhass-add-on.git`

marst444, 2024-03-12:
Running on a "homeassistant green"
Waiting out did not help.
Tried the above mentioned https instead, no luck:

2024-03-12 12:38:06.999 ERROR (MainThread) [homeassistant.components.hassio.handler] /store/repositories return code 400
2024-03-12 12:38:06.999 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /store/repositories - 

Have updated to latest version and restarted. 
No success.

GeoDerp, 2024-03-12:
> Running on a "homeassistant green" Waiting out did not help. Tried the above mentioned https instead, no luck:
>
> 2024-03-12 12:38:06.999 ERROR (MainThread) [homeassistant.components.hassio.handler] [...]

Thanks @marst444 , the error `Failed to to call /store/repositories -` is interesting. 
Is it possible you could get to the terminal and run `ping github.com` ?

GeoDerp, 2024-03-12:
could you also run `uname -a `. the [ssh addon](https://github.com/home-assistant/addons/tree/master/ssh) should allow you to do both commands

davidusb-geek, 2024-03-12:
Actually I juste realized now that the images for ARM buil failed!

[image]

This is intriguing because the build for docker testing passed correctly here on the emhass core tests:

[image]

GeoDerp, 2024-03-12:
> Actually I just realized now that the images for ARM build failed!
> 
> ![image](https://private-user-images.githubusercontent.com/32926550/312076422-67265a3b-a0b8-481e-834f-fe5261fc8ece.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAyNDczNDYsIm5iZiI6MTcxMDI0NzA0NiwicGF0aCI6Ii8zMjkyNjU1MC8zMTIwNzY0MjItNjcyNjVhM2ItYTBiOC00ODFlLTgzNGYtZmU1MjYxZmM4ZWNlLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzEyVDEyMzcyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ3ODE5MTRkYWRmMzY0MDhiNDgwOTZkYzZhMTA2MjRkM2UyYzljZjczZmNhN2FhM2IxMWI3ZTI5OTVhYmEzZTYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.-zSMgk_QHmlOR77lOWo1kOSu4iOHHrmYPtapG66UhZo)
> 
> This is intriguing because the build for docker testing passed correctly here on the emhass core tests:
> 
> ![image](https://private-user-images.githubusercontent.com/32926550/312076807-6c82d2f9-7f38-4ae7-80c2-d371ff82ff23.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTAyNDczNDYsIm5iZiI6MTcxMDI0NzA0NiwicGF0aCI6Ii8zMjkyNjU1MC8zMTIwNzY4MDctNmM4MmQyZjktN2YzOC00YWU3LTgwYzItZDM3MWZmODJmZjIzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzEyVDEyMzcyNlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWMwZjgzNzA1YjQ0ODJlYTRjYWYxOGI4OTk3MTliM2JmNjhlMjg0ZWMxY2U3NzgwYTVkMGIwMDliZGY2NzJlNmMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.8NNv5RZbUYZuTPyMfwD3Fud6QL3whyIJPuw3oYF96I8)

Interesting. Ill have a look at the workflow and see if I can find anything.

GeoDerp, 2024-03-12:
> Oh boy with more than 5k of lines that's some logging there. But it seems that for example Publish-armv7 is trying to pull the `ghcr.io/home-assistant/amd64-builder:latest` image to build this. Probably the cause of the late failure when building?

Probably. im sure its the smallest issue. I think I might know what it is.

GeoDerp, 2024-03-12:
Actually. if 0.8.2 worked wouldn't users just be pulling 0.8.2 version for arm?

davidusb-geek, 2024-03-12:
Ok re-running...

marst444, 2024-03-12:
[github ping.pdf](https://github.com/davidusb-geek/emhass/files/14572680/github.ping.pdf)

GeoDerp, 2024-03-12:
its also having a lot of setuptools issues. maybe a pip install or apt-get may help. but why would it only brake once?

GeoDerp, 2024-03-12:
> [github ping.pdf](https://github.com/davidusb-geek/emhass/files/14572680/github.ping.pdf)

thanks @marst444, my version seems to be different from yours. Mine being `6.1.73-haos-raspi`. Since your running `arch64` it should be good to go, you can also ping github fine I see. So i'm wondering if there is a difference with `haos-raspi` and `haos`

GeoDerp, 2024-03-12:
meanwhile the action worked. 🤦

davidusb-geek, 2024-03-12:
Yes it worked now.
Yes that parameter can be a nice option to not wait 4hrs because of a timeout problem

GeoDerp, 2024-03-12:
We could also add packages like setup tools in so if it needs to compile the packages it can without error.

marst444, 2024-03-12:
Still no go for me though. Trying both suggested https: 
https://github.com/davidusb-geek/emhass-add-on 
https://github.com/davidusb-geek/emhass-add-on.git

2024-03-12 19:50:28.598 ERROR (MainThread) [homeassistant.components.hassio.handler] /store/repositories return code 400
2024-03-12 19:50:28.598 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /store/repositories - 
2024-03-12 19:50:37.436 ERROR (MainThread) [homeassistant.components.hassio.handler] /store/repositories return code 400

2024-03-12 19:50:37.437 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /store/repositories -

davidusb-geek, 2024-03-12:
Well that seems like a problem with your HA instance. Are you avle to add any other repository at all?

marst444, 2024-03-12:
Yes I can. And also add addons from repo. But cannot add the "https://github.com/davidusb-geek/emhass-add-on"
Errors as above....
[image]

marst444, 2024-03-12:
But now it showed up! This time i only had to wait it out a little longer it seems :) Will try to install now! :)

marst444, 2024-03-12:
Yes. Now its installed. Just the configure part to get it up and running i guess :) Thanx for your help!

davidusb-geek, 2024-03-12:
Glad to hear. Closing then.

closed 2024-03-12 by davidusb-geek