davidusb-geek/emhass #230: Huge icons on webui

davidusb-geek, 2024-03-11:
Hi @GeoDerp 
I don't know if you can see this same issue but in my webui there are now huge icons:

[image]

These icons where normal size on upper right corner on v0.8.1

GeoDerp, 2024-03-11:
Interesting, definitely nothing changed from 0.8.1.
By webui, it's like a PWA? 
I'll have a look to see if there is any max sizing on  the buttons. If not I'll set some. 
It may be a good idea to have a PR that just focuses on improved PWA support.

purcell-lab, 2024-03-12:
I get large icons on mobile but not my laptop.

[image]

GeoDerp, 2024-03-12:
Thanks @michaelpiron,  It looks like the same issue with some sort of styling not being propperly enabled. Im having a look at this now.

GeoDerp, 2024-03-12:
Is it possible that one of your can test this out: https://github.com/GeoDerp/emhass/tree/index-icon-fix

GeoDerp, 2024-03-12:
Just added an JS function that checks if stylsheet exists, if not adds it. *(redundancy)*

GeoDerp, 2024-03-12:
Dad had the idea it could be a DNS issue. He might not be wrong. Ie. If you're using DNS it may not be getting called probably inside the webui. The JS function may help with this I'm not sure.

GeoDerp, 2024-03-13:
Let me know if this has been fixed for you guys.  👍

michaelpiron, 2024-03-13:
> Thanks @michaelpiron, It looks like the same issue with some sort of styling not being propperly enabled. Im having a look at this now.
> 
> Its kinda annoying that this doesn't seem to be an issue on my end. Does the other elements styling look correct? _(buttons have shadows, etc)_. If you run http://yourhaip:5000, does it still look that same.
> 
> Also @michaelpiron , is your mobile an IPhone? Mine is a Android.

Hi @GeoDerp , could it be that you addressed the wrong person? I don’t recall this issue.

purcell-lab, 2024-03-13:
Yes I do get styling from the webUI interface.

[image]

GeoDerp, 2024-03-13:
> Yes I do get styling from the webUI interface.
> 
> [image]
> 

Yeah interesting. You do, but even things like the basic info box doesn't show up.

GeoDerp, 2024-03-13:
Sorry @michaelpiron , that was my bad.meant that to be @purcell-lab. 

@purcell-lab , could you clear the app cache in the android settings then force killing the app and rerunning? 

[image]

[image]

purcell-lab, 2024-03-13:
Cache cleared, no change unfortunately.

GeoDerp, 2024-03-13:
> Cache cleared, no change unfortunately.

@purcell-lab does the buttons (svg and actions) work? 
If so I may just embed the CSS file.

purcell-lab, 2024-03-14:
Buttons are functional with large icons.

Not sure how to test svg?

GeoDerp, 2024-03-14:
@purcell-lab could you test this for me. Made a EMHASS fork, Bellow are the steps to build emhass-add-on using EMHASS from my repo and branch. *(feel free to keep your original EMHASS, just running this version on a separate port)*

```bash
cd ~/addons/
git clone https://github.com/davidusb-geek/emhass-add-on
sed -i.bak '/image:/ s/./#&/' ~/addons/emhass-add-on/emhass/config.yml
sed -i.bak "s/build_version:.*/build_version: addon-git/g"  ~/addons/emhass-add-on/emhass/build.yaml
repo=https://github.com/GeoDerp/emhass.git
branch=index-icon-fix
sed -i.bak "s%build_repo:\s.*%build_repo: $repo%g"  ~/addons/emhass-add-on/emhass/build.yaml
sed -i.bak "s/build_branch:\s.*/build_branch: $branch/g"  ~/addons/emhass-add-on/emhass/build.yaml
```

GeoDerp, 2024-03-16:
Update, tested with Nabu Casa . No problems with current version or fork. @davidusb-geek or @michaelpiron . If your still having issues if you could test the fork above and see if that fixes your current solution that would be great.

purcell-lab, 2024-03-18:
I haven't updated since 0.8.4.

Icons are now a good size on laptop and mobile:

[image]

Graphs are also a lot more usable on mobile as well, I even get yellow for PV ;-)

[image]

Great work..

GeoDerp, 2024-03-18:
> I haven't updated since 0.8.4.
> 
> Icons are now a good size on laptop and mobile:
> 
> [image]
> 
> Graphs are also a lot more usable on mobile as well, I even get yellow for PV ;-)
> 
> [image]
> 
> Great work..

I guess it was a cache problem after all. 
@davidusb-geek, if you think the PR is still good (replacing urls with relatives). free to merge it.

Otherwise I'll close it, and I may make another PR with just the CSS JS function removed. As, I believe it slows down the page loading time.

GeoDerp, 2024-03-18:
@purcell-lab, the diagram is just an inverted colour of the light mode. PV being yellow is an accidental benefit. 😂

GeoDerp, 2024-03-18:
@davidusb-geek, feel free to close this issue after reading this. 👍🏼

davidusb-geek, 2024-03-18:
> @davidusb-geek, feel free to close this issue after reading this. 👍🏼

Ok but what do you recommend regarding PR #239?
If load times are affected then it should be better to do this?:
> Otherwise I'll close it, and I may make another PR with just the CSS JS function removed. As I believe it slows down the page loading time.

GeoDerp, 2024-03-26:
@davidusb-geek , If you have any issues @ me and ill help revert.

davidusb-geek, 2024-03-26:
Ok, I merged. Then will test it and see how it goes.

davidusb-geek, 2024-04-01:
Seems solved to me.
Closing

closed 2024-04-01 by davidusb-geek