davidusb-geek/emhass #233: Pulp solver issue in 0.8.3

tanelvakker, 2024-03-12:
When running mpc with custom data, I get 500 error and error in log

2024-03-12 12:00:00,160 - web_server - INFO - Passed runtime parameters: {'load_cost_forecast': [0.156, 0.153, 0.153, 0.154, 0.163, 0.178, 0.195, 0.217, 0.207, 0.175, 0.139, 0.134], 'prod_price_forecast': [0.0807, 0.0784, 0.0783, 0.079, 0.0863, 0.0993, 0.1131, 0.1321, 0.1234, 0.0965, 0.083, 0.0785], 'prediction_horizon': 12, 'soc_init': 0.68, 'soc_final': 0.2, 'def_total_hours': [0, 0], 'pv_power_forecast': [10795, 9971, 6408, 5533, 4192, 2375, 651, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 14, 478, 1654, 2460, 2544]}
2024-03-12 12:00:00,160 - web_server - INFO -  >> Setting input data dict
2024-03-12 12:00:00,160 - web_server - INFO - Setting up needed data
2024-03-12 12:00:00,171 - web_server - INFO - Retrieve hass get data method initiated...
2024-03-12 12:00:03,479 - web_server - INFO - Retrieving weather forecast data using method = list
2024-03-12 12:00:03,479 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2024-03-12 12:00:03,480 - web_server - INFO - Retrieve hass get data method initiated...
2024-03-12 12:00:11,893 - web_server - INFO -  >> Performing naive MPC optimization...
2024-03-12 12:00:11,893 - web_server - INFO - Performing naive MPC optimization
2024-03-12 12:00:11,897 - web_server - INFO - Perform an iteration of a naive MPC controller
2024-03-12 12:00:12,090 - web_server - ERROR - Exception on /action/naive-mpc-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1463, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 872, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 870, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 855, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/web_server.py", line 141, in action_call
    opt_res = naive_mpc_optim(input_data_dict, app.logger)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/command_line.py", line 285, in naive_mpc_optim
    opt_res_naive_mpc = input_data_dict['opt'].perform_naive_mpc_optim(
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/optimization.py", line 628, in perform_naive_mpc_optim
    self.opt_res = self.perform_optimization(df_input_data, P_PV.values.ravel(), P_load.values.ravel(),
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/optimization.py", line 433, in perform_optimization
    opt_model.solve(COIN_CMD(msg=0, path=self.lp_solver_path))
  File "/usr/local/lib/python3.11/dist-packages/pulp/pulp.py", line 1883, in solve
    status = solver.actualSolve(self, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pulp/apis/coin_api.py", line 112, in actualSolve
    return self.solve_CBC(lp, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pulp/apis/coin_api.py", line 181, in solve_CBC
    raise PulpSolverError(
pulp.apis.core.PulpSolverError: Pulp: Error while trying to execute, use msg=True for more details/usr/bin/cbc

 - Home Assistant OS

- OS: HA OS
- Architecture: amd64

**EMHASS installation type**
 - Add-on

Worked before upgrade

tanelvakker, 2024-03-12:
These are already set this way

GeoDerp, 2024-03-12:
mmm. could you test lp_solver to `PULP_CBC_CMD` and `default` and path to `empty`

GeoDerp, 2024-03-12:
tested with MPC with this: 

```json
{
    "load_cost_forecast": [
        0.156,
        0.153,
        0.153,
        0.154,
        0.163,
        0.178,
        0.195,
        0.217,
        0.207,
        0.175,
        0.139,
        0.134
    ], "prod_price_forecast": [
        0.0807,
        0.0784,
        0.0783,
        0.079,
        0.0863,
        0.0993,
        0.1131,
        0.1321,
        0.1234,
        0.0965,
        0.083,
        0.0785
    ], 
    "prediction_horizon": 12,
    "soc_init": 0.68, 
    "soc_final": 0.2, 
    "def_total_hours": [
        0,
        0
    ], 
    "pv_power_forecast": [
        10795,
        9971,
        6408,
        5533,
        4192,
        2375,
        651,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        0,
        14,
        478,
        1654,
        2460,
        2544
    ]
}
```

This worked but Ill need to try and see if I can replicate the issue in Docker.

tanelvakker, 2024-03-12:
setting to PULP_CBC_CMD and empty worked

GeoDerp, 2024-03-12:
> setting to PULP_CBC_CMD and empty worked

Thanks for that. We will see if other have the same issue.

davidusb-geek, 2024-03-14:
Closing as solved

closed 2024-03-14 by davidusb-geek