davidusb-geek/emhass #235: Feature Request: Support for limiting amount that can be exported to the grid

philjohn, 2024-03-13:
**Describe the bug**

At present there is a setting for the amount you can pull in from the grid, however, as Distributed Solar PV continues to grow, many network operators are placing limits on how much power can be sent back to the grid - e.g. g100 export limitation in the UK.

**Expected behavior**

A new configuration setting that allows the maximum export to be defined

davidusb-geek, 2024-03-14:
Hi yes for now this a limitation. The maximum power from and to the grid is the same.
We will split that into separate configurable parameters

kcoffau, 2024-03-19:
+1 on this.

davidusb-geek, 2024-04-01:
Implemented in the fresh release. A new configuration parameter `P_to_grid_max` was added. 
Closing as completed

closed 2024-04-01 by davidusb-geek