davidusb-geek/emhass #240: Feature Request - Forecast Horizon - extend beyond 48 values

kcoffau, 2024-03-19:
**Describe the improvement**
I have been happily using 48 forecast values in 30 min blocks for a long time now.  I am now able to forecast 6 days in 30 min increments, 180 variables.

I updated my curl to push 96 values (as a starting point) and get error: 
"IndexError: index 95 is out of bounds for axis 0 with size 48"

So....

I dont have time this week, but I was going to have a look next week and see what I can do, but before I do, I want to check, is my understanding of 48 values in the forecast the current limitation?  Is that for performance or other?

Cheers,
Kel.

davidusb-geek, 2024-03-19:
Yes this is a hard coded parameter for performance concerns. If extended we could quickly fall into memory issues and that will be bad for your HA instance. However we could be flexible and add an additional optional parameter to extend this limitation if needed by the user.

Why is your goal for extending this limitation and optimizing over 24 h span?

kcoffau, 2024-03-19:
I am trying to allow longer range forecasting to allow better charge/discharge of battery.

Example deeper discharge if it can see a good solar day in 2 days, or higher state of charge if solar forecast is low.  I think 2-3 days would allow battery cycling to be more accurate instead of chasing end charge state 24 hours away, if it was 3 days away the end state influence would be diluted.

Agree it could be a variable to manage resource issues.

MikaelHoogen, 2024-03-19:
+1 on this. The battery forecasting would be better. I live in Sweden with nordpool, so when we get the prices for the upcoming day there is very much use to go beyond 48 values.

For my part an optional parameter would be great. The performance shouldn't be a problem here, my server needs something to do :)

kcoffau, 2024-03-19:
Just on performance, I am running it in a VM, so I can just allocate more resources - if needed.

I appreciate those running on devices with restricted resources will need a way of limiting the forecast either by days or intervals.

davidusb-geek, 2024-03-21:
When these problems fall into the curse of dimensionality, the memory need grows extremely fast exponentially. Even in a machine over teraflops you are done

davidusb-geek, 2024-03-21:
But still this feature will be added, just be warned

davidusb-geek, 2024-04-01:
This is already a possibility. Just set `delta_forecast` to a bigger value, the default is 1, but you can fix it to 2, 3, etc...
Closing for now but reopen if needed.

kcoffau, 2024-04-02:
So, let me start with david, you amazing, and thank you for pointing out the delta_forecast value.   Cant believe I didnt see it.

ive extended that to 7.   Im loading in 96 values, and my optimisation step is 30.

Im getting shape errors below.   Cant understand why its referencing 44 or 22. 

When I run 48 values in, it runs with no issue.

Trying to work out what I am doing wrong.  Any ideas on starting points would be great.

CURL Source:
24 hour:
post_mpc_optim_ev_freelunch: "curl -i -H \"Content-Type: application/json\" -X POST -d '{\"load_cost_forecast\":[{% set current_time = now() %}{% set base_time = current_time.replace(minute=0, second=0, microsecond=0) if current_time.minute < 30 else current_time.replace(minute=30, second=0, microsecond=0) %}{% for offset in range(48) %}{% set nowish = base_time + timedelta(minutes=30*offset) %}{{ '0.081' if 0 <= nowish.hour < 6 else '0.00' if 11 <= nowish.hour < 14 else '0.40' }}{% if not loop.last %}, {% endif %}{% endfor %}], \"prod_price_forecast\":[{% for _ in range(48) %}0.08{% if not loop.last %}, {% endif %}{% endfor %}], \"pv_power_forecast\":{{(((state_attr('sensor.solcast_pv_forecast_forecast_today', 'detailedForecast') +(state_attr('sensor.solcast_pv_forecast_forecast_tomorrow', 'detailedForecast'))) |selectattr('period_start', 'ge', now() - timedelta(minutes=29.99))  | map(attribute='pv_estimate')|map('multiply',1000)|map('int')|list)[:48])}}, \"prediction_horizon\":{{min(48,(state_attr('sensor.amber_feed_in_forecast', 'forecasts')|map(attribute='per_kwh')|list|length)+1)}},\"soc_init\":{{states('sensor.victron_battery_state_of_charge')|float(0)/100}},\"def_total_hours\":[0]}' http://localhost:5000/action/naive-mpc-optim"

48 Hour:
  post_mpc_optim_ev_freelunch_ext: "curl -i -H \"Content-Type: application/json\" -X POST -d '{\"load_cost_forecast\":[{% set current_time = now() %}{% set base_time = current_time.replace(minute=0, second=0, microsecond=0) if current_time.minute < 30 else current_time.replace(minute=30, second=0, microsecond=0) %}{% for offset in range(96) %}{% set nowish = base_time + timedelta(minutes=30*offset) %}{{ '0.081' if 0 <= nowish.hour < 6 else '0.00' if 11 <= nowish.hour < 14 else '0.40' }}{% if not loop.last %}, {% endif %}{% endfor %}], \"prod_price_forecast\":[{% for _ in range(96) %}0.08{% if not loop.last %}, {% endif %}{% endfor %}], \"pv_power_forecast\":{{(((state_attr('sensor.solcast_pv_forecast_forecast_today', 'detailedForecast') +(state_attr('sensor.solcast_pv_forecast_forecast_tomorrow', 'detailedForecast')) +(state_attr('sensor.solcast_pv_forecast_forecast_day_3', 'detailedForecast')) +(state_attr('sensor.solcast_pv_forecast_forecast_day_4', 'detailedForecast')) +(state_attr('sensor.solcast_pv_forecast_forecast_day_5', 'detailedForecast')) +(state_attr('sensor.solcast_pv_forecast_forecast_day_6', 'detailedForecast')) +(state_attr('sensor.solcast_pv_forecast_forecast_day_7', 'detailedForecast'))) |selectattr('period_start', 'ge', now() - timedelta(minutes=29.99))  | map(attribute='pv_estimate')|map('multiply',1000)|map('int')|list)[:96])}}, \"prediction_horizon\":{{min(96,(state_attr('sensor.solcast_pv_forecast_forecast_today', 'detailedForecast')|length + state_attr('sensor.solcast_pv_forecast_forecast_tomorrow', 'detailedForecast')|length + state_attr('sensor.solcast_pv_forecast_forecast_day_3', 'detailedForecast')|length + state_attr('sensor.solcast_pv_forecast_forecast_day_4', 'detailedForecast')|length + state_attr('sensor.solcast_pv_forecast_forecast_day_5', 'detailedForecast')|length + state_attr('sensor.solcast_pv_forecast_forecast_day_6', 'detailedForecast')|length + state_attr('sensor.solcast_pv_forecast_forecast_day_7', 'detailedForecast')|length))}},\"soc_init\":{{states('sensor.victron_battery_state_of_charge')|float(0)/100}},\"def_total_hours\":[0]}' http://localhost:5000/action/naive-mpc-optim"

Curl Translated Output:
24 hour:
  post_mpc_optim_ev_freelunch: "curl -i -H \"Content-Type: application/json\" -X POST -d '{\"load_cost_forecast\":[0.00, 0.00, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.081, 0.081, 0.081, 0.081, 0.081, 0.081, 0.081, 0.081, 0.081, 0.081, 0.081, 0.081, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.00, 0.00, 0.00, 0.00], \"prod_price_forecast\":[0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08, 0.08], \"pv_power_forecast\":[685, 1112, 409, 469, 847, 1182, 1581, 1275, 770, 250, 90, 18, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 60, 811, 1634, 2351, 2982, 3598, 4091, 4180, 4188, 4111, 3961, 3778], \"prediction_horizon\":32,\"soc_init\":0.716,\"def_total_hours\":[0]}' http://localhost:5000/action/naive-mpc-optim"

  48hour:
  post_mpc_optim_ev_freelunch_ext: "curl -i -H \"Content-Type: application/json\" -X POST -d '{\"load_cost_forecast\":[0.00, 0.00, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.081, 0.081, 0.081, 0.081, 0.081, 0.081, 0.081, 0.081, 0.081, 0.081, 0.081, 0.081, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.00, 0.00, 0.00, 0.00, 0.00, 0.00, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0.40, 0 [... cut ...]
    mgr = ndarray_to_mgr(
          ^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/internals/construction.py", line 337, in ndarray_to_mgr
    _check_values_indices_shape_match(values, index, columns)
  File "/usr/local/lib/python3.11/dist-packages/pandas/core/internals/construction.py", line 408, in _check_values_indices_shape_match
    raise ValueError(f"Shape of passed values is {passed}, indices imply {implied}")
**ValueError: Shape of passed values is (44, 1), indices imply (22, 1)**

rixxxx, 2024-04-03:
@kcoffau let us know if you figure it out. I have similar issue when trying 26h forecast with delta_forecast 2.
```emhass-1  | ValueError: Shape of passed values is (4, 1), indices imply (2, 1)```

davidusb-geek, 2024-04-03:
Ok so I went down to this and this is actually a bug!
Actually I realized that we never tested the code well for delta_forecast > 1
Working on fix right now

davidusb-geek, 2024-04-03:
Done! 
Normally solved here: https://github.com/davidusb-geek/emhass/pull/250
Should be available on the next release

rixxxx, 2024-04-04:
My 34h forecast now works. Thanks.

kcoffau, 2024-04-07:
0.8.6 

2 days forecast!!!

Absolutely beautiful.  Thank you David!

[image]
[image]

closed 2024-04-07 by kcoffau

purcell-lab, 2024-04-08:
After 12.30pm each day our energy market operator (AEMO) publishes 30 minute forecasts until 3.30am the day after the following day (ie for the next 39 hours), they then update this set of forecasts every 5 minutes, always finishing at 3.30am.

EMHASS can now consume the full 39 hours and with MPC can update it's optimisation every time the pricing changes.

Thanks @davidusb-geek .

[image]
[image]