davidusb-geek/emhass #242: Feature Request - Changing minimum optimization_time_step to 5 min and prediction_horizon to 288

danielclau, 2024-03-19:
Currently optimization_time_step only allows a minimum of 10 and prediction_horizon to 144

Our energy provider's API provides electricity prices in 5 minute intervals.
To be able to utilise these 5 minute prices, optimization_time_step needs to be set to 5 and prediction_horizon to 288

purcell-lab, 2024-03-19:
The Amber API does provide the 5 minute resolution forecasts and they are heading (slowly) to 5 minute billing, so this is something to get ahead of.

LocalVolts is billing in 5 minute intervals so it is possible to require the resolution for solving.

EMHASS is also solving for solar & household load, which I feed in real time.  I don't want to wait 2-3 minutes to notice the kettle or a cloud so I run MPC every minute to pickup these changes as well.

danielclau, 2024-03-19:
I'm looking at the Localvolts API which returns 287 forecast electricity prices in 5 minutes intervals.
Other users there report they can obtain the Localvolts billing price at around 20 seconds into the 5 min interval. So if EMHASS can mpc_optim in 10 seconds, then we can change our electricity usage in the remaining 4:30 if there is a price spike.

I'm currently on Amber but since Myles has confirmed that Amber is moving to 5 minutes billing this year, Amber users will likely to need this change this year anyway.

closed 2024-04-01 by davidusb-geek

danielclau, 2024-04-02:
Thanks Dave.

Confirmed working in v0.85
Changed EMHASS config optimization_time_step to 5
Added timeout: 30 for REST API (https://www.home-assistant.io/integrations/rest_command) 
Ran naive-mpc-optim via REST
Passed in prediction_horizon: 288
Passed in 288 values for pv_power_forecast, load_power_forecast, load_cost_forecast, prod_price_forecast

Runs for around 15 seconds and gives back 288 lots of 5 minute forecasts (totalling 24 hours)
[image]
[image]

purcell-lab, 2024-04-02:
You can get 5 minute interval data from the AmberAPI, with the following, but only upto around the next 10 intervals. So could run day-ahead optimisation on the 30 minute interval data and then MPC on the 5 minute interval data with the much shorter prediction_horizon.

```
curl -X 'GET'   'https://api.amber.com.au/v1/sites/SITE_ID/prices/current?next=10&resolution=5'   \
 -H 'accept: application/json'   -H 'Authorization: BEARER_PSK_KEY' | \
 jq '.[] | select(.channelType == "general") | .nemTime, .type, .perKwh, .spotPerKwh'
 
"2024-04-02T16:05:00+10:00"
"CurrentInterval"
8.84585
1.29386
"2024-04-02T16:10:00+10:00"
"ForecastInterval"
7.61082
0.1039
"2024-04-02T16:15:00+10:00"
"ForecastInterval"
7.61152
0.10457
"2024-04-02T16:20:00+10:00"
"ForecastInterval"
7.61223
0.10525
"2024-04-02T16:25:00+10:00"
"ForecastInterval"
11.90407
4.2405
...
```