davidusb-geek/emhass #243: Solcast time after day ahead optimization not correct

BDVGitHub, 2024-03-20:
**Describe the bug**
I'm using Solcast for the solar forcast and filled in the paramters in te config en secret file and it worked before.

Since I'v updated to version 0.8.4 the predicted solar data is not on the right time stamp
Its like the solar data is shifted 10h further than expected (see screenshot)

I'm using solcast also in the energy dashboard on HA and there it works fine

**Expected behavior**
That the solar data is shown on the right timeslot

**Screenshots**
log file
[image]

data after day ahaed optimization
[image]
[image]

**Home Assistant installation type**
 - Home Assistant Supervised

**Your hardware**
- OS:  Linux
- Architecture: amd64

**EMHASS installation type**
 - Docker Standalone

davidusb-geek, 2024-03-20:
Are you on TZ +1h?

BDVGitHub, 2024-03-21:
yes correct, I live in Belgium
In the docker container the time is running correct and in the secret file the timezone is also set

davidusb-geek, 2024-03-25:
Very strange.
When starting the app with the `docker run` command try adding the env variable `-e TIME_ZONE="Europe/Paris"` to directly specify the TZ and see if it makes any difference.

GeoDerp, 2024-03-28:
Your timezone parameter may be correct. However, It may be that the timezone on the standalone Docker file itself is incorrect. Presenting incorrect timezone values on the web sever. 

After trying @davidusb-geek 's recommendation. Try: `-e TZ="Europe/Paris"`:
So something line `docker run ... -e TIME_ZONE="Europe/Paris" `-e TZ="Europe/Paris ... `

BDVGitHub, 2024-03-28:
I've removed the container en build it with `docker run ... -e TIME_ZONE="Europe/Paris" -e TZ="Europe/Paris ... `

the time is correct in the container see image, but still same issue 

[image]

BDVGitHub, 2024-03-31:
In the log file I found this error 

2024-03-31 18:47:25,420 - web_server - INFO - Status: Optimal
 2024-03-31 18:47:25,420 - web_server - INFO - Total value of the Cost function = -1.58
 2024-03-31 18:47:25,423 - web_server - INFO - Solving for day: 30-3-2024
 2024-03-31 18:47:25,424 - web_server - ERROR - Exception on /action/perfect-optim [POST]
 Traceback (most recent call last):
   File "/usr/local/lib/python3.11/dist-packages/pandas/core/arrays/datetimes.py", line 2423, in _infer_tz_from_endpoints
     inferred_tz = timezones.infer_tzinfo(start, end)
                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "pandas/_libs/tslibs/timezones.pyx", line 368, in pandas._libs.tslibs.timezones.infer_tzinfo
 AssertionError: Inputs must both have the same timezone, UTC+01:00 != UTC+02:00
 
 The above exception was the direct cause of the following exception:
 
 Traceback (most recent call last):
   File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1463, in wsgi_app
     response = self.full_dispatch_request()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 872, in full_dispatch_request
     rv = self.handle_user_exception(e)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 870, in full_dispatch_request
     rv = self.dispatch_request()
          ^^^^^^^^^^^^^^^^^^^^^^^
   File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 855, in dispatch_request
     return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/usr/local/lib/python3.11/dist-packages/emhass/web_server.py", line 119, in action_call
     opt_res = perfect_forecast_optim(input_data_dict, app.logger)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/usr/local/lib/python3.11/dist-packages/emhass/command_line.py", line 207, in perfect_forecast_optim
     opt_res = input_data_dict['opt'].perform_perfect_forecast_optim(df_input_data, input_data_dict['days_list'])
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/usr/local/lib/python3.11/dist-packages/emhass/optimization.py", line 538, in perform_perfect_forecast_optim
     data_tp = df_input_data.copy().loc[pd.date_range(start=day_start, end=day_end, freq=self.freq)]
                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/usr/local/lib/python3.11/dist-packages/pandas/core/indexes/datetimes.py", line 945, in date_range
     dtarr = DatetimeArray._generate_range(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/usr/local/lib/python3.11/dist-packages/pandas/core/arrays/datetimes.py", line 422, in _generate_range
     tz = _infer_tz_from_endpoints(start, end, tz)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
   File "/usr/local/lib/python3.11/dist-packages/pandas/core/arrays/datetimes.py", line 2426, in _infer_tz_from_endpoints
     raise TypeError(
 TypeError: Start and end cannot both be tz-aware with different timezones

davidusb-geek, 2024-04-01:
I tried but I'm currently not able to reproduce this.

BDVGitHub, 2024-04-06:
When I change in config_emhass.yaml `retrieve_hass_conf:
  freq: 60 # The time step to resample retrieved data from hass in minutes` back to `retrieve_hass_conf:
  freq: 30 # The time step to resample retrieved data from hass in minutes`
The solcast data is shown correct again, the reason why i changed the resample time from **30** to **60** is the following,
I use in homeassistant an sensor to load the price per hour because its not always the same, this data is given per hour.

my question is how can I change the code below to output 30 minute data instead of 60 minute data?

`  - platform: template
    sensors:
      load_cost_forecast_list:
        value_template: >-
          {% if now().weekday() == 5 or now().weekday() == 6 %}
            {%- set load_cost = [0.115, 0.115, 0.115, 0.115, 0.115, 0.115, 0.115, 0.115, 0.115, 0.115, 0.115, 0.115, 0.115, 0.115, 0.115, 0.115, 0.115, 0.115, 0.115, 0.115, 0.115, 0.115, 0.115, 0.115] %}
          {%- else -%}
            {%- set load_cost = [0.115, 0.115, 0.115, 0.115, 0.115, 0.115, 0.155, 0.155, 0.155, 0.155, 0.155, 0.155, 0.155, 0.155, 0.155, 0.155, 0.155, 0.155, 0.155, 0.155, 0.155, 0.155, 0.115, 0.115] %}
          {%- endif %}
          {%- set values_all = namespace(all=[]) %}
          {% for i in range(load_cost | length) %}
            {%- set v = (load_cost[i] | float ) %}
            {%- set values_all.all = values_all.all + [ v ] %}
          {%- endfor %} {{ (values_all.all)[now().hour:24] + (values_all.all)[0:now().hour] }}`

so that I can send 48 data points instead of 24 data points to the load cost forcast...

davidusb-geek, 2024-04-07:
Ok I will try to reproduce this by changing the `retrieve_hass_conf`. It might be a bug, so I will do some tests. 
For your question regarding the code I don't know if I understand well, but you just need to expand your lists to have 48 elements. 

For example this list with 3 elements:
```
list=[1,2,3]
```
Can be expanded to 6 elements like this:
```
list=[1,1,2,2,3,3]
```

davidusb-geek, 2024-05-01:
Are you still having this issue?

BDVGitHub, 2024-05-07:
Yes, even after updating to version 0.8.6

when using `retrieve_hass_conf: freq: 30 #` in the config file it works perfect
when using `retrieve_hass_conf: freq: 60 #` the solcast data starts @ 14h

BDVGitHub, 2024-05-07:
This is an part of the solcast data (yellow) with `retrieve_hass_conf: freq: 30 #`
The solcast data start @ 6h30
[image]

This is an part of the solcast data (yellow) with `retrieve_hass_conf: freq: 60 #`
The solcast data start @ 14h
If you look to the steps off the solcast data, @ 14h the solcast value is "2W", @ 15h I expected "79W" but see "32W" because there is an time difference of 1h and not 30 minutes.
Its like the data array from solcast is stil 30 minute data but the emhass data is 60 minute data.
because  
[image]

davidusb-geek, 2024-05-08:
Ok there is probably a bug with Solcast not adapting to the new `freq` values. However in your original issue the dat was shifted 10h, so that's some improvement! ;-)

As for your second question I already answered before like this:
Just update your lists in your template sensor. 
For example this list with 3 elements:
```
list=[1,2,3]
```
Can be expanded to 6 elements like this:
```
list=[1,1,2,2,3,3]
```

closed 2024-05-13 by davidusb-geek