davidusb-geek/emhass #244: SOC Forecast Never 1.00

JoshC1994, 2024-03-25:
Is it possible to have the SOC Forecast return a value of 1.00 if commanded by EMHASS?  

I find it always defaults to a value of 0.990 or 99%. 
It would be nice for it to return the value of 1.00 which would help with writing code and automations. 

[image]

davidusb-geek, 2024-03-25:
It will default to the value that you tell the system to default.
Are you using day-ahead optimization?
If yes, then this is directly the value that you entered in the configuration file.
Are you saying that you put 1.0 in the configuration file and that the table results always show 0.99.
This is not possible and has been tested.
What is you SOCmax value?

JoshC1994, 2024-03-25:
@davidusb-geek- my bad I had this set at the default 0.99. issue solved

closed 2024-03-25 by JoshC1994