davidusb-geek/emhass #256: Improvement: Use HomeAssistant Statistics

nateguchi, 2024-04-17:
Hi

My energy usage is updated every few seconds and that causes the data fetch from home assistant to take quite a while. Is it possible to use the statistics API to get the totals / averages for an hour or half hour period to speed things up?

davidusb-geek, 2024-04-17:
Yes this could be great, but I just don't know how to do it.
The HA API is only able to retrieve normal data, not HA statistics data.
Here are the two APIs documentations that are used in the code: 
- For using the supervisor: [https://developers.home-assistant.io/docs/api/supervisor/endpoints/#addons](https://developers.home-assistant.io/docs/api/supervisor/endpoints/#addons)
- For using the core API: [https://developers.home-assistant.io/docs/api/rest/](https://developers.home-assistant.io/docs/api/rest/)

In neither of those it is possible to retrieve statistics data.
If anyone finds a solution I'm wiling to take a look at this possibility.

gieljnssns, 2024-04-21:
Can statistics be retrieved through the websocket api?

davidusb-geek, 2024-04-30:
> Can statistics be retrieved through the websocket api?

I don't know, how?

closed 2024-05-13 by davidusb-geek