davidusb-geek/emhass #258: How to handle dynamic prices (Tibber/Nord Pool)

a8291, 2024-04-19:
Currently it's only possible to feed static prices for electricity. It would be nice to feed dynamic prices for each hour smiliar to the PV forecast. If i am correct is this not possible yet.

davidusb-geek, 2024-04-19:
No, you are not correct. 
It is possible to pass dynamic prices for: PV forecast, load forecast, PV production price forecast and load cost forecasts.
Just use the correct key when passing data at runtime.
Documented here: [https://emhass.readthedocs.io/en/latest/intro.html#forecast-data](https://emhass.readthedocs.io/en/latest/intro.html#forecast-data)

davidusb-geek, 2024-04-19:
And there is actually a working example for using Nordpool: [https://emhass.readthedocs.io/en/latest/forecasts.html#example-using-the-nordpool-integration](https://emhass.readthedocs.io/en/latest/forecasts.html#example-using-the-nordpool-integration)

a8291, 2024-04-19:
Wow, thanks for the quick response!

I just worked with the configuration in the addon. There is only possible to enter static data.

[image]

davidusb-geek, 2024-04-19:
Yes those are static from the add-on configuration pane. But then there is the possibility to dynamically pass data as list during the shell or REST commands.
See the documentation on the links that I just posted before

closed 2024-04-19 by a8291