davidusb-geek/emhass #261: Feature request: startup penalty

werdnum, 2024-04-23:
I am using emhass to control my pool pump based on Amber electricity prices. Often, the solution EMHASS comes up with switches the pump on and off repeatedly, which is probably a bit more switching than I'd like. The pump also runs faster (2x energy consumption) for the first few minutes, which probably negates the benefit from running at 22c/kWh rather than 23c/kWh.

I think I'd like to alter the algorithm to allow specifying a penalty (in minutes or cents) for each startup. That way, the optimiser can split the usage in two if it really helps, but won't stop and start every 30 minutes when the power price fluctuates slightly over the day.

davidusb-geek, 2024-05-01:
Turning on the option to make it a continuous load with just one startup is not an option for you?

werdnum, 2024-05-01:
It’s an option, but not the preferred one. Starting up twice is fine, but
it needs to be a big enough win to justify it.

— Andrew Garrett

werdnum, 2024-05-10:
I have this somewhat working in [a branch](https://github.com/werdnum/emhass/tree/start-penalty). Want to clean it up before sending a PR.

davidusb-geek, 2024-05-10:
Hi, ok nice feature!
The annoying part will be to add a new user-define input variable to control this.
I think that by default no penalty should be applied on the number of startups. Then if the user desires as it was your case, this option can be activated. 
This has been a constant struggle, add parameters so that this optimization can be highly configurable, but at the same each new parameter adds more complexity and the people complain about how all this is hard to setup.
But go ahead these are nice features. Try to add some lines to `test_optimization.py` to quickly test the desired/expected behavior of the feature.

davidusb-geek, 2024-06-02:
This is included in the just released new version. 
Closing this for now but reopen if eneded.

closed 2024-06-02 by davidusb-geek