davidusb-geek/emhass #266: Feature Request: Use the home assistent entity 'epex.spot.data.price'

WimGo, 2024-05-01:
There is a home assistant HACS integration called 'EPEX Spot': https://github.com/mampfes/ha_epex_spot .
It gives the base prices for dynamic electricity contracts, from the EPEX market for the country one has configured. This has been working on my home assistant server flawlessly for a few months now.
The prices are rendered in an home assistant sensor: 'sensor.epex_spot_data_price'. It gets updated regularly, so that I have the next day's prices, every day, very soon after they are defined on the EPEX market.
It would be great if EMHASS could use this sensor as input for dynamic electricity prices, as I haven't found a way to download the price CSVs for Belgium on a daily basis.

davidusb-geek, 2024-05-01:
You just need to build a new template sensor that prepares the data from that sensor that you have to be passed to emhass at runtime. There are some examples for this in the documentation.
Can you show what does that sensor looks like? What are its attributes?

WimGo, 2024-05-01:
[image]

The attributes consist of 24 sets of 4 data points. These 24 sets cover the current day and the next day, starting and ending at midnight.
Each of these sets contain 4 data points:
- start_time (date, time till seconds
- end time (idem)
- price_eur_per_MWh
- price_ct_per_kwh

Octofinger, 2024-05-01:
Try something like this to create a variable that contains a list of those elements that you can pass to emhass at runtime. It's NOT tested, just a suggestion to use a contruct similar to my use of Nordpool integration in HA.

```
{%- set hourly_prices=((state_attr('sensor.epex_spot_data_price', 'data') | map(attribute='price_ct_per_kwh') | list
        + state_attr('sensor.epex_spot_data_price', 'data') | map(attribute='price_ct_per_kwh') | list))
        [now().hour:][:24] %}
```

closed 2024-05-10 by davidusb-geek