davidusb-geek/emhass #275: Solcast integration with `delta_forecast` > 1 fails

toggm, 2024-05-07:
**Describe the bug**
Using `solcast` forecast model together with `delta_forecast` set to value > 1 fails with:
```
ERROR - Not enough data retrived from SolCast service, try increasing the time step or use MPC
```
**EMHASS installation type**
 - Docker Standalone

**Additional context**
Without testing or adjust, IMO problem comes from the fixed time frame when calling the API:

https://github.com/davidusb-geek/emhass/blob/d06b829a8de79ccb9a0e70cb5fbc5d03c338dfdb/src/emhass/forecast.py#L238

Value of `hours` should be  a factor of `delta_forecast` to be compatible with the expected number of results. As far as I could see requests are not limited to 24 hours only. Otherwise startup of app should already fail in case of invalid configuration parmeters.

davidusb-geek, 2024-05-10:
Hi. Yes you are right, this is hard coded to retrieve just 24h of data. 
Yes we could make it a function of the passed `delta_forecast`.
But then we need to know what is the maximum possible period to retrieve data from SolCast. 
We need to check that on the SolCast API.

davidusb-geek, 2024-05-18:
Hi, just solved this for out 3 methods available inside EMHASS.
This will be available in the next release.

Here is the result for `delta_forecast=2`:

[image]

davidusb-geek, 2024-06-02:
Solved in the just released new version. 
Closing this for now but reopen if eneded.

closed 2024-06-02 by davidusb-geek