davidusb-geek/emhass #286: Docker not starting, web_server.py: error: unrecognized arguments: python3 src/emhass/web_server.py

smitterer, 2024-05-15:
**Describe the bug**
Getting the following errors after downloading the newest version of EMHASS:
2024-05-15T09:37:13.923695795Z | stdout | web_server.py: error: unrecognized arguments: python3 src/emhass/web_server.py
2024-05-15T09:37:13.923564580Z | stdout | [--no_response NO_RESPONSE]
2024-05-15T09:37:13.922609527Z | stdout | usage: web_server.py [-h] [--url URL] [--key KEY] [--addon ADDON]

**To Reproduce**
My config:
# Configuration file for EMHASS

retrieve_hass_conf:
  - freq: 30 # The time step to resample retrieved data from hass in minutes
  - days_to_retrieve: 30 # We will retrieve data from now and up to days_to_retrieve days
  - var_PV: 'sensor.power_ac_fronius_inverter_1_http_192_168_0_140' # Photovoltaic produced power sensor in Watts
  - var_load: 'sensor.power_load_without_variable_loads' # Household power consumption sensor in Watts (deferrable loads should be substracted)
  - load_negative: False # Set to True if the retrived load variable is negative by convention
  - set_zero_min: True # A special treatment for a minimum value saturation to zero. Values below zero are replaced by nans
  - var_replace_zero: # A list of retrived variables that we would want  to replace nans with zeros
    - 'sensor.power_ac_fronius_inverter_1_http_192_168_0_140'
  - var_interp: # A list of retrived variables that we would want to interpolate nan values using linear interpolation
    - 'sensor.power_ac_fronius_inverter_1_http_192_168_0_140'
    - 'sensor.power_load_without_variable_loads'
  - method_ts_round: 'first' # Set the method for timestamp rounding, options are: first, last and nearest

optim_conf:
  - set_use_battery: True # consider a battery storage
  - delta_forecast: 1 # days
  - num_def_loads: 1
  - P_deferrable_nom: # Watts
    - 643.0
  - def_total_hours: # hours
    - 10
  - treat_def_as_semi_cont: # treat this variable as semi continuous 
    - True
  - set_def_constant: # set as a constant fixed value variable with just one startup for each 24h
    - False
  - weather_forecast_method: 'scrapper' # options are 'scrapper' and 'csv'
  - load_forecast_method: 'naive' # options are 'csv' to load a custom load forecast from a CSV file or 'naive' for a persistance model
  - load_cost_forecast_method: 'hp_hc_periods' # options are 'hp_hc_periods' for peak and non-peak hours contracts and 'csv' to load custom cost from CSV file 
  - list_hp_periods: # list of different tariff periods (only needed if load_cost_forecast_method='hp_hc_periods')
    - period_hp_1:
      - start: '00:00'
      - end: '23:59'
  - load_cost_hp: 0.3 # peak hours load cost in €/kWh (only needed if load_cost_forecast_method='hp_hc_periods')
  - load_cost_hc: 0.3 # non-peak hours load cost in €/kWh (only needed if load_cost_forecast_method='hp_hc_periods')
  - prod_price_forecast_method: 'constant' # options are 'constant' for constant fixed value or 'csv' to load custom price forecast from a CSV file
  - prod_sell_price: 0.25 # power production selling price in €/kWh (only needed if prod_price_forecast_method='constant')
  - set_total_pv_sell: False # consider that all PV power is injected to the grid (self-consumption with total sell)
  - lp_solver: 'PULP_CBC_CMD' # set the name of the linear programming solver that will be used
#  - lp_solver: 'COIN_CMD' # set the name of the linear programming solver that will be used
#  - lp_solver_path: '/usr/bin/cbc' # set the path to the LP solver

plant_conf:
  - P_grid_max: 15000 # The maximum power that can be supplied by the utility grid in Watts
  - module_model:
      - 'JA_Solar_JAM72S01_385_PR' #- 'CSUN_Eurasia_Energy_Systems_Industry_and_Trade_CSUN295_60M' # The PV module model
  - inverter_model:
      - 'Huawei_Technologies_Co___Ltd___SUN2000_5KTL_USL0__240V_' #- 'Fronius_International_GmbH__Fronius_Primo_5_0_1_208_240__240V_' # The PV inverter model
  - surface_tilt:
      - 45 # The tilt angle of your solar panels
  - surface_azimuth:
      - 0 # The azimuth angle of your PV installation
  - modules_per_string:
      - 8 # The number of modules per string
  - strings_per_inverter:
      - 2 # The number of used strings per inverter
  - Pd_max: 4000 # If your system has a battery (set_use_battery=True), the maximum discharge power in Watts
  - Pc_max: 4000 # If your system has a battery (set_use_battery=True), the maximum charge power in Watts
  - eta_disch: 0.95 # If your system has a battery (set_use_battery=True), the discharge efficiency
  - eta_ch: 0.95 # If your system has a battery (set_use_battery=True), the charge efficiency
[... 61 lines cut ...]
  - Pd_max: 4000 # If your system has a battery (set_use_battery=True), the maximum discharge power in Watts
  - Pc_max: 4000 # If your system has a battery (set_use_battery=True), the maximum charge power in Watts
  - eta_disch: 0.95 # If your system has a battery (set_use_battery=True), the discharge efficiency
  - eta_ch: 0.95 # If your system has a battery (set_use_battery=True), the charge efficiency
  - Enom: 6400 # If your system has a battery (set_use_battery=True), the total capacity of the battery stack in Wh
  - SOCmin: 0.1 # If your system has a battery (set_use_battery=True), the minimun allowable battery state of charge
  - SOCmax: 0.95 # If your system has a battery (set_use_battery=True), the minimun allowable battery state of charge
  - SOCtarget: 0.6 # If your system has a battery (set_use_battery=True), the desired battery state of charge at the end of each optimization cycle

**Expected behavior**
A starting Docker container

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Home Assistant installation type**
 - Home Assistant Core (Docker)

**Your hardware**
- OS: Docker on Synology

**EMHASS installation type**
 - Docker Standalone

**Additional context**

davidusb-geek, 2024-05-18:
Please share the command that you are using to lauch the docker container.
What is your `docker run` command?

smitterer, 2024-05-25:
Hi, honestly speaking I don't know the exact one as I'm just starting it via the Synology Docker app.
I just set the TZ and LOCAL_COSTFUN variable.

davidusb-geek, 2024-10-29:
Closing this as stale by lack of information. Feel free to reopen if needed.

closed 2024-10-29 by davidusb-geek