davidusb-geek/emhass #287: Error while using forecast.solar (free version)

RikBast, 2024-05-16:
While using forecast.solar by selecting it in the configuration of EMHASS I get the following error.
Running latest version, but also happens on v0.8.6

2024-05-16 11:23:31,716 - web_server - INFO - Passed runtime parameters: {'load_cost_forecast': [14.087, 13.5443, 13.6566, 13.8305, 16.1317, 19.2142, 20.7035, 22.8564, 25.1905, 23.6503, 22.2076, 20.907], 'prod_price_forecast': [-0.257, -0.769, -0.663, -0.499, 1.672, 4.58, 5.985, 8.016, 10.218, 8.765, 7.404, 6.177], 'load_power_forecast': [218, 200, 200, 300, 200, 200, 300, 300, 200, 200, 300, 200, 1400, 300, 400, 400, 300, 400, 200, 300, 400, 300, 600, 400, 500, 500, 400, 500, 500, 200, 500, 600, 300, 200, 200, 500, 500, 400, 300, 300, 200, 200, 200, 300, 300, 300, 200, 400], 'prediction_horizon': 12, 'alpha': 1, 'beta': 0, 'soc_init': 0.44, 'soc_final': 0.12}
2024-05-16 11:23:31,716 - web_server - INFO -  >> Setting input data dict
2024-05-16 11:23:31,716 - web_server - INFO - Setting up needed data
2024-05-16 11:23:31,743 - web_server - INFO - Retrieve hass get data method initiated...
2024-05-16 11:23:32,005 - web_server - INFO - Retrieving weather forecast data using method = solar.forecast
2024-05-16 11:23:32,204 - web_server - ERROR - Exception on /action/naive-mpc-optim [POST]
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 1463, in wsgi_app
    response = self.full_dispatch_request()
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 872, in full_dispatch_request
    rv = self.handle_user_exception(e)
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 870, in full_dispatch_request
    rv = self.dispatch_request()
         ^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/flask/app.py", line 855, in dispatch_request
    return self.ensure_sync(self.view_functions[rule.endpoint])(**view_args)  # type: ignore[no-any-return]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/web_server.py", line 108, in action_call
    input_data_dict = set_input_data_dict(config_path, str(data_path), costfun,
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/command_line.py", line 125, in set_input_data_dict
    df_weather = fcst.get_weather_forecast(method=optim_conf['weather_forecast_method'])
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/emhass/forecast.py", line 278, in get_weather_forecast
    data_dict = {'ts':list(data_raw['result']['watts'].keys()), 'yhat':list(data_raw['result']['watts'].values())}
                           ~~~~~~~~~~~~~~~~~~^^^^^^^^^
TypeError: string indices must be integers, not 'str'
2024-05-16 11:23:32,223 - web_server - INFO - Passed runtime parameters: {}
2024-05-16 11:23:32,223 - web_server - INFO -  >> Setting input data dict
2024-05-16 11:23:32,223 - web_server - INFO - Setting up needed data
2024-05-16 11:23:32,225 - web_server - INFO -  >> Publishing data...
2024-05-16 11:23:32,225 - web_server - INFO - Publishing data to HASS instance
2024-05-16 11:23:32,238 - web_server - INFO - Successfully posted to sensor.p_pv_forecast = -5.91
2024-05-16 11:23:32,242 - web_server - INFO - Successfully posted to sensor.p_load_forecast = 300
2024-05-16 11:23:32,246 - web_server - INFO - Successfully posted to sensor.p_deferrable0 = 0.0
2024-05-16 11:23:32,249 - web_server - INFO - Successfully posted to sensor.p_deferrable1 = 0.0
2024-05-16 11:23:32,252 - web_server - INFO - Successfully posted to sensor.p_batt_forecast = 305.91
2024-05-16 11:23:32,256 - web_server - INFO - Successfully posted to sensor.soc_batt_forecast = 12.0
2024-05-16 11:23:32,259 - web_server - INFO - Successfully posted to sensor.p_grid_forecast = 0.0
2024-05-16 11:23:32,261 - web_server - INFO - Successfully posted to sensor.total_cost_fun_value = 5.7
2024-05-16 11:23:32,264 - web_server - INFO - Successfully posted to sensor.optim_status = Optimal
2024-05-16 11:23:32,267 - web_server - INFO - Successfully posted to sensor.unit_load_cost = 20.0081
2024-05-16 11:23:32,270 - web_server - INFO - Successfully posted to sensor.unit_prod_price = 5.329

davidusb-geek, 2024-05-18:
Hi,
I've just tested everything and unfortunately I'm not able to reproduce your error.
I tested our three methods, they are working perfectly well.

I've coded a small script to test all PV forecast methods at once.
You can test it yourself if you want to check that you are able to retrieve all forecasts.
The script is here: [https://github.com/davidusb-geek/emhass/blob/master/scripts/script_debug_forecasts.py](https://github.com/davidusb-geek/emhass/blob/master/scripts/script_debug_forecasts.py)
To do this clone the EMHASS repo and launch the script.
You will need to defined the needed variables for `solar.forecast` and `solcast` methods in a secrets.yaml file.

Here is the output of the script for my case, with no errors whatsoever:
[image]

So your problem is elsewhere.
Probably a API number of calls limitation?

RikBast, 2024-05-20:
Hi, yes it seems to be the API call, I tested it. First 2 calls ok then I got the error again. 5 min MPC calls.

closed 2024-06-02 by davidusb-geek