davidusb-geek/emhass #299: Add: Huawei SUN2000-3/4/5/6/8/10KTL-M1 (High Current Version) into inverter database

heinemannj, 2024-06-04:
Please add "Huawei SUN2000-3/4/5/6/8/10KTL-M1 (High Current Version)" into inverter database.

[Datasheet_SUN2000-3-10KTL-M1_EN_High-current.pdf](https://github.com/user-attachments/files/15550619/Datasheet_SUN2000-3-10KTL-M1_EN_High-current.pdf)

I have the SUN2000-6KTL-M1 version. But the others are also really common devices.

davidusb-geek, 2024-06-04:
Hi. No this is not possible. It is not up to me. That is a database maintained by the people developing PVLib. What you can do is not focus on finding the exact model. Search for "6kW" on the webapp and pick the result from there.

heinemannj, 2024-06-05:
I have selected :
Huawei_Technologies_Co___Ltd___SUN2000_7_6KTL_USL0__240V_

The inverter output rate is obviously to high, but by using solCast solar prediction there was NO impact in the past.

But I'm afraid that the introduction of
- P_hybrid_inverter
- P_PV_curtailment

will change the game.

Releases changelog but nothing found in the actual documentation:

- Fixed PV curtailment maximum possible value constraint
- Added PV curtailement to variable to publish to HA
- Added support for hybrid inverters

Is it possible to inject  custom inverter values into PVLib module?

gieljnssns, 2024-06-06:
I've found something to calculate the values needed
https://pvpmc.sandia.gov/modeling-guide/dc-to-ac-conversion/sandia-inverter-model/

closed 2024-07-13 by davidusb-geek