davidusb-geek/emhass #302: Documentation update to help find compatible modules and panels

ABredhauer, 2024-06-05:
Is it possible to update the docs to provide some more guidance on where to get panel and inverter info and which values are the important ones to use when you can’t find your components in the database and need to find alternative ones. 

I cannot find my inverter or panels in the db despite them being very popular here in Australia. I’m not sure what all the values mean and which ones are important to use when I try to find something close to what I have.

davidusb-geek, 2024-06-08:
You are absolutely right. Improvements needed to document this. I will try to tackle this soon. Otherwise contributions are welcomed

davidusb-geek, 2024-06-09:
There is some info here: https://emhass.readthedocs.io/en/latest/config.html#system-configuration-parameters

But probably need to put that info somewhere else? Or improve it?

davidusb-geek, 2024-07-06:
Added some more info about this on the README

closed 2024-07-06 by davidusb-geek