davidusb-geek/emhass #308: Sensor unit: Unit is missing for 'Total cost function value'

heinemannj, 2024-06-10:
**Sensor unit is missing** for 'Total cost function value':

**sensor.total_cost_fun_value**

heinemannj, 2024-06-15:
In addition please change
`icon: mdi:currency-eur`

davidusb-geek, 2024-10-29:
Hi, this is complete. You may find the `sensor.total_cost_fun_value` published after a data publish call.

heinemannj, 2024-10-29:
I can't confirm.
Installed latest Add-On version (0.11.1):

[image]

In addition the following sonsors are not visible in logs after upgrade:

- sensor.p_pv_curtailment
- sensor.p_hybrid_inverter

```
2024-10-29 16:08:00,309 - web_server - INFO - Passed runtime parameters: {'pv_power_forecast': [36.0, 22, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 18, 66, 137, 216, 284, 399, 533, 684, 852, 927, 912, 904, 919, 845, 702, 491, 248, 147, 56, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 'load_cost_forecast': [0.45270000000000005, 0.45275, 0.53971, 0.53971, 0.50596, 0.50596, 0.44216, 0.44216, 0.36692, 0.36692, 0.34723, 0.34723, 0.34013, 0.34013, 0.32229, 0.32229, 0.31946, 0.31946, 0.31923, 0.31923, 0.31642, 0.31642, 0.31629, 0.31629, 0.31928, 0.31928, 0.32613, 0.32613, 0.34662, 0.34662, 0.36583, 0.36583, 0.36491, 0.36491, 0.34758, 0.34758, 0.33646, 0.33646, 0.33013, 0.33013, 0.32162, 0.32162, 0.32688, 0.32688, 0.33207, 0.33207, 0.34974, 0.34974, 0.39001, 0.39001, 0.43033, 0.43033, 0.42915, 0.42915, 0.38934, 0.38934, 0.35855, 0.35855, 0.34559, 0.34559, 0.33687, 0.33687, 0.33333], 'prediction_horizon': 47, 'soc_init': 0.47, 'soc_final': 1.0, 'def_total_hours': [0, 0], 'alpha': 1, 'beta': 0, 'continual_publish': False}
2024-10-29 16:08:00,309 - web_server - INFO -  >> Setting input data dict
2024-10-29 16:08:00,309 - web_server - INFO - Setting up needed data
2024-10-29 16:08:00,310 - web_server - INFO - Retrieve hass get data method initiated...
2024-10-29 16:08:02,679 - web_server - INFO - Retrieving weather forecast data using method = list
2024-10-29 16:08:02,680 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2024-10-29 16:08:02,680 - web_server - INFO - Retrieve hass get data method initiated...
2024-10-29 16:08:13,440 - web_server - INFO -  >> Performing naive MPC optimization...
2024-10-29 16:08:13,441 - web_server - INFO - Performing naive MPC optimization
2024-10-29 16:08:13,444 - web_server - INFO - Perform an iteration of a naive MPC controller
2024-10-29 16:08:13,489 - web_server - WARNING - Solver default unknown, using default
Welcome to the CBC MILP Solver 
Version: 2.10.3 
Build Date: Dec 15 2019 

command line - /usr/local/lib/python3.11/dist-packages/pulp/solverdir/cbc/linux/64/cbc /tmp/f13908fff73041f7839c2e778e6457ca-pulp.mps -max -timeMode elapsed -branch -printingOptions all -solution /tmp/f13908fff73041f7839c2e778e6457ca-pulp.sol (default strategy 1)
At line 2 NAME          MODEL
At line 3 ROWS
At line 664 COLUMNS
At line 6726 RHS
At line 7386 BOUNDS
At line 7904 ENDATA
Problem MODEL has 659 rows, 423 columns and 5591 elements
Coin0008I MODEL read with 0 errors
Option for timeMode changed from cpu to elapsed
Continuous objective value is -7539.85 - 0.00 seconds
Cgl0003I 0 fixed, 0 tightened bounds, 83 strengthened rows, 0 substitutions
Cgl0003I 0 fixed, 0 tightened bounds, 5 strengthened rows, 0 substitutions
Cgl0003I 0 fixed, 0 tightened bounds, 3 strengthened rows, 0 substitutions
Cgl0003I 0 fixed, 0 tightened bounds, 2 strengthened rows, 0 substitutions
Cgl0004I processed model has 267 rows, 224 columns (65 integer (65 of which binary)) and 4857 elements
Cbc0038I Initial state - 10 integers unsatisfied sum - 0.620673
Cbc0038I Pass   1: suminf.    0.44892 (8) obj. 7551.18 iterations 26
Cbc0038I Solution found of 7551.18
Cbc0038I Relaxing continuous gives 7539.85
Cbc0038I Before mini branch and bound, 55 integers at bound fixed and 109 continuous
Cbc0038I Mini branch and bound did not improve solution (0.01 seconds)
Cbc0038I After 0.02 seconds - Feasibility pump exiting with objective of 7539.85 - took 0.00 seconds
Cbc0012I Integer solution of 7539.8544 found by feasibility pump after 0 iterations and 0 nodes (0.02 seconds)
Cbc0001I Search completed - best objective 7539.854401763731, took 0 iterations and 0 nodes (0.02 seconds)
Cbc0035I Maximum depth 0, 0 variables fixed on reduced cost
Cuts at root node changed objective from 7539.85 to 7539.85
Probing was tried 0 times and created 0 cuts of which 0 were active after adding rounds of cuts (0.000 seconds)
Gomory was tried 0 times and created 0 cuts of which 0 were active after adding rounds of cuts (0.000 seconds)
Knapsack was tried 0 times and created 0 cuts of which 0 were active after adding rounds of cuts (0.000 seconds)
Clique was tried 0 times and created 0 cuts of which 0 were active after adding rounds of cuts (0.000 seconds)
MixedIntegerRounding2 was tried 0 times and created 0 cuts of which 0 were active after adding rounds of cuts (0.000 seconds)
FlowCover was tried 0 times and created 0 cuts of which 0 were active after adding rounds of cuts (0.000 seconds)
TwoMirCuts was tried 0 times and created 0 cuts of which 0 were active after adding rounds of cuts (0.000 seconds)
ZeroHalf was tried 0 times and created 0 cuts of which 0 were active after adding rounds of cuts (0.000 seconds)

Result - Optimal solution found

Objective value:                -7539.85440176
Enumerated nodes:               0
Total iterations:               0
Time (CPU seconds):             0.02
Time (Wallclock seconds):       0.02

Option for printingOptions changed from normal to all
Total time (CPU seconds):       0.02   (Wallclock seconds):       0.02

2024-10-29 16:08:13,525 - web_server - INFO - Status: Optimal
2024-10-29 16:08:13,525 - web_server - INFO - Total value of the Cost function = -7539.85
[... cut ...]
2024-10-29 16:08:13,772 - web_server - INFO - Successfully posted to sensor.p_grid_forecast = 0.0
2024-10-29 16:08:13,833 - web_server - INFO - Successfully posted to sensor.total_cost_fun_value = -7.54
2024-10-29 16:08:13,837 - web_server - INFO - Successfully posted to sensor.optim_status = Optimal
2024-10-29 16:08:13,842 - web_server - INFO - Successfully posted to sensor.unit_load_cost = 0.4527
2024-10-29 16:08:13,849 - web_server - INFO - Successfully posted to sensor.unit_prod_price = 0.08
```

[image]

davidusb-geek, 2024-10-29:
You can't confirm but the sensor is there?

For `sensor.p_pv_curtailment` and `sensor.p_hybrid_inverter` you may need to reactivate those options in the new webui configurator

heinemannj, 2024-10-29:
The `sensor.total_cost_fun_value` is available but without fix.

`"... For sensor.p_pv_curtailment and sensor.p_hybrid_inverter you may need to reactivate those options in the new webui configurator ..."`

Done after painful reconfigurations from the sratch ...
No old conf parameters have been take over to the new config ...

[image]
[image]
[image]

After reload of Add-On sensor.p_pv_curtailment and sensor.p_hybrid_inverter are available - strange
After reload of HA in general seems to be ok as well

davidusb-geek, 2024-10-29:
> The sensor.total_cost_fun_value is available but without fix.

What do you mean with this?

davidusb-geek, 2024-10-29:
Oh the unit of measurement is missing?
Well there is already an issue to address this >> #155