davidusb-geek/emhass #309: load_cast_forecast nordpool fails Hello,

renaatdb, 2024-06-10:


JohanAlvedal, 2024-06-12:
Got it working here. 
[https://github.com/JohanAlvedal/homeautomation/tree/main/Emhass/Shell](url)

renaatdb, 2024-06-12:
> https://github.com/JohanAlvedal/homeautomation/tree/main/Emhass/Shell

Thanks, 
I'm going to study it

purcell-lab, 2024-06-13:
There is no doubt adding in  these additional forecasts with some json foo can get quite complex quickly.

It would be useful if we could provide a blueprint/ template for some of the standard suppliers; nordpool, Amber, octopus.  So the end user could just tick a check box, rather than having all this yaml fun.

The other consideration is each time you modify the shell_command or rest_command you need to restart HA, it would also reduce complexity if you could just reload a template/ script each time you needed to modify the payload.

JohanAlvedal, 2024-06-14:
Absolutely, I agree with you. Someone talented who could create a blueprint so you can enter your electricity supplier solcast and the most relevant parts, I think it would help many users of emhass. It was/is the part that many ausers struggle with to get emhass up and running, because it is precisely that part that is extremely complex. When and how should all scripts be run and how should they be configured.

renaatdb, 2024-06-14:
Thanks, I thought I was the only one having trouble with this.  It's amazing how Emhass was made. However, it is difficult for a beginner to get started and difficult to understand the manual. 
Hopefully I'll manage to get everything working in the future.
I want to purchase a battery in the future and see what a dynamic rate (with Nordpool) can teach me.

renaatdb, 2024-06-18:
Hello,
This is also an interesting tool from Thermia .
[](https://classiconlinestorage.blob.core.windows.net/web-resources/SmartPrice_Info_and_FAQ_NL.pdf)]

davidusb-geek, 2024-10-29:
Closing this specific issue.
Feel free to open another one if needed.