davidusb-geek/emhass #310: pv_forecast is not aligning with JSON input for naive-mpc-optim command

Deztak, 2024-06-11:
EMHASS is returning a pv_forecast that doesn't align with the input data. Currently it's outputting 2,270.57 W and I can't see any reason why it's not outputting 839W ... all other output time intervals are aligned with the input

Also note that it seems to be jumping around at random over time and I can't reverse engineer what it's doing.

I am using the Hybrid Inverter option but have no PV curtailment

Input for the naive-mpc-optim command
{ "pv_power_forecast":[839, 1115, 1576, 1892, 1981, 2149, 2366, 2524, 2566, 2481, 2310, 2118, 1910, 1636, 1316, 971, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],"load_cost_forecast":[0.13, 0.13, 0.16, 0.16, 0.15, 0.1, 0.1, 0.12, 0.12, 0.15, 0.14, 0.41, 0.42, 0.41, 0.42, 0.49, 0.52, 0.69, 0.69, 0.72, 0.81, 0.69, 0.69, 0.43, 0.4, 0.33, 0.24, 0.33, 0.24, 0.37, 0.34, 0.34, 0.28, 0.22, 0.21, 0.19],"prod_price_forecast":[0.03, 0.04, 0.06, 0.04, 0.03, -0.01, -0.01, 0.01, 0.01, 0.03, 0.02, 0.32, 0.33, 0.32, 0.33, 0.39, 0.42, 0.58, 0.58, 0.61, 0.68, 0.58, 0.58, 0.31, 0.28, 0.22, 0.13, 0.22, 0.14, 0.25, 0.23, 0.22, 0.17, 0.11, 0.11, 0.09],"prediction_horizon":36,"soc_init":0.53,"soc_final":0.15 }

[image]

[image]

Deztak, 2024-06-11:
I just changed my pv_forecast input algorithm to make the first value 0 and it's still giving back 2022.72 W

[image]

Deztak, 2024-06-12:
Now it's giving a massive error but the second it ticked over 1131, the error reduced to less than 100 and has now increased over the time period.

Data Input
{ "pv_power_forecast":[2418, 2456, 2371, 2290, 2214, 2030, 1752, 1441, 1135, 895, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0],"load_cost_forecast":[0.04, 0.04, 0.1, 0.1, 0.1, 0.39, 0.4, 0.41, 0.42, 0.48, 0.47, 0.52, 0.69, 0.58, 0.54, 0.53, 0.58, 0.31, 0.23, 0.26, 0.23, 0.37, 0.21, 0.37, 0.27, 0.24, 0.21, 0.21, 0.19, 0.18, 0.18, 0.18, 0.18],"prod_price_forecast":[-0.07, -0.07, -0.01, -0.01, -0.01, 0.31, 0.32, 0.32, 0.33, 0.38, 0.38, 0.42, 0.58, 0.48, 0.44, 0.43, 0.48, 0.2, 0.13, 0.15, 0.13, 0.25, 0.11, 0.26, 0.16, 0.14, 0.11, 0.11, 0.09, 0.08, 0.08, 0.08, 0.08],"prediction_horizon":33,"soc_init":0.64,"soc_final":0.15 }

[image]

purcell-lab, 2024-06-12:
Please check your *alpha* and *beta* values as well as the *method_ts_round*:
https://emhass.readthedocs.io/en/latest/forecasts.html#now-current-values-in-forecasts
https://emhass.readthedocs.io/en/latest/config.html#retrieve-hass-data-configuration

I have:

*method_ts_round* set to first.
        "alpha": 1,
        "beta": 0,

Deztak, 2024-06-23:
> Please check your _alpha_ and _beta_ values as well as the _method_ts_round_: https://emhass.readthedocs.io/en/latest/forecasts.html#now-current-values-in-forecasts [...]

Thankyou ... I don't know where this thread can be moved to to help others ... but, if anyone else has this issue here is my rest command change that solved it:

`post_mpc_optim_solcast:
  url: http://homeassistant.local:5000/action/naive-mpc-optim
  method: POST
  content_type: 'application/json'
  payload: >-
    { "pv_power_forecast":{{states('sensor.solcast_24hrs_forecast')}},"load_cost_forecast":{{states('sensor.amber_gen_price_forecast')}},"prod_price_forecast":{{states('sensor.amber_prod_price_forecast')}},"prediction_horizon":{{states('sensor.solcast_24hrs_forecast')|from_json|count}},"soc_init":{{(states('sensor.med_battery_charge')|float(0)/100)|round(2)}},"soc_final":{{float(0.15)|round(2)}},"alpha":{{float(1)}},"beta":{{float(0)}} }`

With this rest command, I'm now getting:
[image]

if alpha = 0 and beta = 1 then this is then I got the below output for this input
[image]

closed 2024-06-23 by Deztak