davidusb-geek/emhass #312: Unable to connect to home assistant

BeFygo, 2024-06-12:
**Describe the bug**
I'm unable to get data from home assistant with the Optimization request.

**To Reproduce**
Home assistant docker en EMHASS docker setup with docker compose both in Network Host mode.

**Expected behavior**
That EMHASS would request data from Home Assistant

**Screenshots**
[image]

**Home Assistant installation type**
 - Home Assistant Docker

**Your hardware**
- OS: Linux
- Architecture: (Odroid n2+) armv7

**EMHASS installation type**
 - Docker Standalone

**Additional context**
Secret_emhass.yaml:
hass_url: http://localhost:8123
long_lived_token: **correct**
time_zone: Europe/Brussels
lat: *****
lon: ****
alt: 24.4

Docker compose:
version: "3.8"
services:

  homeassistant2:
    image: ghcr.io/home-assistant/home-assistant:stable
    container_name: homeAssistant
    environment:
      - TZ=Europe/Brussels

    volumes:
       - /****/Homeassistant:/config
       - /var/run/dbus/system_bus_socket:/var/run/dbus/system_bus_socket

    privileged: true
    network_mode: host

    restart: unless-stopped

  emhass:
    image: davidusb/emhass-docker-standalone:latest
    container_name: emhass
    environment:
      - TZ=Europe/Brussels
      - LOCAL_COSTFUN="profit"
    volumes:
       - /****/Config/EMHASS:/app/
       - /****/Config/EMHASS/data:/app/data

#    ports:
#      - 5000:5000
         
#    privileged: true
    network_mode: host

    restart: unless-stopped

GeoDerp, 2024-06-13:
I don't know if this helps but I tried:

Home Assistant
```bash
# modified version of: https://www.home-assistant.io/installation/linux#install-home-assistant-container
docker run -d --name homeassistant -e TZ="Europe/Paris" -v ~/test:/config -p 8123:8123 ghcr.io/home-assistant/home-assistant:stable
```
EMHASS
```bash
# copied config_emhass.yaml and secrets_emhass.yaml (from secrets_emhass(copy).yaml) into /test2/ 
docker run -it -p 5000:5000 --network=host --name emhass-container -v $(pwd)/test2/config_emhass.yaml:/app/config_emhass.yaml -v $(pwd)/test2/secrets_emhass.yaml:/app/secrets_emhass.yaml davidusb/emhass-docker-standalone:latest
```
Note: It is my understanding that the EMHASS container would need to have `--network=host` in order to access `localhost`. Its my understanding that by default docker runs its networking in bridge mode. So an alternative method would be to obtain the Home Assistant bridged ip address and pass that into the EMHASS container

An example:
```bash
# grab the ip address of the running homeassistant ip address
docker exec -it homeassistant ip addr
```
in my case it was `172.17.0.3` with `eth0@if23`, then pass this ip address into the EMHASS secrets.

BeFygo, 2024-06-13:
Thank you for the help.

I tried your suggestions and in the end it was just the foward slash `/`  at the end of the address :D

closed 2024-06-13 by BeFygo