davidusb-geek/emhass #315: Feature Request: Pd_max/Pc_max as runtime parameters

anline380, 2024-06-21:
HI David ,
I have been using EMHASS since the first version and I am happy and grateful for how this project has developed over the last few years. Currently EMHASS runs great for me as HA Add-On, but I think there is one relevant feature missing for home storage systems (I hope I haven't overlooked anything).

Would it be possible to implement Pd_max: The maximum discharge power and Pc_max: The maximum charge power as runtime parameters (The greatest thing would be as a timeslot/array). A very relevant application scenario would be, for example, to load the Home Storage gently or the Spot Price Market. I am currently setting the Solaredge - Storage Charge Limit from a maximum of 5000W to 1500W via HA. With EMHASS you would now have to set Pd_max (currently only via config) to 1500 during the runtime so that the System know whether the desired SOC, e.g. 100, is achieved at the end of the day with max 1500W charging power. If the Home Storage needs to be charged quickly, e.g. bad weather or hours with a low spot price, you can set Pd_max/Pc_max back to 5000 using the runtime parameter.

 Thanks in advance, Manuel

davidusb-geek, 2024-07-04:
Hi, yes this can be added to the set of parameters that could be set at runtime.
I will try to add this to the following release.

anline380, 2024-07-05:
That would be really great. I think it would be a game changer not just for me. Thank you very much in advance, Manuel🤗

anline380, 2024-07-06:
A big Thank you for the  very quick implementation, the update works excellent.👍👍

closed 2024-07-06 by anline380