davidusb-geek/emhass #317: 400 error on call

smitterer, 2024-06-26:
**Describe the bug**
I get a 400 error when doing a call using:

{"pv_power_forecast":[3374.8,3245.1,3005.7999999999997,2680.8999999999996,2331.2,1985.1000000000001,1598.6,1236.5,875.1999999999999,562.9,343.5,195.1,85.5,13,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0], "prod_price_forecast":[0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04,0.04], "load_cost_forecast":[0.186586,0.186586,2.238408,2.238408,0.944180,0.944180,0.231886,0.231886,0.182804,0.182804,0.041332,0.041332,0.041468,0.041468,0.040380,0.040380,0.040911,0.040911,0.039008,0.039008,0.214656,0.214656,0.203334,0.203334,0.192692,0.192692,0.195807,0.195807,0.190653,0.190653], "prediction_horizon":30, "def_total_hours":[5.00], "soc_init":0.97, "soc_final":0.9}

My conf:
retrieve_hass_conf:
  - freq: 30 # The time step to resample retrieved data from hass in minutes
  - days_to_retrieve: 30 # We will retrieve data from now and up to days_to_retrieve days
  - var_PV: 'sensor.pv_power' # Photovoltaic produced power sensor in Watts
  - var_load: 'sensor.power_load_without_variable_loads' # Household power consumption sensor in Watts (deferrable loads should be subtracted)
  - load_negative: False # Set to True if the retrieved load variable is negative by convention
  - set_zero_min: True # A special treatment for a minimum value saturation to zero. Values below zero are replaced by NaNs
  - var_replace_zero: [] # A list of retrieved variables that we would want to replace NaNs with zeros (should be a list even if empty)
  - var_interp: # A list of retrieved variables that we would want to interpolate NaN values using linear interpolation
    - 'sensor.power_load_without_variable_loads'
  - method_ts_round: 'first' # Set the method for timestamp rounding, options are: first, last and nearest

optim_conf:
  - set_use_battery: True # consider a battery storage
  - delta_forecast: 1 # days
  - num_def_loads: 1
  - P_deferrable_nom: # Watts
    - 643.0
#  - def_total_hours: # hours
#    - 10
  - treat_def_as_semi_cont: # treat this variable as semi continuous 
    - True
  - set_def_constant: # set as a constant fixed value variable with just one startup for each 24h
    - False
  - weather_forecast_method: 'scrapper' # options are 'scrapper' and 'csv'
  - load_forecast_method: 'naive' # options are 'csv' to load a custom load forecast from a CSV file or 'naive' for a persistence model
  - load_cost_forecast_method: 'hp_hc_periods' # options are 'hp_hc_periods' for peak and non-peak hours contracts and 'csv' to load custom cost from CSV file 
  - list_hp_periods: # list of different tariff periods (only needed if load_cost_forecast_method='hp_hc_periods')
    - period_hp_1:
        start: '00:00'
        end: '23:59'
  - load_cost_hp: 0.3 # peak hours load cost in €/kWh (only needed if load_cost_forecast_method='hp_hc_periods')
  - load_cost_hc: 0.3 # non-peak hours load cost in €/kWh (only needed if load_cost_forecast_method='hp_hc_periods')
  - prod_price_forecast_method: 'constant' # options are 'constant' for constant fixed value or 'csv' to load custom price forecast from a CSV file
  - prod_sell_price: 0.4 # power production selling price in €/kWh (only needed if prod_price_forecast_method='constant')
  - set_total_pv_sell: False # consider that all PV power is injected to the grid (self-consumption with total sell)
  - lp_solver: 'PULP_CBC_CMD' # set the name of the linear programming solver that will be used
  - set_nocharge_from_grid: True # avoid battery charging from the grid
  - set_nodischarge_to_grid: True # avoid battery discharging to the grid
  - set_battery_dynamic: False # add a constraint to limit the dynamic of the battery power in power per time unit
  - battery_dynamic_max: 0.9 # maximum dynamic positive power variation in percentage of battery maximum power
  - battery_dynamic_min: -0.9 # minimum dynamic negative power variation in percentage of battery maximum power
  - weight_battery_discharge: 0.0 # weight applied in cost function to battery usage for discharge
  - weight_battery_charge: 0.0 # weight applied in cost function to battery usage for charge
  
  # lp_solver: 'COIN_CMD' # set the name of the linear programming solver that will be used
  # lp_solver_path: '/usr/bin/cbc' # set the path to the LP solver

plant_conf:
  - P_grid_max: 15000 # The maximum power that can be supplied by the utility grid in Watts
  - module_model:
      - 'JA_Solar_JAM72S01_385_PR' # The PV module model
      # - 'CSUN_Eurasia_Energy_Systems_Industry_and_Trade_CSUN295_60M' # Another PV module model option
  - inverter_model:
      - 'Huawei_Technologies_Co___Ltd___SUN2000_5KTL_USL0__240V_' # The PV inverter model
      # - 'Fronius_International_GmbH__Fronius_Primo_5_0_1_208_240__240V_' # Another PV inverter model option
  - surface_tilt:
      - 45 # The tilt angle of your solar panels
  - surface_azimuth:
      - 0 # The azimuth angle of your PV installation
  - modules_per_string:
      - 8 # The number of modules per string
  - strings_per_inverter:
      - 2 # The number of used strings per inverter
  - Pd_max: 4000 # If your system has a battery (set_use_battery=True), the maximum discharge power in Watts
  - Pc_max: 4000 # If your system has a battery (set_use_battery=True), the maximum charge power in Watts
  - eta_disch: 0.95 # If your system has a battery (set_use_battery=True), the discharge efficiency
  - eta_ch: 0.95 # If your system has a battery (set_use_battery=True), the charge efficiency
  - Enom: 6400 # If your system has a battery (set_use_battery=True), the total capacity of the battery stack in Wh
  - SOCmin: 0.1 # If your system has a battery (set_use_battery=True), the minimum allowable battery state of charge
  - SOCmax: 0.95 # If your system has a battery (set_use_battery=True), the maximum allowable battery state of charge
  - SOCtarget: 0.6 # If your system has a battery (set_use_battery=True), the desired battery state of charge at the end of each optimization cycle

**To Reproduce**
Trigger mpc-optim using the above config

**Expected behavior**
No 400

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Home Assistant installation type**
 - Home Assistant Core

**Your hardware**
- Synology
- 
**EMHASS installation type**
 - Docker

**Additional context**

davidusb-geek, 2024-07-13:
Could you please share the same command that you are using to launch that MPC?
Is it a RESTful command?
Share your template payload to debug

davidusb-geek, 2024-10-29:
Hi, still blocked on this 400 error?

closed 2024-12-24 by davidusb-geek