davidusb-geek/emhass #319: Feature Request: Support dynamic electricity prices

andreas-bulling, 2024-06-27:
It'd be great if EMHASS not only supported fixed electricity prices but also dynamic ones, e.g. provided by the Tibber integration for the next 48 hours.

anline380, 2024-06-30:
Emhass supports dynamic pricing without any problems. You should check the HA forum, there are hundreds of examples.

andreas-bulling, 2024-07-01:
I've seen those solutions but they are all rather hacky/manual solutions IMHO. Would be nicer if one could simply define the (e.g. tibber) entity that provides the prices and EMHASS does the rest.

JohanAlvedal, 2024-07-03:
I think most of them want just that for a supplier, but the probability that it will happen is small. As mentioned earlier in the thread, there are many examples on the HA forum. Personally, I have tibber but runs with nordpool, which is much more stable.
If you want some inspiration, you can check my setup, I'm from Sweden.

[https://github.com/JohanAlvedal/homeautomation/tree/main/Emhass](url)

davidusb-geek, 2024-07-04:
This is not going to happen any time soon. Just because each part of the work have their own prices API and it is just hard to keep track of all of them and keep up with API updates.
This is why we support passing any data as inputs on runtime.
People need to trat that themselves in HA with templating and the help of forum members that provide and exchange their own implementations of this data handling

closed 2024-07-04 by davidusb-geek