davidusb-geek/emhass #323: Solcast shows values at night

a8291, 2024-07-07:
Hey,
When doing a dayahead optimization at noon with configured Solcast EMHASS is showing me PV production for the coming night. Doesn't emhass take into account what time it is or am I doing sth wrong?

The one on the screenshot was triggered just now at 11am.

[image]

GeoDerp, 2024-07-07:
> Hey,
> When doing a dayahead optimization at noon with configured Solcast EMHASS is showing me PV production for the coming night. Doesn't emhass take into account what time it is or am I doing sth wrong?
> 
> The one on the screenshot was triggered just now at 11am.
> [image]
> 

Greetings @a8291 ,
I'm not quite sure 100%. 
Maybe check your timezones on HA and/or the configuration.
If your running EMHASS not in HA (in standalone) you may need to pass in the `-e TZ=Country/State` in the docker run.
Other then that maybe the Solcast rooftop is set up wrong? 
Let me know once you check these thing if they are still wrong and I will have a look at the code.

a8291, 2024-07-07:
I am running as an addon of HA. My time zone settings are correct in HA. The rooftop location of Solcast is correctly setup.

GeoDerp, 2024-07-07:
Sorry forgive me, I have just re-read your message. Looking at the screenshot seems fine by me.
Is it okay if you can elaborate on what "for the coming night" means?
The first element looks like 11:00am to me +2:00 indicating `UTC+2:00`

a8291, 2024-07-07:
After setting the freq to 30 minutes it makes more sense. Now I don't get any PV production when it's dark. That was what I meant with night. Sorry the confusion. So it seems to be related to the frequency. 
[image]

closed 2024-10-29 by davidusb-geek