davidusb-geek/emhass #328: Cannot install add-on

Bascht74, 2024-07-14:
**Describe the bug**
I cannot install the add-on

**To Reproduce**
Try to install the add-on within HA '(2024-07)

**Expected behavior**
The add-on will be installed

**Screenshots**
[image]

**Home Assistant installation type**
 - Home Assistant OS

**Your hardware**
- OS: HA OS
- Architecture: amd64 (Proxmox)

**EMHASS installation type**
 - Add-on

**Additional context**
./.

eddysteurs, 2024-07-14:
same error during upgrade from 0.10.5 to 0.10.6

davidusb-geek, 2024-07-14:
You can try no_, it should be solved

Bascht74, 2024-07-14:
> You can try now, it should be solved

Working. Thx!

closed 2024-07-14 by davidusb-geek

poundy, 2024-07-22:
I'm still seeing this on RPi4/x86 (ARMv7?) when upgrading... 
[image]