davidusb-geek/emhass #330: Same as 328 - missing image update to0.10.6

poundy, 2024-07-22:
**Describe the bug**
repeat of https://github.com/davidusb-geek/emhass/issues/328 for Armv7 (RPi x86)

**To Reproduce**
Attempt to update in HA

**Expected behavior**
Updates normally

**Screenshots**
[image]

**Home Assistant installation type**
 - Home Assistant OS
 - Home Assistant Supervised
 - Home Assistant Core

**Your hardware**
- OS: HA OS
- Architecture: armv7

**EMHASS installation type**
 - Add-on
 
**Additional context**
This has happened before, when there was a blocker on (if I recall) a lib on Armv7 that didn't move forward. That cleared up, and periodically I might see a delay when the build happens, but this has been a while now. I saw the previous bug report and posted there but that didn't reopen so here's a new report.

davidusb-geek, 2024-07-24:
ARM runs are all just failing

poundy, 2024-07-24:
> ARM runs are all just failing

no worries, i suspected it wasn't a transient build failure....

GeoDerp, 2024-07-25:
Will look into some more later. This is interesting tho (on armhf/armv7):
```
#11 27.88 ERROR: Ignored the following versions that require a different python version: 0.2.14 Requires-Python >=3.8, <3.10
#11 27.88 ERROR: Could not find a version that satisfies the requirement emhass==0.10.6 (from versions: 0.1.0, 0.1.1, 0.1.2, 0.1.3, 0.1.4, 0.1.5, 0.2.0, 0.2.1, 0.6.0, 0.6.1, 0.6.2, 0.7.0, 0.7.1, 0.7.2, 0.7.3, 0.7.4, 0.7.5, 0.7.6, 0.7.7, 0.7.8, 0.8.0, 0.8.1, 0.8.2, 0.8.3, 0.8.4, 0.8.5, 0.8.6, 0.9.0, 0.9.1, 0.10.0, 0.10.1, 0.10.3)
#11 27.88 ERROR: No matching distribution found for emhass==0.10.6
```

davidusb-geek, 2024-07-25:
Exactly, that I don't understand, the pypi is published and there. The action ran without issue: https://github.com/davidusb-geek/emhass/actions/runs/9923583868

GeoDerp, 2024-07-25:
@davidusb-geek maybe rerun the release? I'm interested if it was maybe  they PyPI database lagginh behind on its arm repositories.
I'm thinking of transferring from `requirements.txt` to `poetry.lock`. I'm not sure if it will help, but im wondering if a tools such as this is a better method of assuring the correct packages are installed on the correct CPU architectures.

davidusb-geek, 2024-07-25:
I've just done that. The release launch fails because pypi won't accept you to overwrite the same version. But anyway the wheels are there available on pypi, so that's ok. I've just relaunched the add-on release, let's see how it goes

GeoDerp, 2024-07-25:
Just saw the same error on the re-run action... defiantly need to look deeper into the issue.

GeoDerp, 2024-07-25:
Worst case scenario, we could input the EMHASS into EMHASS-add-on as a sub modul. then build the EMHASS package like we do on the standalone docker file. *(pypassing PyPI)*
https://github.blog/open-source/git/working-with-submodules/

davidusb-geek, 2024-07-25:
I like that idea.
I confirm that it is better IMO to keep separate repos for the core code and the add-on.

GeoDerp, 2024-07-25:
> I like that idea.
> I confirm that it is better IMO to keep separate repos for the core code and the add-on.

Alright. When I have some time I'll test to see how doable it is.

davidusb-geek, 2024-10-29:
This is deemed solved with the current version, all platform images were compiled correctly.
Closing as completed

closed 2024-10-29 by davidusb-geek