davidusb-geek/emhass #339: Feature Request: Cost/profit comparison

smefa, 2024-08-14:
Would it be possible to add a cost value that is calculateed without any battery usage as a comparison?
You could use this to see how much you potentially save using battery storage.

davidusb-geek, 2024-09-07:
Yes it can be done and could be interesting. It will need some code tweaking to get it done. I will look into it when I have the time

purcell-lab, 2024-09-07:
I have been doing some work using Riemann sum and utility meters to calculate the savings and cost benefits of both solar and battery.

https://community.home-assistant.io/t/add-solar-cost-saving-option-to-native-energy-dashboard/376143/4?u=markpurcell

davidusb-geek, 2024-10-29:
Yes probably something to do better directly in HA.
Closing

closed 2024-10-29 by davidusb-geek

smefa, 2024-10-29:
I don't see how you could calculate the projected missed savings, only the outcome. Am I wrong?
But anyhow, it's ok to close it since it is a niche request.
[image]

davidusb-geek, 2024-10-29:
I just went to the link provided my Mark and saw that he is able to separetly compute PV saving, Battery saving and PV+battery savings. So the missed savings of adding the battery are just the opposite of the battery savings, right?

smefa, 2024-10-29:
Yes, I guess it would work if you use the battery and show historic values. 
I don't see how it would work for the prediction. But anyhow, we can drop this since it is not that important.

davidusb-geek, 2024-10-29:
Ok.
If you are interested a manual script could also achieve the kind of comparison that you are looking for.
For example this script here does just that, compare different use cases (with and without PV and batteries):
https://github.com/davidusb-geek/emhass/blob/master/scripts/use_cases_analysis.py