davidusb-geek/emhass #343: P_Batt forecast value calculation issue

martinarva, 2024-09-10:
**Describe the bug**

P_Batt forecast value calculation uses eta_disch as a multiplier and this causes issue where p_batt is lower than it should be.

For example at the moment max sell from battery should be happening.

Pd_max: 15600
eta_disch: 0.95

P_batt should be 15600, but it is 15600 * 0.95 =14 820 

[image]

I'm using p_batt in automation to set the discharge power and in this case losing some power.

**Expected behavior**
Would be nice to get the real b_batt value. I think it would be beneficial for utilizing battery the most. Maybe it will be a possibility to make the eta_disch calculation for after the p_batt so that load and grid would be correct.

davidusb-geek, 2024-09-28:
It could be that we need to apply the efficiency factor before publishing the battery power sensor? 
It is something we have missed all along the way.
In the meantime you can also apply this efficiency on HA just before controlling your equipment

martinarva, 2024-09-28:
> It could be that we need to apply the efficiency factor before publishing the battery power sensor? It is something we have missed all along the way. In the meantime you can also apply this efficiency on HA just before controlling your equipment

Great and yes, i'm applying this efficiency in HA at the moment.

purcell-lab, 2024-10-18:
I have been setting discharge efficiency to 1.00 to workaround this issue.

davidusb-geek, 2026-11-03:
Probably solved since

closed 2026-11-03 by davidusb-geek