davidusb-geek/emhass #345: Feature Request: Different efficiency for AC and DC charging/discharging

anton4, 2024-09-18:
### Describe the improvement - 1 ###
It would be nice if we could configure different efficiency multiplier/factor when charging battery from AC and DC (solar)
Charging from AC is less efficient than charging from solar (DC) meaning a lot more energy is being wasted as heat.

This feature would make the model calculation even more accurate.

### Describe the improvement - 1.5 ###
It would be nice if we could set different efficiency factors depending on the battery charging/discharging power.
This depends on the length and diameter of your wires (both AC and DC) 
For example when charging at 2kW, the loss of power is 100W, but when charging at 20kW, the loss of power is 2kW.

### Describe the improvement - 2 ###
The lower your battery charging power is the less power is wasted as heat (wires, inverter internals, etc.).
For example - import price of cheapest electricity is the same for 4 hours straight, there is no solar (winter) and battery charging (0-100%) takes 2 hours if done at max charge rate. 
At the moment emhass model wants to charge the battery with 2 hours, but it would be more efficient if it would spread the charge time to max available hours (4 in this case).

This feature would make emhass more efficient and not waste electricity to heat by not charging at max power when it's not needed.

davidusb-geek, 2024-09-28:
Yes a lot of room for improvement.
Ideally we should also use variable efficiencies as a function of the actual power.
However this will often mean more complex MILP formulations.
But a different constant efficiency for AC and DC sides seems like a reasonable easy thing to do

davidusb-geek, 2026-11-03:
This should be covered with the latests works on efficiency modeling.

closed 2026-11-03 by davidusb-geek