davidusb-geek/emhass #346: 'set_def_constant' sometimes ignored

toggm, 2024-09-26:
**Describe the bug**
I've a setup with three deferrable loads. Last one of the loads should switch on and off as few as possible (heatpump), therefore I set 'set_def_constant' to "True" only for this load.
Even though setting to "True" sometime the load is switched on and off multiple times per day. How can I force emhass to consider this flag?

**To Reproduce**
Running the optimization task with the following parameters results to a result where the deferrable load 3 is swtiched on multiple time:

`curl --data ' {"num_def_loads": 3, "P_deferrable_nom": [7000, 4000, 4000], "def_total_hours": [0.77, 2.0, 3.0], "load_power_forecast": [1316.31, 1520.37, 1598.63, 1375.53, 1306.0, 1472.87, 1421.64, 1206.22, 946.88, 943.89, 1107.6, 1241.93, 1118.27, 1197.83, 1699.59, 1525.99, 1037.86, 1174.13, 1256.38, 1443.41, 1454.0, 1316.15, 947.55, 891.86, 1700.98, 1746.63, 910.81, 922.17, 880.78, 1750.73, 906.84, 891.43, 883.23, 886.12, 892.02, 891.21, 903.67, 901.31, 878.09, 900.96, 947.44, 962.61, 978.55, 933.01, 885.06, 1119.9, 1259.34, 1453.96, 1316.31, 1520.37, 1598.63, 1375.53, 1306.0, 1472.87, 1421.64, 1206.22, 946.88, 943.89, 1107.6, 1241.93, 1118.27, 1197.83, 1699.59, 1525.99, 1037.86, 1174.13, 1256.38, 1443.41, 1454.0, 1316.15, 947.55, 891.86, 1700.98, 1746.63, 910.81, 922.17, 880.78, 1750.73, 906.84, 891.43, 883.23, 886.12, 892.02, 891.21, 903.67, 901.31, 878.09, 900.96, 947.44, 962.61, 978.55, 933.01, 885.06, 1119.9, 1259.34, 1453.96, 1316.31, 1520.37, 1598.63, 1375.53, 1306.0, 1472.87, 1421.64, 1206.22, 946.88, 943.89, 1107.6, 1241.93, 1118.27, 1197.83, 1699.59, 1525.99, 1037.86, 1174.13, 1256.38, 1443.41, 1454.0, 1316.15, 947.55, 891.86, 1700.98, 1746.63, 910.81, 922.17, 880.78, 1750.73, 906.84, 891.43, 883.23, 886.12, 892.02, 891.21, 903.67, 901.31, 878.09, 900.96, 947.44, 962.61, 978.55, 933.01, 885.06, 1119.9, 1259.34, 1453.96], "pv_power_forecast": [898.5, 936.9, 956.0999999999999, 1090.3, 1166.9, 1243.5, 1300.8, 1300.8, 1224.3, 1224.3, 1281.7, 1262.6, 1186.1, 1013.7, 744.7, 404.6, 277.9, 150.2, 61.1, 7.6, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 211.9, 1230.8, 2578.3999999999996, 3926.6, 5378.5, 6559.700000000001, 7449.8, 8526.199999999999, 9503.199999999999, 10047.0, 9951.2, 9674.699999999999, 9335.3, 8811.6, 8261.1, 7655.700000000001, 6962.1, 6357.0, 5863.7, 4268.7, 2209.5, 776.0, 220.9, 24.1, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 66.4, 224.0, 439.7, 1021.7, 1625.8, 2225.6, 2706.1000000000004, 3145.7999999999997, 3545.2, 3791.5, 3884.1, 3867.1000000000004, 3696.8, 3696.8, 3810.4, 3715.8, 3393.3, 2955.0, 2418.1, 1657.4, 835.0, 320.7, 126.6, 8.4, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 160.0, 84.0, 290.1, 764.0], "alpha": "0.25", "beta": "0.75", "treat_def_as_semi_cont": ["True", "True", "False"], "set_def_constant": ["False", "False", "True"], "def_start_penalty": [1, 1, 1], "delta_forecast": 3}'  http://localhost:5000/action/dayahead-optim`

Playing around with the parameters showed, that the result highly depends on 'def_total_hours' of the first deferreable load.

**Expected behavior**
I would expect that setting 'set_def_constant' to "True" would force emhass to calculate a single on state in the result.

**Screenshots**
[image]

**Home Assistant installation type**
 - Home Assistant Core

**EMHASS installation type**
 - Docker Standalone

**Additional context**
Add any other context about the problem here.

toggm, 2024-09-26:
Is it possibly because I define `def_total_hours` with a higher precision than we defined the `freq` config parameter?

davidusb-geek, 2024-09-26:
How are you defining these two parameters?

davidusb-geek, 2024-09-26:
Yes no sense of using 0.77 for a 30min resolution. Use either 0.5 or 1.0

toggm, 2024-10-01:
with rounded values of `def_total_hours`, which are calculated based on the temperatur difference didn't encounter the issue anymore. Therefore closing the issue. Thanks for the support.