davidusb-geek/emhass #347: now/ current values  are not updated in real time - alpha =1

purcell-lab, 2024-09-28:
**Describe the bug**
When alpha = 1 p_load_forecast should reflect the current/ now value of power_load_no_var_loads

However there appears to be a constant variance (calling  high frequency MPC every minute)
[image]

power_load_no_var_loads = 2450 W, but p_load_forecast = 1007W

```
2024-09-28 15:42:30,450 - web_server - INFO - Passed runtime parameters: {'prod_price_forecast': [-0.04, -0.02, 0.05, 0.08, 0.11, 0.12, 0.11, 0.1, 0.1, 0.08, 0.08, 0.08, 0.1, 0.12, 0.1, 0.1, 0.08, 0.08, 0.08, 0.06, 0.06, 0.06, 0.06, 0.06, 0.06, 0.06, 0.08, 0.08, 0.08, 0.06, 0.05, 0.03, -0.02, -0.02, -0.03, -0.04, -0.04, -0.04, -0.04, -0.04, -0.04, -0.04, -0.04, -0.04, -0.04, -0.04, -0.02, -0.02, -0.02], 'load_cost_forecast': [0.05, 0.21, 0.29, 0.33, 0.36, 0.37, 0.36, 0.35, 0.35, 0.33, 0.33, 0.2, 0.22, 0.24, 0.23, 0.22, 0.2, 0.2, 0.2, 0.18, 0.18, 0.18, 0.18, 0.18, 0.18, 0.18, 0.2, 0.2, 0.2, 0.18, 0.17, 0.15, 0.09, 0.09, 0.08, 0.05, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.04, 0.05, 0.05, 0.07, 0.07, 0.07], 'prediction_horizon': 47, 'alpha': 1, 'beta': 0, 'soc_init': 0.961, 'def_start_penalty': [1, 1, 1, 1], 'def_current_state': [0, 0, 0, 0], 'def_start_timestep': [0, 0, 0, 0], 'def_end_timestep': [0, 0, 29, 0], 'def_total_hours': [2, 1, 0, 4], 'soc_final': 0.08}
2024-09-28 15:42:30,450 - web_server - INFO -  >> Setting input data dict
2024-09-28 15:42:30,450 - web_server - INFO - Setting up needed data
2024-09-28 15:42:30,452 - web_server - INFO - Retrieve hass get data method initiated...
2024-09-28 15:42:30,664 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2024-09-28 15:42:32,696 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2024-09-28 15:42:32,697 - web_server - INFO - Retrieve hass get data method initiated...
2024-09-28 15:42:33,340 - web_server - INFO -  >> Performing naive MPC optimization...
2024-09-28 15:42:33,340 - web_server - INFO - Performing naive MPC optimization
2024-09-28 15:42:33,344 - web_server - INFO - Perform an iteration of a naive MPC controller
2024-09-28 15:42:33,651 - web_server - INFO - Status: Optimal
2024-09-28 15:42:33,651 - web_server - INFO - Total value of the Cost function = 3.82
2024-09-28 15:42:33,779 - web_server - INFO - Passed runtime parameters: {}
2024-09-28 15:42:33,779 - web_server - INFO -  >> Setting input data dict
2024-09-28 15:42:33,779 - web_server - INFO - Setting up needed data
2024-09-28 15:42:33,780 - web_server - INFO -  >> Publishing data...
2024-09-28 15:42:33,780 - web_server - INFO - Publishing data to HASS instance
2024-09-28 15:42:33,787 - web_server - INFO - Successfully posted to sensor.p_pv_forecast = 1940.97
2024-09-28 15:42:33,790 - web_server - INFO - Successfully posted to sensor.p_load_forecast = 1007.63
2024-09-28 15:42:33,793 - web_server - INFO - Successfully posted to sensor.p_pv_curtailment = 0.0
2024-09-28 15:42:33,798 - web_server - INFO - Successfully posted to sensor.p_hybrid_inverter = -1876.59
2024-09-28 15:42:33,801 - web_server - INFO - Successfully posted to sensor.p_deferrable0 = 0.0
2024-09-28 15:42:33,804 - web_server - INFO - Successfully posted to sensor.p_deferrable1 = 0.0
2024-09-28 15:42:33,809 - web_server - INFO - Successfully posted to sensor.p_deferrable2 = 0.0
2024-09-28 15:42:33,813 - web_server - INFO - Successfully posted to sensor.p_deferrable3 = 0.0
2024-09-28 15:42:33,818 - web_server - INFO - Successfully posted to sensor.p_batt_forecast = -3817.55
2024-09-28 15:42:33,821 - web_server - INFO - Successfully posted to sensor.soc_batt_forecast = 99.64
2024-09-28 15:42:33,825 - web_server - INFO - Successfully posted to sensor.p_grid_forecast = 2884.22
2024-09-28 15:42:33,829 - web_server - INFO - Successfully posted to sensor.total_cost_fun_value = 4.0
2024-09-28 15:42:33,832 - web_server - INFO - Successfully posted to sensor.optim_status = Optimal
2024-09-28 15:42:33,836 - web_server - INFO - Successfully posted to sensor.unit_load_cost = 0.05
2024-09-28 15:42:33,840 - web_server - INFO - Successfully posted to sensor.unit_prod_price = -0.04
```

A similar issue with p_pv_forecast.
[image]

**To Reproduce**
To reproduce run high frequency MPC without specifying power or solar in the payload.

**Expected behavior**
p_pv_forecast should be set to match sensor_power_photovoltaics
p_load_forecast should be set to match sensor_power_load_no_var_loads

For example on another EMHASS instance, where load_power_forecast & pv_power_forecast are included in the payload, the values match exactly.

[image]

**Home Assistant installation type**
 - Home Assistant Supervised

**Your hardware**
- OS: HA OS
- Architecture: amd64

**EMHASS installation type**
 - Add-on

davidusb-geek, 2024-09-28:
What seems to be odd if the load forecast? p_load_forecast = 1007.63
What is the load forecast method that you are using? The load forecast won't necessarily be the power_load_no_var_loads. If for example using the ML forecaster it will be a different value

purcell-lab, 2024-09-28:
Load forecast method is naive.

Which from my understanding is just the last 24/48 hours replayed..

With alpha=0,beta=1 I expect
 load_forecast = power_load_no_var_loads(n-48)

With alpha=1,beta=0 I was expecting
 load_forecast = power_load_no_var_loads(now)

purcell-lab, 2025-04-24:
Still having some issues with the implementation, it seems to change depending if we specify pv_forecast at runtime or use the internal forecaster:

https://emhass.readthedocs.io/en/latest/forecasts.html#now-current-values-in-forecasts
[image]

Newevg, 2025-04-24:
Alpha/Beta 1/0 is accurate when PV is active but it's opposite is true for self consumption. It's constantly overestimate "House Loads".  When Alpha/Beta 0/1 - house loads are spot on but PV is way off.  Currently running 0.5/0.5 as ratio. See at the screenshot itself Power_Load_Forecast isn't matching House_Loads

[image]

purcell-lab, 2025-04-25:
'alpha': 0, 'beta': 1,

[image]

p_pv_forecast matches:

[image]

purcell-lab, 2025-04-25:
alpha: 1, beta 0;

Should return value from sensor_power_photovoltaics for now value, but instead returns open-meteo forecast.

```logs
**'alpha': 1, 'beta': 0,** 'num_def_loads': 7, 'def_total_hours': [0, 0, 2, 3, 2, 4, 0], 'def_current_state': [0, 0, 0, 0, 1, 0, 0], 'def_start_timestep': [0, 0, 4, 0, 0, 4, 0], 'def_end_timestep': [0, 0, 0, 0, 6, 0, 0], 'P_deferrable_nom': [440, 5500, 11520, 3500.0, 600, 11520, 0], 'treat_def_as_semi_cont': [1, 1, 0, 0, 1, 0, 1], 'set_def_constant': [0, 0, 0, 0, 0, 0, 0], 'def_start_penalty': [1, 1, 1, 1, 1, 1, 1], 'weight_battery_charge': 0.0, 'weight_battery_discharge': 0.0, 'battery_nominal_energy_capacity': 40000, 'battery_discharge_power_max': 15000, 'soc_final': 0.17, 'weather_forecast_cache_only': True, 'set_nodischarge_to_grid': 0, 'soc_init': 0.9}
...
[2025-04-25 14:14:27 +1000] [23] [DEBUG] get_power_from_weather returning:
2025-04-25 14:00:00+10:00     **6242.000000**
2025-04-25 14:30:00+10:00     5253.797766
2025-04-25 15:00:00+10:00     4285.500752
2025-04-25 15:30:00+10:00     3217.029994
```

open-meto forecast: 6242 W

```logs
[2025-04-25 14:14:59 +1000] [23] [INFO]  >> Publishing data...
[2025-04-25 14:14:59 +1000] [23] [INFO] Publishing data to HASS instance
[2025-04-25 14:14:59 +1000] [23] [INFO] Successfully posted to sensor.p_pv_forecast = **6242.0**
[2025-04-25 14:14:59 +1000] [23] [INFO] Successfully posted to sensor.p_load_forecast = 1693.0
[2025-04-25 14:14:59 +1000] [23] [INFO] Successfully posted to sensor.p_pv_curtailment = 0.0
[2025-04-25 14:14:59 +1000] [23] [INFO] Successfully posted to sensor.p_deferrable0 = 0.0
[2025-04-25 14:14:59 +1000] [23] [INFO] Successfully posted to sensor.p_deferrable1 = 0.0
[2025-04-25 14:14:59 +1000] [23] [INFO] Successfully posted to sensor.p_deferrable2 = 0.0
[2025-04-25 14:14:59 +1000] [23] [INFO] Successfully posted to sensor.p_deferrable3 = 0.0
[2025-04-25 14:14:59 +1000] [23] [INFO] Successfully posted to sensor.p_deferrable4 = 600.0
[2025-04-25 14:14:59 +1000] [23] [INFO] Successfully posted to sensor.p_deferrable5 = 0.0
[2025-04-25 14:14:59 +1000] [23] [INFO] Successfully posted to sensor.p_deferrable6 = 0.0
[2025-04-25 14:14:59 +1000] [23] [INFO] Successfully posted to sensor.p_batt_forecast = -3949.0
[2025-04-25 14:14:59 +1000] [23] [INFO] Successfully posted to sensor.soc_batt_forecast = 93.95
[2025-04-25 14:14:59 +1000] [23] [INFO] Successfully posted to sensor.p_grid_forecast = 0.0
[2025-04-25 14:14:59 +1000] [23] [INFO] Successfully posted to sensor.total_cost_fun_value = 7.44
```

purcell-lab, 2025-12-10:
```
'alpha': 0, 'beta': 1
```
sensor.power_photovoltaics = 13,000 W
Expectation: sensor.p_pv_forecast = sensor.power_photovoltaics

Incorrect Result: sensor.p_pv_forecast is set to open-meto forecast.

[image]

```logs
[2025-12-11 07:55:33,417] DEBUG in forecast: get_power_from_weather returning:
2025-12-11 07:30:00+10:00     9687.000000
...
[2025-12-11 07:55:53,558] DEBUG in retrieve_hass: Successfully posted to sensor.p_pv_forecast = 9687.0
```

```
'alpha': 1, 'beta': 0
```
sensor.power_photovoltaics = 7200W
Expectation: sensor.p_pv_forecast = open meto forecast

```logs
[2025-12-11 08:06:24,037] DEBUG in forecast: get_power_from_weather returning:
2025-12-11 08:00:00+10:00     8878.000000
...
[2025-12-11 08:06:38,808] DEBUG in retrieve_hass: Successfully posted to sensor.p_pv_forecast = 8878.0
```

closed 2026-01-18 by davidusb-geek