davidusb-geek/emhass #352: Help with def_current_state

MikaelHoogen, 2024-10-05:
Hi

I'm trying to use the "def_current_state" function but can't make it work. Would appreciate some help. My understanding is that if you set "def_current_state" = true the def load will be running in the current period, even if it's not optimal. Typically to not turn off a load thats running. Is this correct?

I have an example below. I set "def_current_state" = true but the result is sensor.p_deferrable0 = 0.0.

```
curl -i -H "Content-Type: application/json" -X POST -d '{
      "load_cost_forecast":[1.075, 1.075, 1.001, 1.001, 0.971, 0.971, 0.942, 0.942, 0.946, 0.946, 0.938, 0.938, 0.938, 0.938, 0.937, 0.937, 0.938, 0.938, 0.984, 0.984, 1.048, 1.048, 1.29, 1.29, 1.521, 1.521, 1.524, 1.524, 1.483, 1.483, 1.432, 1.432, 1.379, 1.379, 1.361, 1.361, 1.431, 1.431, 1.498, 1.498, 1.553, 1.553, 1.643, 1.643, 1.684, 1.684, 1.713, 1.713, 1.612, 1.612, 1.573, 1.573, 1.529, 1.529, 1.451, 1.451],
      "prod_price_forecast":[0.777, 0.777, 0.717, 0.717, 0.693, 0.693, 0.67, 0.67, 0.677, 0.677, 0.671, 0.671, 0.671, 0.671, 0.67, 0.67, 0.671, 0.671, 0.707, 0.707, 0.758, 0.758, 0.952, 0.952, 1.137, 1.137, 1.139, 1.139, 1.107, 1.107, 1.066, 1.066, 1.023, 1.023, 1.009, 1.009, 1.065, 1.065, 1.119, 1.119, 1.163, 1.163, 1.235, 1.235, 1.268, 1.268, 1.291, 1.291, 1.21, 1.21, 1.179, 1.179, 1.144, 1.144, 1.081, 1.081],
      "pv_power_forecast":[0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 8.0, 91.0, 989.0, 1983.0, 3862.0, 5691.0, 6836.0, 7316.0, 7659.0, 7538.0, 7369.0, 7138.0, 6727.0, 6151.0, 5419.0, 4404.0, 3338.0, 2203.0, 1144.0, 374.0, 317.0, 238.0, 115.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0],
      "prediction_horizon":56,
      "alpha": 0.0,
      "beta": 1.0,
      "num_def_loads": 3,
      "num_lags":96,
      "P_deferrable_nom":[
            2200.0,
            1400,
            2000
          ],
      "treat_def_as_semi_cont": [true, true, true],
      "def_current_state": [
                        true,
                        false,
                        false],
      "def_total_hours":[
                        6.5,
                        0.0,
                        0],
      "soc_init": 0.74,
      "set_def_constant": [false,
                          false,false]
      }' http://localhost:5000/action/naive-mpc-optim
```
```
2024-09-30 20:15:32,185 - web_server - INFO - Passed runtime parameters: {'load_cost_forecast': [1.075, 1.075, 1.001, 1.001, 0.971, 0.971, 0.942, 0.942, 0.946, 0.946, 0.938, 0.938, 0.938, 0.938, 0.937, 0.937, 0.938, 0.938, 0.984, 0.984, 1.048, 1.048, 1.29, 1.29, 1.521, 1.521, 1.524, 1.524, 1.483, 1.483, 1.432, 1.432, 1.379, 1.379, 1.361, 1.361, 1.431, 1.431, 1.498, 1.498, 1.553, 1.553, 1.643, 1.643, 1.684, 1.684, 1.713, 1.713, 1.612, 1.612, 1.573, 1.573, 1.529, 1.529, 1.451, 1.451], 'prod_price_forecast': [0.777, 0.777, 0.717, 0.717, 0.693, 0.693, 0.67, 0.67, 0.677, 0.677, 0.671, 0.671, 0.671, 0.671, 0.67, 0.67, 0.671, 0.671, 0.707, 0.707, 0.758, 0.758, 0.952, 0.952, 1.137, 1.137, 1.139, 1.139, 1.107, 1.107, 1.066, 1.066, 1.023, 1.023, 1.009, 1.009, 1.065, 1.065, 1.119, 1.119, 1.163, 1.163, 1.235, 1.235, 1.268, 1.268, 1.291, 1.291, 1.21, 1.21, 1.179, 1.179, 1.144, 1.144, 1.081, 1.081], 'pv_power_forecast': [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 8.0, 91.0, 989.0, 1983.0, 3862.0, 5691.0, 6836.0, 7316.0, 7659.0, 7538.0, 7369.0, 7138.0, 6727.0, 6151.0, 5419.0, 4404.0, 3338.0, 2203.0, 1144.0, 374.0, 317.0, 238.0, 115.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0], 'prediction_horizon': 56, 'alpha': 0.0, 'beta': 1.0, 'num_def_loads': 3, 'num_lags': 96, 'P_deferrable_nom': [2200.0, 1400, 2000], 'treat_def_as_semi_cont': [True, True, True], 'def_current_state': [True, False, False], 'def_total_hours': [6.5, 0.0, 0], 'soc_init': 0.74, 'set_def_constant': [False, False, False]}
```

```
2024-09-30 20:16:26,552 - web_server - INFO - Publishing data to HASS instance
2024-09-30 20:16:26,561 - web_server - INFO - Successfully posted to sensor.p_pv_forecast = 1.61
2024-09-30 20:16:26,566 - web_server - INFO - Successfully posted to sensor.p_load_forecast = 941.34
2024-09-30 20:16:26,571 - web_server - INFO - Successfully posted to sensor.p_deferrable0 = 0.0
2024-09-30 20:16:26,576 - web_server - INFO - Successfully posted to sensor.p_deferrable1 = 0.0
2024-09-30 20:16:26,580 - web_server - INFO - Successfully posted to sensor.p_deferrable2 = 0.0
2024-09-30 20:16:26,585 - web_server - INFO - Successfully posted to sensor.p_batt_forecast = 0.0
2024-09-30 20:16:26,589 - web_server - INFO - Successfully posted to sensor.soc_batt_forecast = 74.0
2024-09-30 20:16:26,594 - web_server - INFO - Successfully posted to sensor.p_grid_forecast = 939.73
2024-09-30 20:16:26,598 - web_server - INFO - Successfully posted to sensor.total_cost_fun_value = 17.18
2024-09-30 20:16:26,601 - web_server - INFO - Successfully posted to sensor.optim_status = Optimal
2024-09-30 20:16:26,606 - web_server - INFO - Successfully posted to sensor.unit_load_cost = 1.075
2024-09-30 20:16:26,611 - web_server - INFO - Successfully posted to sensor.unit_prod_price = 0.777
```

mime24, 2025-01-24:
I'm also struggling with the def_current_state parameter on runtime.

even if I set it to true it will be ignored in the current period. the result is sensor.p_deferrable2 = 0.0.

[image]

```
2025-01-24 11:04:11,298 - web_server - INFO - Passed runtime parameters: {'prediction_horizon': 13, 'entity_save': True, 'publish_prefix': 'emhass_test_', 'soc_init': 0.828, 'soc_final': 0.7, 'battery_minimum_state_of_charge': 0.15, 'load_cost_forecast': [0.3237, 0.3214, 0.3192, 0.319, 0.319, 0.3199, 0.3292, 0.3358, 0.3396, 0.3315, 0.3265, 0.3398, 0.333], 'operating_hours_of_each_deferrable_load': [0, 0, 1, 0], 'def_current_state': [False, False, True, True], 'end_timesteps_of_each_deferrable_load': [0, 3, 5, 0], 'nominal_power_of_deferrable_loads': [1600, 300, 1000, 500], 'def_load_config': [{}, {}, {}, {'thermal_config': {'heating_rate': 0.45, 'cooling_constant': 0.011, 'overshoot_temperature': 22, 'start_temperature': 20.7, 'desired_temperatures': [21.0, 21.0, 21.0, 21.0, 21.0, 21.0, 21.0, 21.0, 21.0, 21.0, 21.0, 21.0, 21.0]}}], 'outdoor_temperature_forecast': [7.2, 7.3, 8.0, 8.7, 8.8, 8.5, 8.8, 9.3, 9.5, 9.8, 9.2, 8.7, 8.2]}
2025-01-24 11:04:11,299 - web_server - INFO -  >> Setting input data dict
2025-01-24 11:04:11,300 - web_server - INFO - Setting up needed data
2025-01-24 11:04:11,347 - web_server - INFO - Retrieve hass get data method initiated...
2025-01-24 11:04:21,042 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2025-01-24 11:04:25,607 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2025-01-24 11:04:25,609 - web_server - INFO - Retrieve hass get data method initiated...
2025-01-24 11:04:41,482 - web_server - INFO -  >> Performing naive MPC optimization...
2025-01-24 11:04:41,483 - web_server - INFO - Performing naive MPC optimization
2025-01-24 11:04:41,517 - web_server - INFO - Perform an iteration of a naive MPC controller
2025-01-24 11:04:42,070 - web_server - INFO - Status: Optimal
2025-01-24 11:04:42,071 - web_server - INFO - Total value of the Cost function = 0.19
2025-01-24 11:04:42,101 - web_server - INFO - Publishing data to HASS instance
2025-01-24 11:04:42,116 - web_server - INFO - Successfully posted to sensor.emhass_test_p_pv_forecast = 1458.8
2025-01-24 11:04:42,125 - web_server - INFO - Successfully posted to sensor.emhass_test_p_load_forecast = 444.03
2025-01-24 11:04:42,134 - web_server - INFO - Successfully posted to sensor.emhass_test_p_hybrid_inverter = 942.36
2025-01-24 11:04:42,143 - web_server - INFO - Successfully posted to sensor.emhass_test_p_deferrable0 = 0.0
2025-01-24 11:04:42,152 - web_server - INFO - Successfully posted to sensor.emhass_test_p_deferrable1 = 0.0
2025-01-24 11:04:42,161 - web_server - INFO - Successfully posted to sensor.emhass_test_p_deferrable2 = 0.0
2025-01-24 11:04:42,170 - web_server - INFO - Successfully posted to sensor.emhass_test_p_deferrable3 = 498.33
2025-01-24 11:04:42,180 - web_server - INFO - Successfully posted to sensor.emhass_test_temp_predicted3 = 20.7
2025-01-24 11:04:42,190 - web_server - INFO - Successfully posted to sensor.emhass_test_p_batt_forecast = -516.44
2025-01-24 11:04:42,200 - web_server - INFO - Successfully posted to sensor.emhass_test_soc_batt_forecast = 89.19
2025-01-24 11:04:42,209 - web_server - INFO - Successfully posted to sensor.emhass_test_p_grid_forecast = 0.0
2025-01-24 11:04:42,216 - web_server - INFO - Successfully posted to sensor.emhass_test_total_cost_fun_value = 0.23
2025-01-24 11:04:42,221 - web_server - INFO - Successfully posted to sensor.emhass_test_optim_status = Optimal
2025-01-24 11:04:42,230 - web_server - INFO - Successfully posted to sensor.emhass_test_unit_load_cost = 0.3237
2025-01-24 11:04:42,239 - web_server - INFO - Successfully posted to sensor.emhass_test_unit_prod_price = 0.082

```

davidusb-geek, 2025-01-31:
> I'm trying to use the "def_current_state" function but can't make it work. Would appreciate some help. My understanding is that if you set "def_current_state" = true the def load will be running in [...]

This is a hard one to debug.
Those constraints are hard coded, just checked the code now and it seems correct to me.
Sometimes solvers relax some constraints automatically.
Maybe the logs could give us a hint? Do you have the logs when running your MPC command?

davidusb-geek, 2025-01-31:
> I'm also struggling with the def_current_state parameter on runtime.
> 
> even if I set it to true it will be ignored in the current period. the result is sensor.p_deferrable2 = 0.0.

In your case is maybe more explainable.
That load is the one that you defined as a thermal load.
Unfortunately , in its current form, the code for the thermal load constraints is not comptaible with the `def_current_state` constraint.
Of course we could improve this in the future.

MikaelHoogen, 2025-01-31:
> This is a hard one to debug. Those constraints are hard coded, just checked the code now and it seems correct to me. Sometimes solvers relax some constraints automatically. Maybe the logs could give [...]

Hi

In my post there are logs. Is there another log you think of?

davidusb-geek, 2025-01-31:
Yes the logger changed a bit now with the guicorn integration. 
The complete logging message from the solver is now shown.

mime24, 2025-01-31:
> In your case is maybe more explainable. That load is the one that you defined as a thermal load. Unfortunately , in its current form, the code for the thermal load constraints is not comptaible with [...]

Hi @davidusb-geek you might have overseen, that I have passed two "true" states whereas just one is thermal load. the def2 (Wasching Maschine) is what I'm struggling with...

here is a further example where def 1 (dishwasher) behaves as expected and def 2 (wasching maschine) not. 

```
[2025-01-31 14:00:01 +0100] [23] [INFO] Passed runtime parameters: {'load_forecast_method': 'mlforecaster', 'prediction_horizon': 24, 'entity_save': True, 'publish_prefix': 'emhass_', 'soc_init': 1.0, 'soc_final': 0.7, 'battery_minimum_state_of_charge': 0.15, 'load_cost_forecast': [0.3746, 0.3729, 0.4087, 0.4306, 0.4508, 0.4499, 0.4434, 0.4239, 0.4121, 0.4021, 0.3867, 0.3933, 0.3844, 0.3782, 0.3721, 0.3718, 0.3738, 0.3816, 0.4026, 0.413, 0.3999, 0.3832, 0.3733, 0.3616], 'operating_hours_of_each_deferrable_load': [0, 2, 1, 0], 'def_current_state': [False, True, True, True], 'end_timesteps_of_each_deferrable_load': [0, 0, 0, 0], 'nominal_power_of_deferrable_loads': [1600, 300, 1000, 1074.3333333333335], 'def_load_config': [{}, {}, {}, {'thermal_config': {'heating_rate': 1.5, 'cooling_constant': 0.02, 'overshoot_temperature': 22, 'start_temperature': 21.6, 'desired_temperatures': [21, 21, 21, 21, 21, 21, 21, 21, 21, 21, 21, 21, 21, 21, 21, 21, 21, 21, 21, 21, 21, 21, 21, 21]}}], 'outdoor_temperature_forecast': [4.1, 4.6, 5.0, 4.6, 3.1, 1.0, 0.5, 0.0, -0.4, -0.9, -1.0, -1.2, -1.5, -1.8, -1.8, -2.0, -2.1, -2.2, -2.2, -2.1, -1.2, 0.4, 1.8, 3.0]}
[2025-01-31 14:00:01 +0100] [23] [INFO]  >> Setting input data dict
[2025-01-31 14:00:01 +0100] [23] [INFO] Setting up needed data
[2025-01-31 14:00:01 +0100] [23] [INFO] Retrieve hass get data method initiated...
[2025-01-31 14:00:11 +0100] [23] [INFO] Retrieving weather forecast data using method = scrapper
[2025-01-31 14:00:16 +0100] [23] [INFO] Retrieving data from hass for load forecast using method = mlforecaster
[2025-01-31 14:00:16 +0100] [23] [INFO] Retrieve hass get data method initiated...
[2025-01-31 14:00:47 +0100] [23] [DEBUG] Number of ML predict forcast data generated (lags_opt): 24
[2025-01-31 14:00:47 +0100] [23] [DEBUG] Number of forcast dates obtained: 24
[2025-01-31 14:00:47 +0100] [23] [INFO]  >> Performing naive MPC optimization...
[2025-01-31 14:00:47 +0100] [23] [INFO] Performing naive MPC optimization
[2025-01-31 14:00:47 +0100] [23] [INFO] Perform an iteration of a naive MPC controller
[2025-01-31 14:00:47 +0100] [23] [DEBUG] Deferrable load 0: Proposed optimization window: 0 --> 0
[2025-01-31 14:00:47 +0100] [23] [DEBUG] Deferrable load 0: Validated optimization window: 0 --> 0
[2025-01-31 14:00:47 +0100] [23] [DEBUG] Deferrable load 1: Proposed optimization window: 0 --> 0
[2025-01-31 14:00:47 +0100] [23] [DEBUG] Deferrable load 1: Validated optimization window: 0 --> 0
[2025-01-31 14:00:47 +0100] [23] [DEBUG] Deferrable load 2: Proposed optimization window: 0 --> 0
[2025-01-31 14:00:47 +0100] [23] [DEBUG] Deferrable load 2: Validated optimization window: 0 --> 0
[2025-01-31 14:00:47 +0100] [23] [DEBUG] Deferrable load 3: Proposed optimization window: 0 --> 0
[2025-01-31 14:00:47 +0100] [23] [DEBUG] Deferrable load 3: Validated optimization window: 0 --> 0
[2025-01-31 14:00:48 +0100] [23] [INFO] Status: Optimal
[2025-01-31 14:00:48 +0100] [23] [INFO] Total value of the Cost function = -0.09
[2025-01-31 14:00:48 +0100] [23] [INFO] Publishing data to HASS instance
[2025-01-31 14:00:48 +0100] [23] [INFO] Successfully posted to sensor.emhass_p_pv_forecast = 2548.53
[2025-01-31 14:00:48 +0100] [23] [DEBUG] Saved sensor.emhass_p_pv_forecast to json file
[2025-01-31 14:00:48 +0100] [23] [INFO] Successfully posted to sensor.emhass_p_load_forecast = 250.06
[2025-01-31 14:00:48 +0100] [23] [DEBUG] Saved sensor.emhass_p_load_forecast to json file
[2025-01-31 14:00:48 +0100] [23] [INFO] Successfully posted to sensor.emhass_p_hybrid_inverter = 2548.53
[2025-01-31 14:00:48 +0100] [23] [DEBUG] Saved sensor.emhass_p_hybrid_inverter to json file
[2025-01-31 14:00:48 +0100] [23] [INFO] Successfully posted to sensor.emhass_p_deferrable0 = 0.0
[2025-01-31 14:00:48 +0100] [23] [DEBUG] Saved sensor.emhass_p_deferrable0 to json file
[2025-01-31 14:00:48 +0100] [23] [INFO] Successfully posted to sensor.emhass_p_deferrable1 = 300.0
[2025-01-31 14:00:48 +0100] [23] [DEBUG] Saved sensor.emhass_p_deferrable1 to json file
[2025-01-31 14:00:48 +0100] [23] [INFO] Successfully posted to sensor.emhass_p_deferrable2 = 0.0
[2025-01-31 14:00:48 +0100] [23] [DEBUG] Saved sensor.emhass_p_deferrable2 to json file
[2025-01-31 14:00:48 +0100] [23] [INFO] Successfully posted to sensor.emhass_p_deferrable3 = 0.0
[2025-01-31 14:00:48 +0100] [23] [DEBUG] Saved sensor.emhass_p_deferrable3 to json file
[2025-01-31 14:00:48 +0100] [23] [INFO] Successfully posted to sensor.emhass_temp_predicted3 = 21.6
[2025-01-31 14:00:48 +0100] [23] [DEBUG] Saved sensor.emhass_temp_predicted3 to json file
[2025-01-31 14:00:48 +0100] [23] [INFO] Successfully posted to sensor.emhass_p_batt_forecast = 0.0
[2025-01-31 14:00:48 +0100] [23] [DEBUG] Saved sensor.emhass_p_batt_forecast to json file
[2025-01-31 14:00:48 +0100] [23] [INFO] Successfully posted to sensor.emhass_soc_batt_forecast = 100.0
[2025-01-31 14:00:48 +0100] [23] [DEBUG] Saved sensor.emhass_soc_batt_forecast to json file
[... cut ...]
[2025-01-31 14:00:48 +0100] [23] [DEBUG] Saved sensor.emhass_optim_status to json file
[2025-01-31 14:00:48 +0100] [23] [INFO] Successfully posted to sensor.emhass_unit_load_cost = 0.3746
[2025-01-31 14:00:48 +0100] [23] [DEBUG] Saved sensor.emhass_unit_load_cost to json file
[2025-01-31 14:00:48 +0100] [23] [INFO] Successfully posted to sensor.emhass_unit_prod_price = 0.082
[2025-01-31 14:00:48 +0100] [23] [DEBUG] Saved sensor.emhass_unit_prod_price to json file
```

last week, I tried to solve it by giving a penalty per start of def load but without success.