davidusb-geek/emhass #355: pvlib.forecast module deprecated

abriotde, 2024-10-19:
**Describe the bug**
[pvlib.forecast](https://pvlib-python.readthedocs.io/en/v0.9.5/user_guide/forecasts.html) module is deprecated since pvlib 0.9.1

**To Reproduce**
On Python 3.12.3 (Default on Ubuntu 24.04), pvlib==0.11.1 and when I install emhass with pip, emhass module get error`ModuleNotFoundError: No module named 'pvlib.forecast'`.

**Expected behavior**
At least you should set in pip install restriction on Python pvlib versions.

The best should be to replace pvlib.forecast whith [solarforecastarbiter](https://github.com/SolarArbiter/solarforecastarbiter-core)

**EMHASS installation type**
 - Legacy Python virtual environment

abriotde, 2024-10-19:
solarforecastarbiter is MIT license, so it's ok for that.
But solarforecastarbiter require pvlib-0.8.0 max and emhass require pvlib version superior (numpy : np.Inf)

davidusb-geek, 2024-10-19:
We are not using pvlib.forecast.
We are just using PVLib to model the solar PV plant, which is certainly not deprecated. 
The weather forecast with the included solar irradiance comes from other sources: ex Clear Outside

abriotde, 2024-10-20:
Exact sous la dernière version de emhass, mais sous Python 3.12.3,  pip install emhass==0.2.1. Et alors sous site-packages/emhass/forecast.py utilise bien pvlib.forecast. D'après git, pvlib.forecast a été supprimé de l'historique le 2022-03-29 12:14:32. Et d'ailleurs la version 0.2.1 date du 22 Dec 2021...

Autrement dit, est-ce possible de mettre à jour la version de pip? A moins que j'ai une vielle version à cause d'un autre problème de compatibilité. Mais je ne crois pas car je suis dans un venv propre avec une version de Python récente.

davidusb-geek, 2024-10-20:
EMHASS est en version 0.10.6. Énormément des choses ont changé depuis la v0.2.1.
Il faut y aller vers la dernière version dispo.

abriotde, 2024-10-20:
Oui mais avec pip, je ne peux pas avoir la dernière version.
Je crois que normalement cela devrait être automatique avec les "actions github" (CI/CD) mais quelques chose doit être cassé. Pourtant si le pipeline n'a pas fonctionné pour la 0.10.6, il a fonctionné pour 0.10.5, cette dernière devrait donc être dispo pour pip...

abriotde, 2024-10-20:
Je crois que j'ai une piste sur la cause possible : requirement.txt
Jusqu'au 2023-01-18 il y avait `numpy>=1.20.1` après il y a `numpy==1.20.1` (Commit "Fix - Fixing dependencies, prepared new version"). Or j'ai "numpy==2.1.2" La dernière version satisfaisant requirement.txt avec  `numpy==1.20.1` est la v0.2.1 qui date de peu avant.
Je suppose que cela a été fait parce que emhass ne supportait pas la dernière  version de l'époque de numpy. 
Je suggère quelque chose comme : `numpy>=1.20,<=2.1`

davidusb-geek, 2024-10-20:
> Je crois que normalement cela devrait être automatique avec les "actions github" (CI/CD) mais quelques chose doit être cassé. Pourtant si le pipeline n'a pas fonctionné pour la 0.10.6, il a fonctionné pour 0.10.5, cette dernière devrait donc être dispo pour pip...

Non le pipeline failed n'a rien à voir avec le déploiement vers pip. La dernière version sur pip est bien 0.10.6.

Je ne comprends pas les problèmes avec la version de Numpy.
En partant d'un environnement vierge `python -m venv .venv` puis en activant l'environnement `source .venv/bin/activate`, alors on doit pouvoir installer la dernière version de EMHASS 0.10.6 avec la bonne version de Numpy (1.26.4) avec:
```
pip install emhass
```

abriotde, 2024-10-20:
Avez vous essayer?

>  En partant d'un environnement vierge python -m venv .venv puis en activant l'environnement source .venv/bin/activate, alors on doit pouvoir installer la dernière version de EMHASS 0.10.6 avec la bonne version de Numpy (1.26.4) avec `pip install emhass`

C'est exactement ce que j'ai fais.

On a la dernière version que si `numpy==1.20.1`. Sinon, pip installe la dernière version compatible. En l'occurence celle qui accepte `numpy==2.1.2` dans mon cas. Ce qui correspond à  `emhass==0.2.1` car à cette époque le prérequis était `numpy>=1.20.1`.

Je ne sais pas comment l'expliquer autrement, pip ne va pas installer un emhass non compatible, c'est le but d'un système de paquets: gérer les dépendances.

davidusb-geek, 2024-10-20:
Le CI/CD le fais tout le temps et ça se passe sans problèmes.
Exemple: https://github.com/davidusb-geek/emhass/actions/runs/11182079770/job/31087531322

abriotde, 2024-10-21:
Exact si je fais `docker run -it python:3.11 bash` alors il install bien emhass-0.10.6... et il install `numpy==1.26.4`... ce qui est une grosse limitation.

Mais sur mon PC, Python est en version 3.12, et ça ne marche pas. Il install emhass-0.2.1.

Je ne pense pas qu'intrinsèquement, emhass ait des limitations pour tourner en python:3.12, c'est juste les conséquences de requirement.txt...

davidusb-geek, 2024-10-21:
A la dernière vérification certaines dépendences n'étaient pas compatible avec Python 3.12, d'où la contrainte `python_requires='>=3.10, <3.12'`. Il faudra vérifier à nouveau

abriotde, 2024-10-21:
Ok, c'est une explication qui me va.
Mais alors, ce serait bien de virer emhass-0.2.1 pour que pip ne le propose pas

davidusb-geek, 2024-10-29:
Otherwise a better approach is to directly use docker images. These are readily available for multiple platforms.
Closing

closed 2024-10-29 by davidusb-geek

abriotde, 2024-10-29:
Docker image is not a good think to use it in Python. I choose to use it from source code directly (In Python 3.12). It's simpler to give back Python's datas. 
But I tested both : https://github.com/abriotde/openhems-sample/blob/main/scripts/emhass/test_emhass.py