davidusb-geek/emhass #365: Upgrade to Add-on 0.11.1: failed

heinemannj, 2024-10-29:
```
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:245
integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 07:26:35 (1 occurrences)
Last logged: 07:26:35

[140318071754320] Unexpected exception
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 168, in async_install
    await async_update_addon(self.hass, slug=self._addon_slug, backup=backup)
  File "/usr/src/homeassistant/homeassistant/components/hassio/handler.py", line 58, in _wrapper
    data = await funct(*argv, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/hassio/handler.py", line 112, in async_update_addon
    return await hassio.send_command(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/hassio/handler.py", line 541, in send_command
    raise HassioAPIError(error.get(ATTR_MESSAGE))
homeassistant.components.hassio.handler.HassioAPIError: Unknown error, see supervisor

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 245, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2761, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2804, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 996, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 1068, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 154, in async_install
    await entity.async_install_with_progress(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 463, in async_install_with_progress
    await self.async_install(version, backup)
  File "/usr/src/homeassistant/homeassistant/components/hassio/update.py", line 170, in async_install
    raise HomeAssistantError(f"Error updating {self.title}: {err}") from err
homeassistant.exceptions.HomeAssistantError: Error updating EMHASS: Unknown error, see supervisor

SolaX_Power_Network_Technology__Zhe_jiang__Co___Ltd___A1_Hybrid_7_0_US__240V_
```

davidusb-geek, 2024-10-29:
What does the supervisor logger says?

heinemannj, 2024-10-29:
```
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'list_start_timesteps_of_each_deferrable_load' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'list_end_timesteps_of_each_deferrable_load' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'list_peak_hours_periods_start_hours' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'list_peak_hours_periods_end_hours' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'list_treat_deferrable_load_as_semi_cont' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'list_set_deferrable_load_single_constant' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'list_set_deferrable_startup_penalty' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'load_peak_hours_cost' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'load_offpeak_hours_cost' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'photovoltaic_production_sell_price' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'list_pv_module_model' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'list_pv_inverter_model' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'list_surface_tilt' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'list_surface_azimuth' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'list_modules_per_string' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'list_strings_per_inverter' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'inverter_is_hybrid' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'compute_curtailment' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'set_use_battery' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'battery_nominal_energy_capacity' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'optimization_time_step' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'historic_days_to_retrieve' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'method_ts_round' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'lp_solver' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'lp_solver_path' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'set_battery_dynamic' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'battery_dynamic_max' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'battery_dynamic_min' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'weight_battery_discharge' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'weight_battery_charge' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'load_forecast_method' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'battery_discharge_power_max' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:41:06.448 WARNING (MainThread) [supervisor.addons.options] Option 'battery_charge_power_max' does not exist in the schema for EMHASS (5b918bf2_emhass)
[... cut ...]
2024-10-29 16:44:57.536 WARNING (MainThread) [supervisor.addons.options] Option 'load_negative' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:44:57.536 WARNING (MainThread) [supervisor.addons.options] Option 'load_cost_forecast_method' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:44:57.536 WARNING (MainThread) [supervisor.addons.options] Option 'weather_forecast_method' does not exist in the schema for EMHASS (5b918bf2_emhass)
2024-10-29 16:44:57.536 WARNING (MainThread) [supervisor.addons.options] Option 'delta_forecast_daily' does not exist in the schema for EMHASS (5b918bf2_emhass)
```

bert1111, 2024-11-10:
Hi, I just updated emhass and I see the same supervisor error logs. I did the config conversion.

bert1111, 2024-11-10:
Anyone else see these errors in supervisor after updating emhass to 0.11.2?
I tried to get rid of it but not succesfull up until now. The list keeps getting longer with the same errors.
I did the config conversion.

`[33m2024-11-10` 15:55:44.535 WARNING (MainThread) [supervisor.addons.options] Option 'battery_dynamic_min' does not exist in the schema for EMHASS (5b918bf2_emhass)[0m
[33m2024-11-10 15:55:44.535 WARNING (MainThread) [supervisor.addons.options] Option 'weight_battery_discharge' does not exist in the schema for EMHASS (5b918bf2_emhass)[0m
[33m2024-11-10 15:55:44.536 WARNING (MainThread) [supervisor.addons.options] Option 'weight_battery_charge' does not exist in the schema for EMHASS (5b918bf2_emhass)[0m
[33m2024-11-10 15:55:44.536 WARNING (MainThread) [supervisor.addons.options] Option 'load_forecast_method' does not exist in the schema for EMHASS (5b918bf2_emhass)[0m
[33m2024-11-10 15:55:44.537 WARNING (MainThread) [supervisor.addons.options] Option 'battery_discharge_power_max' does not exist in the schema for EMHASS (5b918bf2_emhass)[0m
[33m2024-11-10 15:55:44.537 WARNING (MainThread) [supervisor.addons.options] Option 'battery_charge_power_max' does not exist in the schema for EMHASS (5b918bf2_emhass)[0m
[33m2024-11-10 15:55:44.537 WARNING (MainThread) [supervisor.addons.options] Option 'battery_discharge_efficiency' does not exist in the schema for EMHASS (5b918bf2_emhass)[0m
[33m2024-11-10 15:55:44.537 WARNING (MainThread) [supervisor.addons.options] Option 'battery_charge_efficiency' does not exist in the schema for EMHASS (5b918bf2_emhass)[0m
[33m2024-11-10 15:55:44.537 WARNING (MainThread) [supervisor.addons.options] Option 'battery_minimum_state_of_charge' does not exist in the schema for EMHASS (5b918bf2_emhass)[0m
[33m2024-11-10 15:55:44.537 WARNING (MainThread) [supervisor.addons.options] Option 'battery_maximum_state_of_charge' does not exist in the schema for EMHASS (5b918bf2_emhass)[0m
[33m2024-11-10 15:55:44.537 WARNING (MainThread) [supervisor.addons.options] Option 'battery_target_state_of_charge' does not exist in the schema for EMHASS (5b918bf2_emhass)[0m
[33m2024-11-10 15:55:44.537 WARNING (MainThread) [supervisor.addons.options] Option 'delta_forecast_daily' does not exist in the schema for EMHASS (5b918bf2_emhass)[0m
[33m2024-11-10 15:55:44.538 WARNING (MainThread) [supervisor.addons.options] Option 'weather_forecast_method' does not exist in the schema for EMHASS (5b918bf2_emhass)[0m
[33m2024-11-10 15:55:44.538 WARNING (MainThread) [supervisor.addons.options] Option 'load_cost_forecast_method' does not exist in the schema for EMHASS (5b918bf2_emhass)[0m
[33m2024-11-10 15:55:44.538 WARNING (MainThread) [supervisor.addons.options] Option 'set_zero_min' does not exist in the schema for EMHASS (5b918bf2_emhass)[0m
[33m2024-11-10 15:55:44.538 WARNING (MainThread) [supervisor.addons.options] Option 'production_price_forecast_method' does not exist in the schema for EMHASS (5b918bf2_emhass)[0m
[33m2024-11-10 15:55:44.538 WARNING (MainThread) [supervisor.addons.options] Option 'optional_solcast_api_key' does not exist in the schema for EMHASS (5b918bf2_emhass)[0m
[33m2024-11-10 15:55:44.538 WARNING (MainThread) [supervisor.addons.options] Option 'optional_solar_forecast_kwp' does not exist in the schema for EMHASS (5b918bf2_emhass)[0m
[33m2024-11-10 15:55:44.538 WARNING (MainThread) [supervisor.addons.options] Option 'optional_solcast_rooftop_id' does not exist in the schema for EMHASS (5b918bf2_emhass)[0m`

heinemannj, 2024-11-12:
I was able to fix by deinstalling and reinstalling the addon.

However - the complete configuration procedure must be done afterwards.
**Have MUST habe a backup of your YAML configuration available before the reinstallation**!

purcell-lab, 2024-11-19:
> However - the complete configuration procedure must be done afterwards.
> **You MUST have a backup of your YAML configuration available before the reinstallation**!

This is a difficult step for many folks as you do the upgrade, then you realize you needed to save the configuration YAML from before the upgrade.

Is there a way the legacy configuration file can be saved and made available after the upgrade?

Can the upgrade from the legacy configuration to the new configuration be automatic?

RafAustralia, 2024-12-17:
@GeoDerp 

Hey morning ...

I have had numerous attempts with upgrade and I get stuck each and every time...
Upgrade goes through but I cannot expand any buttons in you gui to paste the emhass settings info... and without it - the add on is failing.... done it with @purcell-lab and we both do not really know where it goes wrong... suggestions please?

GeoDerp, 2024-12-20:
> @GeoDerp
> 
> Hey morning ...
> 
> I have had numerous attempts with upgrade and I get stuck each and every time... Upgrade goes through but I cannot expand any buttons in you gui to paste the emhass settings info... and without it - the add on is failing.... done it with @purcell-lab and we both do not really know where it goes wrong... suggestions please?
> 
> I cannot click any buttons see below:
> 
> ![image](https://private-user-images.githubusercontent.com/168625844/396704673-31276d4b-7d07-4628-a974-a13029de95ea.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ2NzYwOTgsIm5iZiI6MTczNDY3NTc5OCwicGF0aCI6Ii8xNjg2MjU4NDQvMzk2NzA0NjczLTMxMjc2ZDRiLTdkMDctNDYyOC1hOTc0LWExMzAyOWRlOTVlYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyMFQwNjIzMThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02ZDdmMzgyYzFkYzZhNTNmN2MxNmJlYWQ0MDgyYzEzNmFmYWZiYWI1NmMwMGE0ZmExNzM3YTA5MTdiZWRmZWNkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.RPamLE5JEPPdnLtqvrod6zcnIhIhMbNwuM_7cs5v_Bg) ![image](https://private-user-images.githubusercontent.com/168625844/396704727-10d399aa-af05-45dd-af0f-e91c8aad253f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ2NzYwOTgsIm5iZiI6MTczNDY3NTc5OCwicGF0aCI6Ii8xNjg2MjU4NDQvMzk2NzA0NzI3LTEwZDM5OWFhLWFmMDUtNDVkZC1hZjBmLWU5MWM4YWFkMjUzZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyMFQwNjIzMThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kMmI3NjMxOTRhNGQ0NTU4ZjkzZDVlNTM1ZDYwY2YwYTU5NzY2YmUzMjFkOGE5NGM3NzY4YWY1ZTRmZTliMzUzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.9z0n45-canHilYx0K7ImrNLLbJk5cPWiuwzPEHxUvNI)
> 
> nothing expands nothing is live cant put back my settings.... which I have saved out...

First off im sorry for the late reply.
One reason why nothing is working could be that javascript is disabled on your web browser. Try another web browser

RafAustralia, 2024-12-20:
> > @GeoDerp
> > Hey morning ...
> > I have had numerous attempts with upgrade and I get stuck each and every time... Upgrade goes through but I cannot expand any buttons in you gui to paste the emhass settings info... and without it - the add on is failing.... done it with @purcell-lab and we both do not really know where it goes wrong... suggestions please?
> > I cannot click any buttons see below:
> > ![image](https://private-user-images.githubusercontent.com/168625844/396704673-31276d4b-7d07-4628-a974-a13029de95ea.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ2NzYwOTgsIm5iZiI6MTczNDY3NTc5OCwicGF0aCI6Ii8xNjg2MjU4NDQvMzk2NzA0NjczLTMxMjc2ZDRiLTdkMDctNDYyOC1hOTc0LWExMzAyOWRlOTVlYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyMFQwNjIzMThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02ZDdmMzgyYzFkYzZhNTNmN2MxNmJlYWQ0MDgyYzEzNmFmYWZiYWI1NmMwMGE0ZmExNzM3YTA5MTdiZWRmZWNkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.RPamLE5JEPPdnLtqvrod6zcnIhIhMbNwuM_7cs5v_Bg) ![image](https://private-user-images.githubusercontent.com/168625844/396704727-10d399aa-af05-45dd-af0f-e91c8aad253f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ2NzYwOTgsIm5iZiI6MTczNDY3NTc5OCwicGF0aCI6Ii8xNjg2MjU4NDQvMzk2NzA0NzI3LTEwZDM5OWFhLWFmMDUtNDVkZC1hZjBmLWU5MWM4YWFkMjUzZi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMjIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTIyMFQwNjIzMThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kMmI3NjMxOTRhNGQ0NTU4ZjkzZDVlNTM1ZDYwY2YwYTU5NzY2YmUzMjFkOGE5NGM3NzY4YWY1ZTRmZTliMzUzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.9z0n45-canHilYx0K7ImrNLLbJk5cPWiuwzPEHxUvNI)
> > nothing expands nothing is live cant put back my settings.... which I have saved out...
> 
> First off im sorry for the late reply. One reason why nothing is working could be that javascript is disabled on your web browser. Try another web browser. If that doesnt work we can look deeper into the issue.

hey @GeoDerp 

Thank you so much for your reply first and foremost...

I just checked and my java script is enabled.
This has also happened both in Chrome as well as Edge.

Chrome snipped of settings below:

[image]

Any other ideas? Happy to chat on discord if you need to check anything else... (raf_t_aud)

GeoDerp, 2024-12-20:
Sure, mine is: geo_the_noodle

GeoDerp, 2024-12-20:
```
{
  "battery_charge_efficiency": 0.95,
  "battery_charge_power_max": 1000,
  "battery_discharge_efficiency": 0.95,
  "battery_discharge_power_max": 1000,
  "battery_dynamic_max": 0.9,
  "battery_dynamic_min": -0.9,
  "battery_maximum_state_of_charge": 0.9,
  "battery_minimum_state_of_charge": 0.3,
  "battery_nominal_energy_capacity": 5000,
  "battery_target_state_of_charge": 0.6,
  "compute_curtailment": false,
  "continual_publish": false,
  "costfun": "profit",
  "delta_forecast_daily": 1,
  "end_timesteps_of_each_deferrable_load": [
    0,
    0
  ],
  "historic_days_to_retrieve": 2,
  "inverter_is_hybrid": false,
  "load_cost_forecast_method": "hp_hc_periods",
  "load_forecast_method": "naive",
  "load_negative": false,
  "load_offpeak_hours_cost": 0.1419,
  "load_peak_hour_periods": {
    "period_hp_1": [
      {
        "start": "02:54"
      },
      {
        "end": "15:24"
      }
    ],
    "period_hp_2": [
      {
        "start": "17:24"
      },
      {
        "end": "20:24"
      }
    ]
  },
  "load_peak_hours_cost": 0.1907,
  "logging_level": "INFO",
  "lp_solver": "default",
  "lp_solver_path": "empty",
  "maximum_power_from_grid": 9000,
  "maximum_power_to_grid": 9000,
  "method_ts_round": "nearest",
  "modules_per_string": [
    16
  ],
  "nominal_power_of_deferrable_loads": [
    3000,
    750
  ],
  "number_of_deferrable_loads": 2,
  "operating_hours_of_each_deferrable_load": [
    4,
    0
  ],
  "optimization_time_step": 30,
  "photovoltaic_production_sell_price": 0.1419,
  "production_price_forecast_method": "constant",
  "pv_inverter_model": [
    "Fronius_International_GmbH__Fronius_Primo_5_0_1_208_240__240V_"
  ],
  "pv_module_model": [
    "CSUN_Eurasia_Energy_Systems_Industry_and_Trade_CSUN295_60M"
  ],
  "sensor_linear_interp": [
    "sensor.power_photovoltaics",
    "sensor.power_load_no_var_loads"
  ],
  "sensor_power_load_no_var_loads": "sensor.power_load_no_var_loads",
  "sensor_power_photovoltaics": "sensor.power_photovoltaics",
  "sensor_replace_zero": [
    "sensor.power_photovoltaics"
  ],
  "set_battery_dynamic": false,
  "set_deferrable_load_single_constant": [
    false,
    false
  ],
  "set_deferrable_startup_penalty": [
    0,
    0
  ],
  "set_nocharge_from_grid": false,
  "set_nodischarge_to_grid": true,
  "set_total_pv_sell": false,
  "set_use_battery": false,
  "set_zero_min": true,
  "start_timesteps_of_each_deferrable_load": [
    0,
    0
  ],
  "strings_per_inverter": [
    1
  ],
  "surface_azimuth": [
    205
  ],
  "surface_tilt": [
    30
  ],
  "treat_deferrable_load_as_semi_cont": [
    true,
    true
  ],
  "weather_forecast_method": "scrapper",
  "weight_battery_charge": 1,
  "weight_battery_discharge": 1
}
```

davidusb-geek, 2025-01-31:
In the latest version I haven't seen this type of issues.
If it presist maybe be worth opening a dedicated issue for that.
Closing this as the original issue is solved

closed 2025-01-31 by davidusb-geek