davidusb-geek/emhass #369: Bugs in optimization with Battery enabled

GeoDerp, 2024-11-02:
@purcell-lab 

ive compared our config:
Mine is below for reference.

I note the following differences (that just arent general settings)
**"battery_maximum_state_of_charge": 1,**
 - I had an issue with setting 1, and had to set to 0.99

**"battery_minimum_state_of_charge": 0,**
- I had an issue with setting 0, and had to set 0.01

**"set_battery_dynamic": true,"**
- i could never get this to work. Did you want to try disabling for troubleshooting?

**"weight_battery_charge": 1,"
"weight_battery_discharge": 0.05"**
- both of these values after the last release i was getting very obscure behaviour.  I set them both to "0" and it resolved it.

Ohh, and your SOC_final in your CURL is also 1.  Did you want to troubleshoot with a 0.99?

Otherwise your config looks within reason the same as mine, +/- inverter/grid/solar settings.

```

{
  "battery_charge_efficiency": 0.99,
  "battery_charge_power_max": 8000,
  "battery_discharge_efficiency": 0.99,
  "battery_discharge_power_max": 8000,
  "battery_dynamic_max": 0.9,
  "battery_dynamic_min": -0.9,
  "battery_maximum_state_of_charge": 0.99,
  "battery_minimum_state_of_charge": 0.01,
  "battery_nominal_energy_capacity": 75000,
  "battery_target_state_of_charge": 0.98,
  "compute_curtailment": false,
  "continual_publish": false,
  "costfun": "profit",
  "delta_forecast_daily": 2,
  "end_timesteps_of_each_deferrable_load": [
    0,
    0
  ],
  "historic_days_to_retrieve": 3,
  "inverter_is_hybrid": true,
  "load_cost_forecast_method": "hp_hc_periods",
  "load_forecast_method": "naive",
  "load_negative": false,
  "load_offpeak_hours_cost": 0.08,
  "load_peak_hour_periods": {
    "period_hp_1": [
      {
        "start": "06:00"
      },
      {
        "end": "11:00"
      }
    ],
    "period_hp_2": [
      {
        "start": "14:00"
      },
      {
        "end": "23:59"
      }
    ]
  },
  "load_peak_hours_cost": 0.48,
  "logging_level": "INFO",
  "lp_solver": "COIN_CMD",
  "lp_solver_path": "/usr/bin/cbc",
  "maximum_power_from_grid": 15000,
  "maximum_power_to_grid": 5000,
  "method_ts_round": "nearest",
  "modules_per_string": [
    11
  ],
  "nominal_power_of_deferrable_loads": [
    0,
    0
  ],
  "number_of_deferrable_loads": 2,
  "operating_hours_of_each_deferrable_load": [
    0,
    0
  ],
  "optimization_time_step": 30,
  "photovoltaic_production_sell_price": 0.03,
  "production_price_forecast_method": "constant",
  "pv_inverter_model": [
    "Fronius_International_GmbH__Fronius_Primo_5_0_1_208_240__240V_"
  ],
  "pv_module_model": [
    "CSUN_Eurasia_Energy_Systems_Industry_and_Trade_CSUN295_60M"
  ],
  "sensor_linear_interp": [
    "sensor.power_ac_fronius_inverter_1_192_168_10_122"
  ],
  "sensor_power_load_no_var_loads": "sensor.victron_ac_l1",
  "sensor_power_photovoltaics": "sensor.power_ac_fronius_inverter_1_192_168_10_122",
  "sensor_replace_zero": [
    "sensor.power_ac_fronius_inverter_1_192_168_10_122",
    "sensor.victron_ac_l1"
  ],
  "set_battery_dynamic": false,
  "set_deferrable_load_single_constant": [
    false,
    false
  ],
  "set_deferrable_startup_penalty": [
    0,
    0
  ],
  "set_nocharge_from_grid": false,
  "set_nodischarge_to_grid": false,
  "set_total_pv_sell": false,
  "set_use_battery": true,
  "set_zero_min": false,
  "start_timesteps_of_each_deferrable_load": [
    0,
    0
  ],
  "strings_per_inverter": [
    2
  ],
  "surface_azimuth": [
    10
  ],
  "surface_tilt": [
    15
  ],
  "treat_deferrable_load_as_semi_cont": [
    true,
    true
  ],
  "weather_forecast_method": "scrapper",
  "weight_battery_charge": 0,
  "weight_battery_discharge": 0
}
```

_Originally posted by @kcoffau in https://github.com/davidusb-geek/emhass/issues/334#issuecomment-2450869516_

purcell-lab, 2024-11-07:
`weight_battery_discharge` should not default to 1 ($/ kWh) it goes infeasible almost immediately.

Defaults should be zero, and then I find tuning to around 0.05 or 0.10 give useful results.

purcell-lab, 2024-11-07:
> **"battery_maximum_state_of_charge": 1,**
> 
> * I had an issue with setting 1, and had to set to 0.99
> 
> **"battery_minimum_state_of_charge": 0,**
> 
> * I had an issue with setting 0, and had to set 0.01

Understood, I can setup as a workaround.

However my battery does go to 100% and 0% and EMHASS reports infeasible if SOC is above or below its desired limits.

mk2lehe, 2024-11-09:
Seeing similar issues related to battery weights. I also notice the soc_init value doesn't seem to be handled correctly. 

Testing weight_battery_discharge: 0 and weight_battery_charge: 1 gives very interesting result, not as expected.
I use rest_command: for mpc (for testing purpose I set load_power_forecast to 550W)

I'll be happy to help you guys test more if needed.
```
  trigger_mpc_forecast_dummy_load_power:
    url: http://localhost:5000/action/naive-mpc-optim
    method: POST
    content_type: "application/json"
    payload: >-
      {
      "prediction_horizon":{{((state_attr('sensor.nordpool', 'raw_today') | map(attribute='value') | list  + state_attr('sensor.nordpool', 'raw_tomorrow') | map(attribute='value') | list ))[now().hour:][:24] |count}},
      "soc_init":{{states.sensor.sh80rt_a2290552450_battery_level_soc.state|float(0.6)|round(0)/100}},
      "soc_final":0.6,
      "load_cost_forecast":{{((state_attr('sensor.nordpool', 'raw_today') | map(attribute='value') | list  + state_attr('sensor.nordpool', 'raw_tomorrow') | map(attribute='value') | list))[now().hour:][:24] |tojson}},
      "prod_price_forecast":{{((state_attr('sensor.nordpool', 'raw_today') | map(attribute='value') | list  + state_attr('sensor.nordpool', 'raw_tomorrow') | map(attribute='value') | list))[now().hour:][:24] |tojson}},
      "delta_forecast":2,
      "alpha": 1,
      "beta": 0,
      "def_total_hours":[0, 0],
      "def_end_timestep":[0, 0],
      "load_power_forecast":{{ ([550] * ((state_attr('sensor.nordpool', 'raw_today') | map(attribute='value') | list + state_attr('sensor.nordpool', 'raw_tomorrow') | map(attribute='value') | list))[now().hour:][:24] | count) | tojson }}
      }
```

```
{
    "logging_level": "DEBUG",
    "costfun": "profit",
    "optimization_time_step": 60,
    "historic_days_to_retrieve": 2,
    "method_ts_round": "first",
    "continual_publish": false,
    "data_path": "default",
    "set_total_pv_sell": false,
    "lp_solver": "COIN_CMD",
    "lp_solver_path": "empty",
    "set_nocharge_from_grid": false,
    "set_nodischarge_to_grid": false,
    "set_battery_dynamic": false,
    "battery_dynamic_max": 0.9,
    "battery_dynamic_min": -0.9,
    "weight_battery_discharge": 0,
    "weight_battery_charge": 1,
    "sensor_power_photovoltaics": "sensor.sh80rt_a2290552450_mppt_total_power",
    "sensor_power_load_no_var_loads": "sensor.house_consumption_power",
    "sensor_replace_zero": [
        "sensor.sh80rt_a2290552450_mppt_total_power"
    ],
    "sensor_linear_interp": [
        "sensor.sh80rt_a2290552450_mppt_total_power",
        "sensor.house_consumption_power"
    ],
    "load_negative": false,
    "set_zero_min": true,
    "number_of_deferrable_loads": 2,
    "nominal_power_of_deferrable_loads": [
        300,
        75
    ],
    "operating_hours_of_each_deferrable_load": [
        4,
        0
    ],
    "weather_forecast_method": "scrapper",
    "load_forecast_method": "naive",
    "delta_forecast_daily": 1,
    "load_cost_forecast_method": "hp_hc_periods",
[... 77 lines cut ...]
2024-11-09 14:37:45,510 - web_server - INFO - Home Assistant data fetch will be performed using url: http://supervisor/core/api
2024-11-09 14:37:45,510 - web_server - INFO - The data path is: /share
2024-11-09 14:37:45,510 - web_server - INFO - The logging is: DEBUG
2024-11-09 14:37:45,516 - web_server - INFO - Using core emhass version: 0.11.2
waitress   INFO  Serving on http://0.0.0.0:5000
2024-11-09 14:37:57,117 - web_server - INFO - Passed runtime parameters: {'prediction_horizon': 24, 'soc_init': 0.87, 'soc_final': 0.6, 'load_cost_forecast': [1.013, 1.354, 1.545, 1.637, 1.644, 1.642, 1.624, 1.506, 1.339, 1.336, 0.917, 0.775, 0.767, 0.725, 0.762, 1.014, 0.994, 1.033, 1.175, 1.253, 1.267, 1.317, 1.278, 1.214], 'prod_price_forecast': [1.013, 1.354, 1.545, 1.637, 1.644, 1.642, 1.624, 1.506, 1.339, 1.336, 0.917, 0.775, 0.767, 0.725, 0.762, 1.014, 0.994, 1.033, 1.175, 1.253, 1.267, 1.317, 1.278, 1.214], 'delta_forecast': 2, 'alpha': 1, 'beta': 0, 'def_total_hours': [0, 0], 'def_end_timestep': [0, 0], 'load_power_forecast': [550, 550, 550, 550, 550, 550, 550, 550, 550, 550, 550, 550, 550, 550, 550, 550, 550, 550, 550, 550, 550, 550, 550, 550]}
2024-11-09 14:37:57,117 - web_server - INFO -  >> Setting input data dict
2024-11-09 14:37:57,117 - web_server - INFO - Setting up needed data
2024-11-09 14:37:57,141 - web_server - WARNING - warning `days_to_retrieve` is set to a value less than 9, this could cause an error with the fit
2024-11-09 14:37:57,141 - web_server - WARNING - setting`passed_data:days_to_retrieve` to 9 for fit/predict/tune
2024-11-09 14:37:57,142 - web_server - WARNING - lp_solver=COIN_CMD but lp_solver_path=empty, attempting to use lp_solver_path=/usr/bin/cbc
2024-11-09 14:37:57,143 - web_server - INFO - Retrieve hass get data method initiated...
2024-11-09 14:37:58,826 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2024-11-09 14:38:00,147 - web_server - INFO -  >> Performing naive MPC optimization...
2024-11-09 14:38:00,148 - web_server - INFO - Performing naive MPC optimization
2024-11-09 14:38:00,160 - web_server - INFO - Perform an iteration of a naive MPC controller
2024-11-09 14:38:00,209 - web_server - DEBUG - Deferrable load 0: Proposed optimization window: 0 --> 0
2024-11-09 14:38:00,209 - web_server - DEBUG - Deferrable load 0: Validated optimization window: 0 --> 0
2024-11-09 14:38:00,211 - web_server - DEBUG - Deferrable load 1: Proposed optimization window: 0 --> 0
2024-11-09 14:38:00,211 - web_server - DEBUG - Deferrable load 1: Validated optimization window: 0 --> 0
2024-11-09 14:38:01,508 - web_server - INFO - EMHASS server online, serving index.html...
2024-11-09 14:38:02,937 - web_server - INFO - Status: Optimal
2024-11-09 14:38:02,937 - web_server - INFO - Total value of the Cost function = 46.24
```
[image]

GeoDerp, 2024-11-15:
@davidusb-geek . Whats your thoughts on these issues? Happy to make a PR with modified defaults on request.

mk2lehe, 2024-11-15:
As added input, not being sure if related, changing costfun to self-consumption seemed to resolve all issues. However p_batt went "unstable" (alternating charging and discharging) tonight when load/price_cost_forecast went negative. So I tried added a value in both lists entries so it never goes negative, and the strange p_batt behaviour disappeared and went back to normal.

mime24, 2024-12-13:
Hello everyone,

My problem might be related to this bug. As soon as I activate the battery in the configuration, I see the status "infeasible" for day-ahead and MPC. I have been trying to change this with various parameters for weeks, but without success. Overall, the results look promising despite the "infeasible" status. Here is my current configuration.

```
{
  "battery_charge_efficiency": 0.95,
  "battery_charge_power_max": 7680,
  "battery_discharge_efficiency": 0.95,
  "battery_discharge_power_max": 7680,
  "battery_dynamic_max": 0.8,
  "battery_dynamic_min": -0.8,
  "battery_maximum_state_of_charge": 1,
  "battery_minimum_state_of_charge": 0.15,
  "battery_nominal_energy_capacity": 7680,
  "battery_target_state_of_charge": 0.6,
  "compute_curtailment": false,
  "continual_publish": false,
  "costfun": "cost",
  "delta_forecast_daily": 1,
  "end_timesteps_of_each_deferrable_load": [
    0,
    0
  ],
  "historic_days_to_retrieve": 9,
  "inverter_is_hybrid": true,
  "load_cost_forecast_method": "hp_hc_periods",
  "load_forecast_method": "naive",
  "load_negative": false,
  "load_offpeak_hours_cost": 0.25,
  "load_peak_hour_periods": {
    "period_hp_1": [
      {
        "start": "00:00"
      },
      {
        "end": "02:59"
      }
    ],
    "period_hp_2": [
      {
        "start": "05:00"
      },
      {
        "end": "23:59"
      }
    ]
  },
  "load_peak_hours_cost": 0.35,
  "logging_level": "INFO",
  "lp_solver": "COIN_CMD",
  "lp_solver_path": "/usr/bin/cbc",
  "maximum_power_from_grid": 9000,
  "maximum_power_to_grid": 9000,
  "method_ts_round": "nearest",
  "modules_per_string": [
    14,
    11
  ],
  "nominal_power_of_deferrable_loads": [
    1600
  ],
  "number_of_deferrable_loads": 1,
  "operating_hours_of_each_deferrable_load": [
    1
  ],
  "optimization_time_step": 60,
  "photovoltaic_production_sell_price": 0.082,
  "production_price_forecast_method": "constant",
  "pv_inverter_model": [
    "Fronius_International_GmbH__Primo_GEN24_6_0_208_240_Plus__240V_"
  ],
  "pv_module_model": [
    "AXITEC_AC_400MH_144S"
  ],
  "sensor_linear_interp": [
    "sensor.haus_ohne_gesteuerte_lasten_power"
  ],
  "sensor_power_load_no_var_loads": "sensor.haus_ohne_gesteuerte_lasten_power",
  "sensor_power_photovoltaics": "sensor.pv_leistung",
  "sensor_replace_zero": [
    "sensor.pv_leistung"
  ],
  "set_battery_dynamic": true,
  "set_deferrable_load_single_constant": [
    true,
    true
  ],
  "set_deferrable_startup_penalty": [
    0,
    0
  ],
  "set_nocharge_from_grid": false,
  "set_nodischarge_to_grid": true,
  "set_total_pv_sell": false,
  "set_use_battery": true,
  "set_zero_min": true,
  "start_timesteps_of_each_deferrable_load": [
    0,
    0
  ],
  "strings_per_inverter": [
    2
  ],
  "surface_azimuth": [
    90,
    270
  ],
  "surface_tilt": [
    45,
    45
  ],
  "treat_deferrable_load_as_semi_cont": [
    true,
    true
  ],
  "weather_forecast_method": "scrapper",
  "weight_battery_charge": 0.05,
  "weight_battery_discharge": 0.05
}
```

any ideas?

```
2024-12-13 20:36:26,847 - web_server - INFO - Passed runtime parameters: {'entity_save': True, 'publish_prefix': 'emhass_dah_', 'load_cost_forecast': [0.4006, 0.3663, 0.3531, 0.331, 0.3315, 0.3248, 0.3209, 0.3101, 0.3035, 0.3013, 0.3022, 0.3133, 0.3277, 0.341, 0.3362, 0.3202, 0.3169, 0.3148, 0.3073, 0.3092, 0.3181, 0.3364, 0.3032, 0.3061]}
2024-12-13 20:36:26,848 - web_server - INFO -  >> Setting input data dict
2024-12-13 20:36:26,848 - web_server - INFO - Setting up needed data
2024-12-13 20:36:26,864 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2024-12-13 20:36:29,602 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2024-12-13 20:36:29,604 - web_server - INFO - Retrieve hass get data method initiated...
2024-12-13 20:36:40,313 - web_server - INFO -  >> Performing dayahead optimization...
2024-12-13 20:36:40,313 - web_server - INFO - Performing day-ahead forecast optimization
2024-12-13 20:36:40,339 - web_server - INFO - Perform optimization for the day-ahead
2024-12-13 20:36:40,587 - web_server - INFO - Status: Infeasible
2024-12-13 20:36:40,588 - web_server - INFO - Total value of the Cost function = -4.09
2024-12-13 20:36:40,623 - web_server - INFO - Publishing data to HASS instance
```

```
2024-12-13 20:38:41,560 - web_server - INFO - Passed runtime parameters: {'entity_save': True, 'publish_prefix': 'emhass_', 'prediction_horizon': 24, 'load_cost_forecast': [0.4006, 0.3663, 0.3531, 0.331, 0.3315, 0.3248, 0.3209, 0.3101, 0.3035, 0.3013, 0.3022, 0.3133, 0.3277, 0.341, 0.3362, 0.3202, 0.3169, 0.3148, 0.3073, 0.3092, 0.3181, 0.3364, 0.3032, 0.3061], 'soc_init': 0.15, 'soc_final': 0.6}
2024-12-13 20:38:41,563 - web_server - INFO -  >> Setting input data dict
2024-12-13 20:38:41,563 - web_server - INFO - Setting up needed data
2024-12-13 20:38:41,606 - web_server - INFO - Retrieve hass get data method initiated...
2024-12-13 20:38:49,936 - web_server - INFO - Retrieving weather forecast data using method = scrapper
2024-12-13 20:38:52,575 - web_server - INFO - Retrieving data from hass for load forecast using method = naive
2024-12-13 20:38:52,577 - web_server - INFO - Retrieve hass get data method initiated...
2024-12-13 20:39:09,938 - web_server - INFO -  >> Performing naive MPC optimization...
2024-12-13 20:39:09,939 - web_server - INFO - Performing naive MPC optimization
2024-12-13 20:39:09,967 - web_server - INFO - Perform an iteration of a naive MPC controller
2024-12-13 20:39:10,253 - web_server - INFO - Status: Infeasible
2024-12-13 20:39:10,254 - web_server - INFO - Total value of the Cost function = -5.00
2024-12-13 20:39:10,292 - web_server - INFO - Publishing data to HASS instance
```

PR #380 (2024-12-20): Update default battery weights
Battery weights dis/charge should be 0.0

Fixes: #369
commit 7fce7794 (2024-12-20): Update default battery weights

Battery weights dis/charge should be 0.0

Fixes: #369
closed 2024-12-20 by davidusb-geek