davidusb-geek/emhass #390: MPC optim status infeasible when SOC lower than min SOC

martinarva, 2024-12-03:
**Describe the bug**
When SOC init is lower than min SOC, MPC run optim status will be infeasible. 

This night my automations did not work because of infeasible optim status. I checked the logs and i'm pretty sure it's because my min SOC is 15%, but whatever reason my real SOC went to 14%.

For now I made a workaround like this:
`        "soc_init": {{
          [((states('sensor.ss_battery_soc') | int(0)) / 100), 0.15] | max
        }},
`
**To Reproduce**
Set min SOC higher than your real SOC that you use as SOC init in MPC run

**Expected behavior**
I would expect optim status won't be infeasible in this scenario.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Home Assistant installation type**
 - Home Assistant Core

**Your hardware**
- OS: Linux
- Architecture: amd64

**EMHASS installation type**
 - Docker Standalone,

twids, 2024-12-13:
I have seen this aswell, same thing happens if SOC is above the maximum battery charge percentage

davidusb-geek, 2024-12-13:
> Expected behavior
I would expect optim status won't be infeasible in this scenario.

I don't understand this issue.
If you pass a SOC init that is lower than the SOC min then I do expect that the outcome is infeasible.
How would you expect this to not be infeasible?

RudolfRendier, 2024-12-16:
Maybe the constraint could be less strict; can it be adjusted to only prevent discharges below the soc_min instead of treating it as an invalid value in general?

That way EMHASS could recover by charging the battery first, after a deep discharge, beyond the usual levels.
(i.e. in winter, when the battery is not charged with solar and simple loses charge over time)

davidusb-geek, 2025-01-31:
Proposed and merged a solution for this in #442

closed 2025-01-31 by davidusb-geek