davidusb-geek/emhass #391: Feature request: Providing operating periods for deferrable loads as list of time steps

RenatusRo, 2024-12-05:
Overruling the start_timesteps_of_each_deferrable_load, end_timesteps_of_each_deferrable_load params
it would be nice to convey the operating hours as a list of all eligible time steps:
{"operating_timesteps_load0": [0,0,0,0,0,0,1,1,1,1,1,1,1,0,0,0,0,0,0,0,1,1,1]}
in order to enable more than one operating time window per day.

I greatly appreciate your efforts!

davidusb-geek, 2024-12-21:
This seems like a nice option indeed, more intuitive when compared to the current implementation of this feature.
At some point in the past i thought about this and then someone implemented the current version.
It is a matter of improving the ease of use of the code.

davidusb-geek, 2026-11-03:
Closing this out as we want to keep the deferrable load window parameters simple, and there are now two ways to achieve multiple operating windows per day:

Multiple virtual deferrable loads: Define two deferrable load slots linked to the same physical device, each with its own start_timesteps_of_each_deferrable_load and end_timesteps_of_each_deferrable_load window.

Per-load cost mask (cost_forecast_per_deferrable_load): Pass a custom per-timestep cost array for that load where eligible timesteps use your normal tariff and forbidden timesteps use a high penalty cost (e.g., 100.0) so the solver only schedules the load inside your allowed windows.

closed 2026-11-03 by davidusb-geek