davidusb-geek/emhass #394: Feature request: Nord Pool integration get data from within HA

BDVGitHub, 2024-12-14:
Since there is a [Nord Pool integration](https://www.home-assistant.io/integrations/nordpool/) for Home Assistant is it possible to use this data in Emhass?

davidusb-geek, 2024-12-15:
Hi and sorry but I don't understand your feature request. Isn't that shell command that you are posting good enough? What is the new feature that you are seeking? That is how it works, you pass the data available from the many different providers out there as argument data in a shell command.

BDVGitHub, 2024-12-15:
What I want to say now that Home Assistant has added Nord Pool as an integration. 
[image]

Is it possible to automatically retrieve the Nord Pool data contained in Home Assistant? So that you no longer have to use the shell command? But only have to indicate e.g. in the config file that you use Nord Pool data available in Home Assistant?
[image]
[image]
[image]

davidusb-geek, 2024-12-23:
I understand that it can be even easier to just tell EMHASS to use Nordpool and it connects directly with HA and retrieves the data.
But in reality this is not a smart move because each integration is unique, so if we do this for Nordpool then we need to do it for Amber, etc. The same principle for the many options to provide solar forecasts for example. The approach that we took is that data coming from an external provider should passed to EMHASS at runtime. So possibly we need to focus on how to ease in the setup of shell commands. The blueprints are a good option, provide a blueprint for each external data provider. Open to contributions on this.

davidusb-geek, 2024-12-23:
Closing as not planned

closed 2024-12-23 by davidusb-geek

BDVGitHub, 2024-12-24:
A think I understand what you mean, the blueprint in HomeAssistant then acts as a software connector to let the HA integration exchange data with Emhass. Is this correct?