davidusb-geek/emhass #397: Prediction Horizon runtime parameter not understood

altonius, 2024-12-23:
**Describe the bug**
After upgrading from 0.11.2 to 0.11.3 the MPC prediction is not working as it previously did. From looking at the logs it appears to not be detecting / using the `prediciton_horizon` runtime parameter, as a 41 is the horizon, however the logs say that the list should be 48.

**To Reproduce**
include a runtime_parameter under 48 in the MPC request.

**Expected behavior**
If the prediction horizon is set at a number lower than 48 then it should accept the prediction_horizon parameter and predict accordingly.

**Logs**
```
[2024-12-24 02:19:00,314] INFO in web_server: Passed runtime parameters: {'load_power_forecast': [187, 400, 300, 300, 300, 300, 300, 300, 300, 300, 500, 400, 400, 700, 1500, 1100, 400, 2400, 1300, 900, 1700, 2100, 2600, 2400, 2000, 1600, 1200, 1200, 1200, 1300, 1200, 1200, 2600, 2000, 800, 900, 700, 1000, 600, 500, 400, 300, 500, 300, 800, 300, 500, 400], 'load_cost_forecast': [0.16, 0.14, 0.14, 0.14, 0.14, 0.14, 0.14, 0.14, 0.13, 0.11, 0.08, 0.06, 0.06, 0.05, 0.04, 0.03, 0.03, 0.02, 0.02, 0.02, 0.02, 0.02, 0.02, 0.02, 0.02, 0.03, 99.0, 99.0, 99.0, 99.0, 99.0, 99.0, 99.0, 99.0, 99.0, 99.0, 99.0, 99.0, 0.15, 0.15, 0.15], 'prod_price_forecast': [0.08, 0.06, 0.06, 0.06, 0.06, 0.06, 0.06, 0.06, 0.06, 0.03, 0.0, -0.01, -0.01, -0.02, -0.03, -0.03, -0.05, -0.06, -0.06, -0.06, -0.06, -0.06, -0.06, -0.06, -0.06, -0.05, -0.04, -0.04, -0.01, -0.0, 0.01, 0.03, 0.04, 0.07, 0.09, 0.11, 0.11, 0.1, 0.07, 0.07, 0.07], 'pv_power_forecast': [0, 0, 0, 0, 0, 0, 0, 53, 934, 1505, 2324, 3414, 4493, 5515, 6472, 7294, 8053, 8630, 8981, 9481, 9724, 9836, 9807, 9657, 9383, 8987, 8470, 7819, 7051, 6184, 5222, 4207, 3112, 2011, 957, 188, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 'prediction_horizon': 41, 'operating_hours_of_each_deferrable_load': [4], 'soc_init': 0.06, 'soc_final': 0.05, 'alpha': 1, 'beta': 0, 'battery_charge_power_max': 10006, 'def_current_state': [0]}
2024-12-24 02:19:00,314 - web_server - INFO - Passed runtime parameters: {'load_power_forecast': [187, 400, 300, 300, 300, 300, 300, 300, 300, 300, 500, 400, 400, 700, 1500, 1100, 400, 2400, 1300, 900, 1700, 2100, 2600, 2400, 2000, 1600, 1200, 1200, 1200, 1300, 1200, 1200, 2600, 2000, 800, 900, 700, 1000, 600, 500, 400, 300, 500, 300, 800, 300, 500, 400], 'load_cost_forecast': [0.16, 0.14, 0.14, 0.14, 0.14, 0.14, 0.14, 0.14, 0.13, 0.11, 0.08, 0.06, 0.06, 0.05, 0.04, 0.03, 0.03, 0.02, 0.02, 0.02, 0.02, 0.02, 0.02, 0.02, 0.02, 0.03, 99.0, 99.0, 99.0, 99.0, 99.0, 99.0, 99.0, 99.0, 99.0, 99.0, 99.0, 99.0, 0.15, 0.15, 0.15], 'prod_price_forecast': [0.08, 0.06, 0.06, 0.06, 0.06, 0.06, 0.06, 0.06, 0.06, 0.03, 0.0, -0.01, -0.01, -0.02, -0.03, -0.03, -0.05, -0.06, -0.06, -0.06, -0.06, -0.06, -0.06, -0.06, -0.06, -0.05, -0.04, -0.04, -0.01, -0.0, 0.01, 0.03, 0.04, 0.07, 0.09, 0.11, 0.11, 0.1, 0.07, 0.07, 0.07], 'pv_power_forecast': [0, 0, 0, 0, 0, 0, 0, 53, 934, 1505, 2324, 3414, 4493, 5515, 6472, 7294, 8053, 8630, 8981, 9481, 9724, 9836, 9807, 9657, 9383, 8987, 8470, 7819, 7051, 6184, 5222, 4207, 3112, 2011, 957, 188, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], 'prediction_horizon': 41, 'operating_hours_of_each_deferrable_load': [4], 'soc_init': 0.06, 'soc_final': 0.05, 'alpha': 1, 'beta': 0, 'battery_charge_power_max': 10006, 'def_current_state': [0]}
[2024-12-24 02:19:00,315] INFO in web_server:  >> Setting input data dict
2024-12-24 02:19:00,315 - web_server - INFO -  >> Setting input data dict
[2024-12-24 02:19:00,315] WARNING in web_server: profit
2024-12-24 02:19:00,315 - web_server - WARNING - profit
[2024-12-24 02:19:00,315] INFO in command_line: Setting up needed data
2024-12-24 02:19:00,315 - web_server - INFO - Setting up needed data
[2024-12-24 02:19:00,318] ERROR in utils: ERROR: The passed data is either not a list or the length is not correct, length should be 48
2024-12-24 02:19:00,318 - web_server - ERROR - ERROR: The passed data is either not a list or the length is not correct, length should be 48
[2024-12-24 02:19:00,318] ERROR in utils: Passed type is <class 'list'> and length is 41
2024-12-24 02:19:00,318 - web_server - ERROR - Passed type is <class 'list'> and length is 41
[2024-12-24 02:19:00,318] ERROR in utils: ERROR: The passed data is either not a list or the length is not correct, length should be 48
2024-12-24 02:19:00,318 - web_server - ERROR - ERROR: The passed data is either not a list or the length is not correct, length should be 48
[2024-12-24 02:19:00,319] ERROR in utils: Passed type is <class 'list'> and length is 41
2024-12-24 02:19:00,319 - web_server - ERROR - Passed type is <class 'list'> and length is 41
```

**Home Assistant installation type**
- Core 2024.12.5
- Supervisor 2024.12.0
- Operating System 14.1
- Frontend 20241127.8

**Your hardware**
- OS: HA OS
- Architecture: amd64

**EMHASS installation type**
0.11.3

**Additional context**
Add any other context about the problem here.

altonius, 2024-12-23:
maybe related to #396 ?

GeoDerp, 2024-12-24:
Thanks for the detailed info @altonius , it's now fixed in that PR.

altonius, 2024-12-25:
resolved in 0.11.4

closed 2024-12-25 by altonius