davidusb-geek/emhass #398: ModuleNotFoundError: No module named 'skforecast.ForecasterAutoreg'

adamlonsdale, 2024-12-23:
**Describe the bug**
The update to skforecast 0.14 includes some breaking changes that impact the ML Forecaster. Looks like there is a [migration guide](https://github.com/skforecast/skforecast/blob/88da967670344453cdc1ab775e4c228cdc394617/docs/user_guides/migration-guide.ipynb#L50) after some organisation changes in the upstream packages.

`ModuleNotFoundError: No module named 'skforecast.ForecasterAutoreg'`

**To Reproduce**
Sending a call to `/action/forecast-model-predict` in my case shows the error message via console output in the container. 

**Expected behavior**
SKForecast 0.14 should be used with the breaking changes

**Home Assistant installation type**
 - Home Assistant Core

**Your hardware**
- OS: Arch Linux
- Architecture: amd64

**EMHASS installation type**
 - Docker via Kubernetes

**Additional context**
 I'm happy to take a look and raise a PR but I don't have an Emhass test environment at the moment.

davidusb-geek, 2024-12-23:
I did followed the migration guide.
Please try again but first erase the ML (.pkl) files saved with the previous version of EMHASS.
Then relaunch a train command and finally you should be able to do a predict with no issue.

adamlonsdale, 2024-12-23:
ah, yep, my apologies. I'd seen the older code in the repo but it was just one of the scripts. I've taken your advice and that is working now, thanks David - apologies for the report!

closed 2024-12-23 by adamlonsdale